esphome/issues #5925: 2024.6.1: `status_code` was not declared in this scope or HttpContainer>` has no member named `status_code`

andrewjswan, 2024-06-20:
### The problem

In lambda function, error: `status_code` was not declared in this scope
or
In lambda function, error: `class std::shared_ptr<esphome::http_request::HttpContainer>` has no member named `status_code`

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.3

### What platform are you using?

ESP32

### Board

Esp32 Node MCU

### Component causing the issue

http_request

### Example YAML snippet

```yaml
# 2024.5.3 Config
script:
  - id: send_state
    then:
      - http_request.get: 
          url: !lambda |-
            return ((std::string) "https://api.site.com/ping?key=" + id(my_key).state).c_str();
          on_response:
            then:
              - lambda: |-
                  if (status_code == 200)
                  {
                    id(send_status).publish_state(true);
                    id(ok)++;
                  }
                  else
                  {
                    id(send_status).publish_state(false);
                    id(error)++;
                  }
              - component.update: quality
    mode: queued

# 2024.6.1 Config
script:
  - id: send_state
    then:
      - http_request.get: 
          url: !lambda |-
            return ((std::string) "https://api.site.com/ping?key=" + id(my_key).state).c_str();
          on_response:
            then:
              - lambda: |-
                  if (response.status_code == 200)
                  {
                    id(send_status).publish_state(true);
                    id(ok)++;
                  }
                  else
                  {
                    id(send_status).publish_state(false);
                    id(error)++;
                  }
              - component.update: quality
    mode: queued
```

### Anything in the logs that might be useful for us?

```txt
When building with ESPHome 2024.6.1, if I specify in the config as it was in 2024.5.3 (current Documentation, section `http_request.get Action`), the error is:

In lambda function:\nsvitlobot.yaml:83:11: error: 'status_code' was not declared in this scope
```

When building with ESPHome 2024.6.1, if I specify it as in the current documentation in the `on_response Trigger` section, then the error is:
```
In lambda function:\nsvitlobot.yaml:82:20: error: 'class std::shared_ptr<esphome::http_request::HttpContainer>' has no member named 'status_code'
```
```

mveinot, 2024-06-20:
Seeing this issue as well.

mveinot, 2024-06-20:
Friend of mine dug into the code base a bit and discovered that if you dereference, it works:

`response->status_code`

andrewjswan, 2024-06-20:
> it works:

Yes, it works. Looks like the Documentation needs to be updated.

guillempages, 2024-07-21:
Fixed with documentation update (esphome/esphome-docs#3987)

closed 2024-07-21 by guillempages