esphome/issues #5926: Error: Downlading ESPHome firmware for ESP8266 failed (404)

Aad3229, 2024-06-20:
### The problem

After succesfully connecting over USB in Chrome (https://web.esphome.io/) to an ESP8266 I try to "Prepare for first use" --> "install" I get"Error: Downlading ESPHome firmware for ESP8266 failed (404)" 

### Which version of ESPHome has the issue?

2024.5.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.2

### What platform are you using?

ESP8266

con0aks, 2024-06-20:
Same here, also for different models like the ESP32S3

remcom, 2024-06-24:
Same here. maybe the issue that is caused now the file names have been changed since last 2024.6?

BarBaar44, 2024-07-05:
I can confirm this one, and make it bigger.

When creating a firmware file, it cannot be downloaded using the option to flash using ESPHome Web or manual download. The browser encounters a 404.

`2024-07-05 16:11:09,380 INFO Running command 'esphome --dashboard compile /config/esphome/human-sensor.yaml'
2024-07-05 16:11:57,337 INFO Process exited with return code 0
2024-07-05 16:11:58,242 INFO 304 GET /downloads?configuration=human-sensor.yaml (0.0.0.0) 1.42ms
2024-07-05 16:12:01,463 WARNING 404 GET /download.bin?configuration=human-sensor.yaml&file=firmware.factory.bin&download=screek-human-sensor.factory.bin (0.0.0.0) 957.61ms`

Checked, and the file does exist in the filesystem:
`root@5c53de3b-esphome:/data/build/screek-human-sensor/.pioenvs/screek-human-sensor# ls
FrameworkArduino  bootloader.bin  firmware-factory.bin	firmware.bin  firmware.elf  firmware.map  idedata.json	lib01c	lib0bd	lib333	lib4d0	lib4fc	lib64d	lib679	lib915	libFrameworkArduino.a  libbc6  libbd7  libd65  partitions.bin  src
`

ssieb, 2024-07-08:
This should have been fixed already.  Do you have the latest version?

BarBaar44, 2024-07-09:
Yes. I did a little further digging and found removing the code below fixed the issue for me

external_components:
  - source:
      type: git
      url: https://github.com/screekworkshop/custom_components_support_for_screek_2a
      ref: main
    components: [esp32, uart]

closed 2024-07-09 by ssieb

restauffer, 2024-07-17:
I note @BarBaar44 was having this problem with an Screek repository. I wonder if the others were also trying with an Screek repository? It would be interesting to know if this is a problem with ESPHome or only with Screek repositories. I can say I have seen it with two different Screek repositories although I suspect the contain similar code.

The solution offered by @BarBaar44 is not a good one as the external components he commented out, are needed. It may be possible to work around the problems created by the lost if those components.

After quite a bit of testing I know how to make it work but not why it is necessary. If you try to install the problematic yaml via USB cable it gives the "404 Error" but if you put the device on wifi first and then try to install the full YAML it works.

I have opened an issue on his problem at:
https://github.com/screekworkshop/screek-human-sensor/issues/22
In it, in a recent comment I summarized what I have learned so far.