esphome/issues #5927: Modbus CRC Check failed! still happening !!!!

dr3amr, 2024-06-20:
### The problem

Hi team,

Bellow errors can be still seen even on latest ESP 2024.6.1, the data still randomly comes through, but 80% is as bellow:

[13:04:28][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:04:28][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:04:28][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:04:28][W][modbus:114]: Modbus CRC Check failed! C071!=00
[13:04:28][W][modbus:114]: Modbus CRC Check failed! C071!=4842

Only real workaround, found from other threads, is using arduino version 2.0.6, which when used works perfect, but that is old version:

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino
    version: 2.0.6  ===> workaround

Is there any other way to make it work and stay updated ?

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.3

### What platform are you using?

ESP32

### Board

az-delivery-devkit-v4

### Component causing the issue

modbus

### Example YAML snippet

```yaml
uart:
  id: rs485_uart
  rx_pin: GPIO17
  tx_pin: GPIO18
  baud_rate: 9600
  stop_bits: 1
#  debug:
#    direction: BOTH
#    dummy_receiver: false
#    after:
  #    delimiter: "\n"
  #  sequence:
  #   - lambda: UARTDebug::log_string(direction, bytes);
  
modbus:
  uart_id: rs485_uart
  id: modbus_sdm
  send_wait_time: 500ms
#  flow_control_pin: GPIO21

  - platform: sdm_meter
    modbus_id: modbus_sdm
    phase_a:
      current:
        name: "${name} Current"
      voltage:
        name: "${name} Voltage"
      active_power:
        name: "${name} Power"
      power_factor:
        name: "${name} Power Factor"
      apparent_power:
        name: "${name} Apparent Power"
      reactive_power:
        name: "${name} Reactive Power"
      phase_angle:
        name: "${name} Phase Angle"
    frequency:
      name: "${name} Frequency"
    import_active_energy:
      name: "${name} Import Active Energy"
    export_active_energy:
      name: "${name} Export Active Energy"
    import_reactive_energy:
      name: "${name} Import Reactive Energy"
    export_reactive_energy:
      name: "${name} Export Reactive Energy"
```

TheOriginalMrWolf, 2024-10-04:
I had a similar problem, also with a power meter.

Setting `stop_bits: 2` cured it - 100% perfect comms from that point on.  Never got to the bottom of it as manufacturer swore it should only be 1 stop bit…  might be minuscule timing differences… who knows???

Either way, give it a go and see! 😁

dr3amr, 2024-10-04:
@TheOriginalMrWolf  do you also have an Eastron meter ? If so what model ? platform: sdm_meter is for Eastron power meters. UART is generic not specific clearly but stop_bits: 2 might not work for Eastron power meters, but for your specific meter.

TheOriginalMrWolf, 2024-10-04:
Try the suggestion @dr3amr & report back - does

dr3amr, 2024-10-04:
@TheOriginalMrWolf tried it. Not working. Using same workaround which seems to work fine to using 2.0.6 only version of Arduino framework.

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino
    version: 2.0.6

closed 2024-10-29 by jesserockz

dr3amr, 2024-10-29:
I confirm that the fix in #7674 is working, glad it was merged. Will stay with the workaround until the new ESPHome version comes. This was a very old and frustrating issue.