esphome/issues #5932: [2024.6.1] GPIO8 can no longer be used for UART/RX on the Denky D4

alucryd, 2024-06-20:
### The problem

The latest update broke ESPHome on the Denky D4, typically used for Linky Teleinfo in France. We can no longer select GPIO8 for UART/RX, it was working fine in 2024.5.x.

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.3

### What platform are you using?

ESP32-IDF

### Board

Denky D4

### Component causing the issue

uart

### Example YAML snippet

```yaml
esphome:
  name: linky
  friendly_name: Linky

esp32:
  board: denky_d4
  framework:
    type: esp-idf

uart:
  rx_pin: GPIO8
  baud_rate: 9600
  parity: EVEN
  data_bits: 7
  stop_bits: 1
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/linky.yaml...
Failed config

uart: [source /config/esphome/linky.yaml:40]
  - 
    This pin cannot be used on ESP32s and is already used by the flash interface (function: Flash Data 1).
    rx_pin: GPIO8
    baud_rate: 9600
    parity: EVEN
    data_bits: 7
    stop_bits: 1
```

ezlo-picori, 2024-06-21:
I also just faced this issue, it is a consequence of https://github.com/esphome/esphome/pull/5615 .
You may replace

```yaml
  rx_pin: GPIO8
```

by

```yaml
  rx_pin:
    number: GPIO8
    ignore_pin_validation_error: true
```
to silence this warning

seblang, 2024-06-21:
@ezlo-picori 

same issue for me but if i change in my yaml 👍 

uart:
  rx_pin: GPIO8
  ignore_pin_validation_error: true
  baud_rate: 9600
  parity: EVEN
  data_bits: 7
  stop_bits: 1

with 

uart:
  rx_pin:
    number: GPIO8
    ignore_pin_validation_error: true
  baud_rate: 9600
  parity: EVEN
  data_bits: 7
  stop_bits: 1

it's work than you

alucryd, 2024-06-21:
Worked like a charm, thank you!

closed 2024-06-21 by alucryd