esphome/issues #5933: WPA Enterprise compile fails on framework arduino after update 2024.6.1

gnsbglora, 2024-06-20:
### The problem

After the update 2024.6.1 Wifi WPA Enterprise doesn't compile on framework arduino anymore. On framework esp-idf it's working. The parameter ttls_phase_2: mschapv2 is accepted on esp-idf only.

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.3

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

wifi wpa enterprise

### Example YAML snippet

```yaml
esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  domain: .domain.net
  networks:
  - ssid: ssid
    eap:
      username: !secret wifi_username
      identity: !secret wifi_identity
      password: !secret wifi_eap_pwd
      #ttls_phase_2: mschapv2
  - ssid: ssid
    eap:
      identity: !secret wifi_identity
      certificate_authority: /ssl/cacert.pem
```

### Anything in the logs that might be useful for us?

```txt
> INFO ESPHome 2024.6.1
> INFO Reading configuration /config/esphome/esp32-pool.yaml...
> WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
> Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
> See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
> INFO Generating C++ source...
> INFO Compiling app...
> Processing esp32-pool (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)

> --------------------------------------------------------------------------------
> Removing unused dependencies...
> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
>  - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
> Dependency Graph
> |-- WiFi @ 2.0.0
> |-- ESPmDNS @ 2.0.0
[... 101 lines cut ...]
> Building .pioenvs/esp32-pool/bootloader.bin
> Creating esp32 image...
> Successfully created esp32 image.
> Generating partitions .pioenvs/esp32-pool/partitions.bin
> Compiling .pioenvs/esp32-pool/lib64d/WiFi/WiFi.cpp.o
> Compiling .pioenvs/esp32-pool/lib64d/WiFi/WiFiAP.cpp.o
> Compiling .pioenvs/esp32-pool/lib64d/WiFi/WiFiClient.cpp.o
> src/main.cpp: In function 'void setup()':
> src/main.cpp:141:23: error: 'ESP_EAP_TTLS_PHASE2_MSCHAPV2' was not declared in this scope
>        .ttls_phase_2 = ESP_EAP_TTLS_PHASE2_MSCHAPV2,
>                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
> Compiling .pioenvs/esp32-pool/lib64d/WiFi/WiFiGeneric.cpp.o
> src/main.cpp:142:3: error: 'esphome::wifi::EAPAuth' has no non-static data member named 'ttls_phase_2'
>    });
>    ^
> src/main.cpp:156:23: error: 'ESP_EAP_TTLS_PHASE2_MSCHAPV2' was not declared in this scope
>        .ttls_phase_2 = ESP_EAP_TTLS_PHASE2_MSCHAPV2,
>                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
> src/main.cpp:157:3: error: 'esphome::wifi::EAPAuth' has no non-static data member named 'ttls_phase_2'
>    });
>    ^
> Compiling .pioenvs/esp32-pool/lib64d/WiFi/WiFiMulti.cpp.o
> *** [.pioenvs/esp32-pool/src/main.cpp.o] Error 1
> ========================== [FAILED] Took 8.94 seconds ==========================
```

Dark-Sarmat, 2024-06-21:
I have exactly the same problem.
I'm waiting for the fix

kc2zgu, 2024-06-26:
I also ran into this; seems to be introduced in 2024.6.0. On a ESP32C3 XIAO board I could compile with esp-idf but then nothing would actually run. It appeared to just reboot every 10 seconds with some log messages being sent to the serial port. Had to roll back to 2024.5.5.

nanctil, 2024-07-02:
ditto

Pack3tL0ss, 2024-07-02:
Same here, all eap-tls devices fail to compile now

jesserockz, 2024-07-03:
Looks this was an oversight of https://github.com/esphome/esphome/pull/6436 that set the default to mschapv2 for all configs.
Ill try put out a fix today

jesserockz, 2024-07-03:
```yaml
external_components:
  - source: github://pr#7033
  	components: wifi
```

This above snippet should fix the issue for you until 2024.6.5 is released.