esphome/issues #5934: Wifi with EAP does not compile

Sven3er, 2024-06-20:
### The problem

The default ESP32 example with an EAP network does not compile due to compiler error:
'esphome::wifi::EAPAuth' has no non-static data member named 'ttls_phase_2'

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

wifi

### Example YAML snippet

```yaml
substitutions:
  device_name: "device_name"
  friendly_name: "friendly_name"

esphome:
  name: ${device_name}
  friendly_name: ${friendly_name}

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  id: api_id
  encryption:
    key: !secret esp-home_apiKey

ota:
  - platform: esphome
    password: !secret esp-home_otaPW

wifi:
  networks:
  - ssid: !secret wifi_ssid_eap
    eap:
      username: !secret eap_user
      password: !secret eap_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    password: !secret esp-home_wifiDefaultPW

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
src/main.cpp: In function 'void setup()':
src/main.cpp:98:23: error: 'ESP_EAP_TTLS_PHASE2_MSCHAPV2' was not declared in this scope
       .ttls_phase_2 = ESP_EAP_TTLS_PHASE2_MSCHAPV2,
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/device_name/lib64d/WiFi/WiFiServer.cpp.o
src/main.cpp:99:3: error: 'esphome::wifi::EAPAuth' has no non-static data member named 'ttls_phase_2'
   });
   ^
*** [.pioenvs/device_name/src/main.cpp.o] Error 1
```

randybb, 2024-06-20:
Duplicate of https://github.com/esphome/issues/issues/5933

ziolelle, 2024-06-21:
same issue here...
what does "closed this as not planned" mean ?