esphome/issues #5935: safe_mode error with Apollo Air-1 + 2024.6.1 update

shurickdaryin, 2024-06-20:
### The problem

I cannot update 2024.5.5 -> 2024.6.1 on Apollo Air-1 because of the following error:

```
INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/apollo-air-1-bd54fc.yaml...
Failed config

ota.esphome: [source /config/esphome/apollo-air-1-bd54fc.yaml:11]
  password: apolloautomation
  
  'safe_mode' (and its related configuration variables) has moved from 'ota' to its own component. See https://esphome.io/components/safe_mode.
  safe_mode: False
  platform: esphome
```

The line in question (11) does not contain either password or safe_mode:

```yaml
ota: { platform: esphome }
```

which probably means that safe_mode is inserted automatically by the updater in the wrong place.

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.3

### What platform are you using?

ESP8266

randybb, 2024-06-20:
[safe_mode](https://esphome.io/components/safe_mode) has been taken out from OTA ([release notes 2024.6](https://esphome.io/changelog/2024.6.0.html)), it is now a separate component. To disable it you need to add:
```yaml
safe_mode:
  disabled: true
```

If you including an external config, you need to wait until it is updated by the vendor.

shurickdaryin, 2024-06-21:
I wonder if it's possible to migrate the config format in a more graceful way.

There were at least two changes:

* added required field ota.platform
* ota.safe_mode moved to a separate component

It is possible to fix "platform" by editing the configs so that the merged config contains "platform". However, I could not find a way to definitely remove safe_mode by editing the configs - the merged config still contains safe_mode and the validation fails.

Maybe it would be possible, for some migration time window, to have a default for "ota.platform" (esphome) and to automatically copy ota.safe_mode if it's specified there.

keniji, 2024-06-22:
> [safe_mode](https://esphome.io/components/safe_mode) has been taken out from OTA ([release notes 2024.6](https://esphome.io/changelog/2024.6.0.html)), it is now a separate component. To disable it [...]

Tried with a new created test device, but could no make it complied.

If I put it as a separate component, like:
```
ota:
  - platform: esphome
    password: "7d8de822dcfd9478c4aa4daa0d2ea67d"

safe_mode:
  disabled: true
```
Will get an error:
`ERROR Circular dependency detected! Please run with -v option to see what functions failed to complete.`
If I edit like:
```
ota:
  - platform: esphome
    password: "7d8de822dcfd9478c4aa4daa0d2ea67d"
    safe_mode: false

safe_mode:
  disabled: true
```
Or
```
ota:
  - platform: esphome
    password: "7d8de822dcfd9478c4aa4daa0d2ea67d"
    safe_mode: false
```
It would not pass the validation at all. Anyone knows how to disable the safe mode?

labodj, 2024-06-22:
Same problem on my end using a different platform, this yaml does not work:

```yaml
esphome:
  name: device-name

esp8266:
  board: d1_mini_pro

safe_mode:
  disabled: true

ota:
  platform: esphome

wifi:
  networks:
    - ssid: Wifi_SSID
      password: WIFI_PWD
```

error code:

> ERROR Circular dependency detected! Please run with -v option to see what functions failed to complete.

error using -v:

> (infinite loop of) DEBUG Running to_code in esphome.components.wifi (num 7)

The only "solution" is to enable safe mode

```yaml
safe_mode:
  #disabled: true
```

Anyway this is not a real solution, since I need safe mode to be disabled.

shurickdaryin, 2024-06-22:
The error is gone. Not sure what changed (HA core update?), but now the config is valid with the same ESPHome version and the same base config from the vendor. Magic ;)

N1c093, 2024-06-25:
I can also confirm, that this code 
```yaml
safe_mode:
  disabled: true
```
create this error during installing the firmware:

`ERROR Circular dependency detected! Please run with -v option to see what functions failed to complete.`

If I enable the safe mode, I can create the firmware successfully:
```yaml
safe_mode:
  #disabled: true
```

closed 2024-06-25 by kbx81