esphome/issues #5936: the new one_wire  / dallas_temp are not working (Found no devices!)

Links2004, 2024-06-20:
### The problem

after updateing to 2024.6.1 i get "Found no devices!" for my one wire bus.
this happens with multiple devices ESP32 and ESP32-C3 based

where there any changes other the the rename like to the timing, internal pullups?

example ESP32-C3:

log:
```
[17:01:45][C][gpio.one_wire:016]: GPIO 1-wire bus:
[17:01:45][C][gpio.one_wire:017]:   Pin: GPIO5
[17:01:45][W][gpio.one_wire:078]:   Found no devices!
```

changes to yaml for the braking changes:
```diff
--- a/esp32-keller-technik-ac.yaml
+++ b/esp32-keller-technik-ac.yaml
@@ -29,6 +29,7 @@ api:
     key: "bbbbbbbbbbbbbbbbbbbb"
 
 ota:
+  platform: esphome
   password: "aaaaaaaaaaaaaaa"
 
 wifi:
@@ -100,11 +102,12 @@ status_led:
     number: 8
     inverted: true
 
-dallas:
-  - pin: 5
-    update_interval: 10s
+one_wire:
+  - platform: gpio
+    pin: 5
     id: DA1
 
+
 # UART config
 uart:
   tx_pin: 4
@@ -209,23 +212,25 @@ sensor:
     lambda: |-
       return (id(midea_climate).preset == CLIMATE_PRESET_ECO);
 
-  - platform: dallas
+  - platform: dallas_temp
     address: 0x04c3ed7312646128
+    update_interval: 10s
     id: temp_room
     name: "Room"
-    dallas_id: DA1
+    one_wire_id: DA1
     filters:
     - median:
         window_size: 6
         send_every: 1
         send_first_at: 1
 
-  - platform: dallas
+  - platform: dallas_temp
     icon: mdi:snowflake-thermometer
     address: 0xbe9d177312646128
+    update_interval: 10s
     name: "Evaporator"
     id: temp_evaporator
-    dallas_id: DA1
+    one_wire_id: DA1
     filters:
     - median:
         window_size: 6
```

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

esp32 c3 super mini

### Component causing the issue

dallas_temp

### Example YAML snippet

```yaml
one_wire:
  - platform: gpio
    pin: 5
    id: DA1

sensors:
  - platform: dallas_temp
    address: 0x04c3ed7312646128
    update_interval: 10s
    id: temp_room
    name: "Room"
    one_wire_id: DA1
    filters:
    - median:
        window_size: 6
        send_every: 1
        send_first_at: 1

  - platform: dallas_temp
    icon: mdi:snowflake-thermometer
    address: 0xbe9d177312646128
    update_interval: 10s
    name: "Evaporator"
    id: temp_evaporator
    one_wire_id: DA1
    filters:
    - median:
        window_size: 6
        send_every: 1
        send_first_at: 1
```

### Anything in the logs that might be useful for us?

```txt
did work 100% with 2024.5.5 and flashing back via 2024.5.5 solves the problem.
```

Links2004, 2024-06-20:
ok looks like a duplicate to https://github.com/esphome/issues/issues/5908
my search did not bring this up at first :(

closed 2024-06-20 by Links2004

Findarato, 2024-07-10:
Wanted to revive this as I am having the same issues with 2024.6.6

```
[W][gpio.one_wire:078]:   Found no devices!

....

[09:00:37][W][dallas.temp.sensor:139]: 'Temperature 1' - Scratch pad checksum invalid!
[09:00:37][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[09:00:37][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[09:00:38][W][dallas.temp.sensor:139]: 'Temperature 2' - Scratch pad checksum invalid!
[09:00:38][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[09:00:47][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[09:00:47][W][dallas.temp.sensor:139]: 'Temperature 1' - Scratch pad checksum invalid!
[09:00:47][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[09:00:47][W][component:170]: Component dallas_temp.sensor cleared Warning flag
[09:00:48][W][dallas.temp.sensor:139]: 'Temperature 2' - Scratch pad checksum invalid!
[09:00:48][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
```

My Config is below minus some PII
```
substitutions:
  name: fishtank
  friendly_name: fishtank

# external_components:
#   - source:
#       type: git
#       url: https://github.com/ssieb/esphome
#       ref: onewire
#     components: [ gpio, one_wire ]
#     refresh: 1min

interval:
  # - interval: 1s
  #   then:
  #     if:
  #       condition:
  #         wifi.connected:
  #       then:
  #         - output.turn_on: wifi_connected
  #         - delay: 200ms
  #         - output.turn_off: wifi_connected
  #       else:
  #         - output.turn_off: wifi_connected
  - interval: 10s
    then:
      - component.update: temperature1
      - delay: 0.8s
      - component.update: temperature2

packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

wifi:

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret fish_api

ota:
 - platform: esphome
   password: !secret ota_password

web_server:
  css_url: https://esphome.io/_static/webserver-v1.min.css 
  port: 80

switch:
  - platform: restart
    name: 'REBOOT' 

bluetooth_proxy:
  active: true

one_wire:
  - platform: gpio
    pin: GPIO22
    id: dallas_d7

sensor:
  - platform: uptime
    name: Uptime Sensor

  - platform: dallas_temp
    address: 0xe33ce104574a5528
    id: temperature1
    one_wire_id: dallas_d7
    name: "Temperature 1" 
    unit_of_measurement: "°C"
    icon: "mdi:thermometer-plus"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 2
    update_interval: never
    filters:
      - filter_out: nan
      - filter_out: 85.0

  - platform: dallas_temp
    address: 0x803ce1e380f9d328
    id: temperature2
    one_wire_id: dallas_d7
    name: "Temperature 2"
    unit_of_measurement: "°C"
    icon: "mdi:thermometer-plus"
    device_class: "temperature"
    state_class: "measurement"
    accuracy_decimals: 2
    update_interval: never
    filters:
      - filter_out: nan
      - filter_out: 85.0

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
    ssid:
      name: ESP Connected SSID
    bssid:
      name: ESP Connected BSSID
    mac_address:
      name: ESP Mac Wifi Address

concierge-at, 2024-07-29:
hello, now i have done the update to, when i plug out the 3 sensor reboot and plug them in then is all okay until the power fell down,
the opener is writink about hardware break how to identifi this? 
the esp is new, the 3 sensors are new
how kan identify the problem please
thank you