esphome/issues #5937: New update to 1-wire sensors - update interval no longer works

chrisghz, 2024-06-20:
### The problem

I've just updated to the new ESPhome version 2024.6.0 which after check the breaking changes made the adjustments to the 1-wire configuration, but unfortunately the update interval option has been removed or name changed and I can't find an update on any name change. This is causing a significant issue with my 1-wire sensors that I originally had setup as an update interval of 5 seconds, I'm no longer able to set this variable. 

Anyone know of something I've missed?

### Which version of ESPHome has the issue?

2024.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.2

### What platform are you using?

ESP32

### Board

Seeed Studio XIAO ESP32C3

### Component causing the issue

1-wire dallas sensor

emefff, 2024-06-20:
The update_interval option is now below the sensor, before it was below the bus. Like this:

`one_wire:
  - platform: gpio
    pin: 5  
    id: dallas5
    #update_interval: 5s
  
# Individual sensors
sensor:
  - platform: dallas_temp
    one_wire_id: dallas5
    address: 0xblablabla
    name: "Sensor 1"
    accuracy_decimals: 1
    update_interval: 5s`

chrisghz, 2024-06-21:
Amazing thankyou @emefff !! Work perfectly!!!

closed 2024-06-24 by ssieb