esphome/issues #5939: Update to 1-wire sensors in 2024.6.1 - Very unstable on ESP32

little-chef, 2024-06-20:
### The problem

With the update of the one-wire implementation, the ESP32 board nodemcu-32s runs into freezes and reboots about 2 times per minute. I have 7 Dallas temperature sensors, with an update_interval of 2 seconds each, and since the update 2024.6.1, i see the frequent hangups/reboots of the devices, this issue was not existing on the latest ESPHome version before 2024.6.0.

I have configured it like:
```
one_wire:
  - platform: gpio
    pin: GPIO27

sensor:
# the following section 7 times
  - platform: dallas_temp
    address: 0xXXXXXXXXXXXXXXXX
    accuracy_decimals: 1
    id: panel_temperature
    name: "1Wire: Solar Collector Temperature"
    unit_of_measurement: °C
    device_class: temperature
    update_interval: 2s
```

The problem is gone when I fully disable the one-wire sensors. After disabling the board runs stable again, but of course without the sensor data.

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.3

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

esphome

### Example YAML snippet

```yaml
one_wire:
  - platform: gpio
    pin: GPIO27

sensor:
# the following section 7 times
  - platform: dallas_temp
    address: 0xXXXXXXXXXXXXXXXX
    accuracy_decimals: 1
    id: panel_temperature
    name: "1Wire: Solar Collector Temperature"
    unit_of_measurement: °C
    device_class: temperature
    update_interval: 2s
```

### Anything in the logs that might be useful for us?

```txt
No stack traces or useful logs, board just freezes/hangs, and then reboots the board after some time
```

to4ko, 2024-06-20:
same here, 3 esp32 devices with dallas - 2 of them randomly restarts and one hangs till power reset.

flashing back to 2024.5.1 solves issue.

bottlesbo, 2024-06-20:
[st-dr_logs.txt](https://github.com/user-attachments/files/15919384/st-dr_logs.txt)
My log 
Board esp d1 mini v3
config 

```
one_wire:
 - platform: gpio
   pin: GPIO0
    update_interval: 5s
sensor:
  - platform: dallas-temp
    name: "Temperature aquarium"
    address: 0x6408dcd446b80628
    update_interval: 5s
```
The problem is gone when I fully disable the one-wire sensors.

little-chef, 2024-06-20:
Sounds to me like a deadlock in the code after the restructuring of the code, followed by a watchdog reset after that. The last logging seems to confirm that as well.

to4ko, 2024-06-21:
do another one check - once boards downgraded to 2024.5.5 no issues at all.

bedson87, 2024-06-21:
Same for me. With the new 1wire comp. with dallas, ESP hangs after a few seconds or minutes. I don't see anything interesting in the logs. After removing new 1wire with Dallas, ESP works properly.

to4ko, 2024-06-21:
same board, same sensors, diff version

2024.6.1
[image]

2024.5.5
[image]

ssieb, 2024-06-21:
@to4ko you have a different issue.  This should fix it:
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ gpio ]
    refresh: 1min
```

ssieb, 2024-06-21:
The main issue here is that each sensor takes about 1s to update, so things get backed up quickly.  The old code triggered an update on all sensors at once with a global command, but now we're supporting more devices than just temperature.  It might still be possible, but will require some restructuring.  In the meantime, I'll see why it's crashing, since that shouldn't happen anyway.  Need serial logs to see what's happening.

ssieb, 2024-06-21:
@bottlesbo do you have any pwm configured?

to4ko, 2024-06-21:
> @to4ko you have a different issue.  This should fix it:
> 
> ```yaml
> 
> external_components:
> 
>   - source:
> 
>       type: git
> 
>       url: https://github.com/ssieb/esphome
> 
>       ref: onewire
> 
>     components: [ gpio ]
> 
>     refresh: 1min
> 
> ```

Doesnt look like it helps - board restarting over and over again

Rolling back to 2024.5.5 

[image]

ssieb, 2024-06-21:
@to4ko ok, you're showing two different issues.  You showed the devices not being found, but I missed that you were also having restart issues.  Are those on the same device?  Please don't mix issues if those are different devices.  The not found issue should be fixed by what I gave you, since you're also using the (apparently common) gpio13.

to4ko, 2024-06-22:
> @to4ko ok, you're showing two different issues.  You showed the devices not being found, but I missed that you were also having restart issues.  Are those on the same device?  Please don't mix [...]

This is the same device. 
2024.6.1 - sensors nor found
2024.6.1 with your fix - sensors found but board is restarting
2024.5.5 - all ok

ssieb, 2024-06-22:
How many sensors and what update interval?  Can you get serial logs?

ellull, 2024-06-22:
@ssieb I'm having the same issue with just one sensor and an update interval of 5 seconds. But sometimes the ESP32 takes several hours to reset.

And I have a PWM output (you asked this in a previous comment).

I will try to get the serial logs. Do you need any specific log level?

to4ko, 2024-06-22:
> How many sensors and what update interval?  Can you get serial logs?

3 sensors. 1 min update. No logs at the moment - im on vacation with remote access only

ssieb, 2024-06-22:
> The main issue here is that each sensor takes about 1s to update, so things get backed up quickly. The old code triggered an update on all sensors at once with a global command, but now we're [...]

This is actually wrong.  Everything is interleaved.  Of course the sensor update doesn't hold anything up.
I just tested with 3 sensors, but with 3 instances for each sensor, so it's doing 9 updates every second.  There's no indication of a problem.  I'll let it run and see what happens, but I don't see any reason that there would be a problem.  I will need logs.

ssieb, 2024-06-22:
> @ssieb I'm having the same issue with just one sensor and an update interval of 5 seconds. But sometimes the ESP32 takes several hours to reset.
> 
> And I have a PWM output (you asked this in a previous comment).
> 
> I will try to get the serial logs. Do you need any specific log level?

The pwm question was only relevant to an 8266.  The log level of the serial logs doesn't really matter as I'm looking for the crash reason, but debug would probably be good to provide some context.  If it's taking that long, you will need to save the logs somewhere unless you expect to be able to catch it when it happens.

ellull, 2024-06-22:
I've got a serial log: [serial.zip](https://github.com/user-attachments/files/15936235/serial.zip).

I hope this helps you find the cause of the issue.

I think you are mainly interested in:
```
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4008e0f2  PS      : 0x00060735  A0      : 0x80081328  A1      : 0x3ffb2630  
A2      : 0x00000001  A3      : 0x3ffbf0d8  A4      : 0x3ffc430c  A5      : 0x00060323  
A6      : 0x00060323  A7      : 0x00000001  A8      : 0x8008e0f2  A9      : 0x3ffb2610  
A10     : 0x3ffc4348  A11     : 0x0049dbbd  A12     : 0x0049dbbd  A13     : 0x00060723  
A14     : 0x00060723  A15     : 0x00000001  SAR     : 0x00000011  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400849c5  LEND    : 0x400849cd  LCOUNT  : 0x00000027  

Backtrace:0x4008e0ef:0x3ffb26300x40081325:0x3ffb2650 0x400e747e:0x3ffb2670 0x40081341:0x3ffb2690 0x400814db:0x3ffb26b0 0x40081507:0x3ffb26d0 0x40088b89:0x3ffb
26f0 0x400db56a:0x3ffb2710 0x400db705:0x3ffb2740 0x400e7d0f:0x3ffb2780 0x400e57f5:0x3ffb27d0 0x400e7f12:0x3ffb2800 0x400ef7a1:0x3ffb2820 

Core  0 register dump:
PC      : 0x4017098a  PS      : 0x00060135  A0      : 0x800fa110  A1      : 0x3ffbd520  
A2      : 0x00000000  A3      : 0x80000001  A4      : 0x8008ee9c  A5      : 0x3ffbd440  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800f9c8e  A9      : 0x3ffbd4f0  
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x8008ee9c  A13     : 0x3ffbd420  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x40170987:0x3ffbd5200x400fa10d:0x3ffbd540 0x4008d000:0x3ffbd560 
```

In case you need it, this is my current yaml.
<details>
<summary>Current YAML</summary>

```yaml
esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino

logger:
  level: DEBUG

wifi:
  id: wifi_id
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name} Hotspot

api:
  encryption:
    key: !secret api_key
  reboot_timeout: 0s

ota:
  - platform: esphome
    password: !secret ota_password

debug:
  update_interval: 5s

one_wire:
  - platform: gpio
    pin: GPIO17

status_led:
  pin:
    number: GPIO02
    inverted: true

button:
  - platform: restart
    id: reboot
    name: Reboot
    disabled_by_default: true

sensor:
  # Wifi signal sensor.
  - platform: wifi_signal
    id: rssi_sensor
    name: RSSI
    disabled_by_default: true

  - platform: uptime
    name: Uptime

  - platform: dallas_temp
    id: temperature_sensor_internal
    internal: true
    update_interval: 5s
    accuracy_decimals: 2
    filters:
      - sliding_window_moving_average: 
          window_size: 3
          send_every: 1

  - platform: copy
    id: temperature_sensor
    name: Temperature
    source_id: temperature_sensor_internal
    filters:
      - throttle_average: 10s

  - platform: pulse_counter
    id: fan_speed_sensor
    name: Fan Speed
    pin:
      number: GPIO16
      inverted: true
    icon: mdi:fan
    unit_of_measurement: 'RPM'
    accuracy_decimals: 2
    update_interval: 10s
    filters:
      - multiply: 0.5

  - platform: pid
    id: fan_duty_cycle_sensor
    name: Duty Cycle
    type: COOL
    accuracy_decimals: 2
    entity_category: DIAGNOSTIC
    state_class: measurement
    disabled_by_default: true

  - platform: pid
    id: pid_proportional_term
    name: PID Proportional Term
    type: PROPORTIONAL
    accuracy_decimals: 2
    entity_category: DIAGNOSTIC
    disabled_by_default: true
    filters: 
      - multiply: -1.0
          
  - platform: pid
    id: pid_integral_term
    name: PID Integral Term
    type: INTEGRAL
    accuracy_decimals: 2
    entity_category: DIAGNOSTIC
    disabled_by_default: true
    filters: 
      - multiply: -1.0

  - platform: pid
    id: pid_derivative_term
    name: PID Derivative Term
    type: DERIVATIVE
    accuracy_decimals: 2
    entity_category: DIAGNOSTIC
    disabled_by_default: true
    filters: 
      - multiply: -1.0

  - platform: pid
    id: pid_error
    name: PID error
    type: ERROR
    accuracy_decimals: 2
    entity_category: DIAGNOSTIC
    disabled_by_default: true
    filters: 
      - multiply: -1.0

  - platform: debug
    loop_time:
      name: "Loop Time"

binary_sensor:
  # Status sensor.
  - platform: status
    id: status_sensor
    name: Status
    disabled_by_default: true

text_sensor:
  # Version sensor.
  - platform: version
    name: Version
    disabled_by_default: true

  # Wifi info sensor.
  - platform: wifi_info
    ip_address:
      name: IP Address
      disabled_by_default: true
    dns_address:
      name: DNS Address
      disabled_by_default: true
    ssid:
      name: SSID
      disabled_by_default: true
    bssid:
      name: BSSID
      disabled_by_default: true

  - platform: debug
    reset_reason:
      name: "Reset Reason"      

output:
  # Fan PWM output.
  - platform: ledc
    id: fan_duty_cycle
    pin: GPIO27
    frequency: 25000Hz
    min_power: 4%
    zero_means_zero: true

climate:
  - platform: pid
    id: thermostat
    name: Thermostat
    default_target_temperature: 30.0°C
    visual:
      min_temperature: 20.0°C
      max_temperature: 40.0°C
      temperature_step:
        current_temperature: 0.01
        target_temperature: 0.5
    sensor: temperature_sensor_internal
    cool_output: fan_duty_cycle
    control_parameters:
      kp: 0.5
      ki: 0.0005
      kd: 0
      max_integral: 0.0
      starting_integral_term: -0.20
    deadband_parameters:
      threshold_high: 0.15°C
      threshold_low: -0.15°C
      kp_multiplier: 0.0
      ki_multiplier: 0.1
      kd_multiplier: 0.0
      deadband_output_averaging_samples: 15      
```
</details>

little-chef, 2024-06-22:
> How many sensors and what update interval? Can you get serial logs?

I indicated it in my ticket: 7 sensors, 2 second update interval each. It reboots every 30 seconds or so. I tried putting the sensors on 50, 52, 54, 56, 58, 60, 62 seconds to at least make sure they do not get updated all at the same time, but that did not solve it. I do not have serial logs, I can only access the device remotely.

HpNoTiQ56, 2024-06-22:
Not working here also esp with i2c + onewire not working.
Switch to dallasng working.

[image]

`# Sonde Dallas
one_wire:
  - platform: gpio
    pin: GPIO32
sensor:
  - platform: dallas_temp
    name: "T° Eau Piscine"
    id: temp_piscine
    address: 0x0c236c761f64ff28
    accuracy_decimals: 1
    update_interval: 5s
`

bakerkj, 2024-06-22:
I am also seeing this as well on multiple devices.

jadenv, 2024-06-23:
After updating to v2024.6.0, my ESP32 is stuck in a boot loop indefinitely after initializing the 1-wire bus.

I have 4 `dallas_temp` sensors that used to work just fine prior to the update.

Logs:
```
[C][safe_mode:079]: There have been 9 suspected unsuccessful boot attempts
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][i2c.idf:017]: Setting up I2C bus...
[I][i2c.idf:252]: Performing I2C bus recovery
[D][esp-idf:000]: I (79) gpio: GPIO[22]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 

[D][esp-idf:000]: I (85) gpio: GPIO[23]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 

[C][gpio.one_wire:011]: Setting up 1-wire bus...Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 
[10:32:00]
[10:32:00]Core  1 register dump:
[10:32:00]PC      : 0x4008cae9  PS      : 0x00060234  A0      : 0x800824b2  A1      : 0x3ffbf390  
[10:32:00]A2      : 0x3ffb3f48  A3      : 0x3ffb3f48  A4      : 0x00000000  A5      : 0x00000001  
[10:32:00]A6      : 0x00000001  A7      : 0x00000000  A8      : 0x8008cae9  A9      : 0x3ffbf380  
[10:32:00]A10     : 0x00000003  A11     : 0x00060223  A12     : 0x00060223  A13     : 0x00060223  
[10:32:00]A14     : 0x3f40564c  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000006  
[10:32:00]EXCVADDR: 0x00000000  LBEG    : 0x40083189  LEND    : 0x40083191  LCOUNT  : 0x00000027  
[10:32:00]
[10:32:00]
[10:32:00]Backtrace: 0x4008cae6:0x3ffbf390 0x400824af:0x3ffbf3b0 0x400824d1:0x3ffbf3d0 0x4008bf25:0x3ffbf3f0 0x40081221:0x3ffbf410 0x400e5ca6:0x3ffbf430 0x4008123d:0x3ffbf450 0x400813d7:0x3ffbf470 0x40081474:0x3ffbf490 0x400ddc66:0x3ffbf4c0 0x400dbef5:0x3ffbf510 0x400dbf02:0x3ffbf530 0x40161f09:0x3ffbf550 0x40161f95:0x3ffbf570 0x400e5148:0x3ffbf590 0x400e784d:0x3ffbf5c0 0x400dada3:0x3ffbf6c0
[10:32:00]
[10:32:00]
[10:32:00]Core  0 register dump:
[10:32:00]PC      : 0x40162bd6  PS      : 0x00060334  A0      : 0x800feb71  A1      : 0x3ffba410  
[10:32:00]A2      : 0x00000000  A3      : 0x40001d48  A4      : 0x00060120  A5      : 0x00060123  
[10:32:00]A6      : 0x00000003  A7      : 0x00060023  A8      : 0x8008ff68  A9      : 0x3ffba3f0  
[10:32:00]A10     : 0x3ffb9d78  A11     : 0x3ffb0f74  A12     : 0x3ffb0f74  A13     : 0x00000000  
[10:32:00]A14     : 0x00060120  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000006  
[10:32:00]EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
[10:32:00]
[10:32:00]
[10:32:00]Backtrace: 0x40162bd3:0x3ffba410 0x400feb6e:0x3ffba430 0x4008aec8:0x3ffba450
[10:32:00]
[10:32:00]
[10:32:00]
[10:32:00]
[10:32:00]ELF file SHA256: fe53b16e5af49441
[10:32:00]
[10:32:00]Rebooting...
```

ssieb, 2024-06-24:
Try my external component again.  I think I found the issue.

jadenv, 2024-06-24:
After flashing directly via USB with the external component (was not able to flash remotely via wifi due to infinite boot loop), the ESP32 does get past the 1-wire panic, boot again, and report correct temperatures.

Will keep an eye on it to see if it remains stable through the night, thanks @ssieb 🎉

MiguelAngelLV, 2024-06-24:
> @to4ko you have a different issue. This should fix it:
> 
> ```yaml
> external_components:
>   - source:
>       type: git
>       url: https://github.com/ssieb/esphome
>       ref: onewire
>     components: [ gpio ]
>     refresh: 1min
> ```

This solve my problem. Thanks you

closed 2024-06-24 by kbx81

little-chef, 2024-06-24:
> does get past the 1-wire panic, boot again, and report correct temperatures.
> 
> Will keep an eye on it to see if it remains stable through the ni

This looks like it solved it. It is now running for 30 minutes or so. 
Thanks.

bakerkj, 2024-06-25:
> Try my external component again. I think I found the issue.

This has fixed the issues for me. My two devices have been running for more than 10 hours without a reboot.

Thanks!