esphome/issues #5943: mdns wifi on start disabled error

iqubik, 2024-06-20:
### The problem

### Problem Description

When I use the following configuration in my ESPHome YAML:

YAML

wifi:
  enable_on_boot: False

And then manually connect to the network in STA mode using the code:

WiFi.disconnect();
WiFi.mode(WIFI_STA);
WiFi.begin("$staNameSTR", "$staPassSTR");

The device name, instead of being the one specified in the esphome section:

YAML

esphome:
  name: electrowaterpool

changes to esp32-54391C.local. This causes an ESPHome API error and a connection error to the smart home system. I believe this issue is due to mDNS and related services not being configured correctly.
- [api.connection:078] | : Network unavailable, disconnecting

### Steps to Reproduce

1. Use the following configuration in ESPHome YAML:
    
YAML

    wifi:
      enable_on_boot: False
    esphome:
      name: electrowaterpool
    

2. Manually connect to Wi-Fi in STA mode with the following code:
    
C++

    WiFi.disconnect();
    WiFi.mode(WIFI_STA);
    WiFi.begin("$staNameSTR", "$staPassSTR");
    

3. Observe that the device name changes to esp32-54391C.local instead of electrowaterpool.local.

### Expected Behavior

The device should retain the name specified in the esphome section (electrowaterpool.local).

### Additional Information

It seems that mDNS and related services are not being configured correctly when manually connecting to Wi-Fi in STA mode. How can this be fixed?

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

all versions

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

mdns

### Example YAML snippet

```yaml
esphome:
  name: electrowaterpool

wifi:
  enable_on_boot: False
....
on_click:
  then:...
    WiFi.disconnect();
    WiFi.mode(WIFI_STA);
    WiFi.begin("$staNameSTR", "$staPassSTR");
```

### Anything in the logs that might be useful for us?

```txt
- [api.connection:078] | : Network unavailable, disconnecting
```

robin-thoni, 2024-07-11:
I have a similar problem. I disabled Wi-Fi on boot for a battery operated device. My understanding is that mDNS will fail to init if there's no network (Wi-Fi or Ethernet).
The work around I have is to re-setup mDNS after enabling Wi-Fi:
```
mdns:
  id: mdns_id

wifi:
  id: wifi_id
  enable_on_boot: false

binary_sensor:
  - id: my_sensor
    on_release:
      then:
        - wifi.enable:
        - lambda: id(mdns_id).setup();
```

Sounds to work for now

iqubik, 2024-07-11:
> I have a similar problem. I disabled Wi-Fi on boot for a battery operated device. My understanding is that mDNS will fail to init if there's no network (Wi-Fi or Ethernet). The work around I have is [...]

not work for me
-------
<html>
<body>

23:48:04 | [W] | [api.connection:078] | : Network unavailable, disconnecting
-- | -- | -- | --
23:48:05 | [W] | [api.connection:078] | : Network unavailable, disconnecting
23:48:06 | [W] | [api.connection:078] | : Network unavailable, disconnecting
23:48:27 | [W] | [mdns:047] | Failed to register mDNS service _esphomelib: ESP_ERR_INVALID_ARG
23:48:27 | [W] | [mdns:047] | Failed to register mDNS service _http: ESP_ERR_INVALID_ARG
23:48:27 | [W] | [api.connection:078] | : Network unavailable, disconnecting
23:48:28 | [W] | [api.connection:078] | : Network unavailable, disconnecting
23:48:29 | [W] | [api.connection:078] | : Network unavailable, disconnecting
23:48:30 | [W] | [api.connection:078] | : Network unavailable, disconnecting

</body>
</html>

iqubik, 2024-07-16:
why this closed? Please add mdns.restart in main esphome code and all be good

sorrypqa, 2024-11-06:
> - lambda: id(mdns_id).setup();

thank you! Your solution has solved my problem