esphome/issues #5944: Regression with AR-01

madbrain76, 2024-06-20:
### The problem

I have 2 AR-01s, and cannot update them with the latest ESPHome version

### Which version of ESPHome has the issue?

2024.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.3

### What platform are you using?

ESP8266

### Board

esp825 - AtHom AR-01

### Component causing the issue

Unknown

### Example YAML snippet

```yaml
substitutions:
  name: "guestroom-infrared"
  friendly_name: "Guestroom infrared"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"

esp8266:
  board: esp8285
  restore_from_flash: true

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "<removed>"

ota:
  password: "<removed>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  #use_address: 192.168.100.151
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Guestroom-Infrared"
    password: "<removed>"

captive_portal:

web_server:
  port: 80

sensor:
  - platform: wifi_signal
    name: "${friendly_name} WiFi Signal"
    update_interval: 60s

  - platform: uptime
    name: "${friendly_name} Uptime"

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} IP Address"
      disabled_by_default: true
    ssid:
      name: "${friendly_name} SSID"
    bssid:
      name: "${friendly_name} BSSID"
    mac_address:
      name: "${friendly_name} Mac Wifi Address"
    scan_results:
      name: "${friendly_name} Latest Scan Results"

switch:
  - platform: restart
    name: "${friendly_name} restart"

binary_sensor:
  - platform: status
    name: "${friendly_name} IR Status"

  - platform: gpio
    pin: GPIO0
    id: physical_button

  - platform: remote_receiver
    name: "Lights ceiling off"
    lg:
      data: 0x08E76897
      nbits: 32

  - platform: remote_receiver
    name: "Lights ceiling on"
    lg:
      data: 0x08E7708F
      nbits: 32

  - platform: remote_receiver
    name: "Lights sides 1 off"
    lg:
      data: 0x08E750AF
      nbits: 32

  - platform: remote_receiver
    name: "Lights sides 1 on"
    lg:
      data: 0x08E7A857
      nbits: 32

  - platform: remote_receiver
    name: "Lights sides 2 off"
    lg:
      data: 0x08E7906F
      nbits: 32

  - platform: remote_receiver
    name: "Lights sides 2 on"
    lg:
      data: 0x08E728D7
      nbits: 32

  - platform: remote_receiver
    name: "Lights all off"
    lg:
      data: 0x08E710EF
      nbits: 32

  - platform: remote_receiver
    name: "Lights all on"
    lg:
      data: 0x08E730CF
      nbits: 32

status_led:
  pin: GPIO13

remote_transmitter:
  pin:
    number: GPIO4
  carrier_duty_percent: 50%

remote_receiver:
  pin:
    number: GPIO5
    inverted: true
  dump: all
```

### Anything in the logs that might be useful for us?

```txt
1) at first

INFO ESPHome 2024.6.0
INFO Reading configuration /config/esphome/ht-infrared.yaml...
Failed config
```

At least one platform must be specified for 'ota'; add 'platform: esphome' for original OTA functionality

2) after I added the platform

```
ota
  platform: esphome
  key: <removed>
```

I got this error :

```
INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/guestroom-infrared.yaml...
ERROR Error while reading config: Invalid YAML syntax:

while parsing a block collection
  in "/config/esphome/guestroom-infrared.yaml", line 22, column 3
expected <block end>, but found '?'
  in "/config/esphome/guestroom-infrared.yaml", line 23, column 3
```

3) after I commented out the key

```
ota
  platform: esphome
#  key: <removed>
```

I get this :

```
INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/guestroom-infrared.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing guestroom-infrared (board: esp8285; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
RAM:   [=====     ]  45.8% (used 37560 bytes from 81920 bytes)
Flash: [======    ]  55.7% (used 570213 bytes from 1023984 bytes)
========================= [SUCCESS] Took 1.39 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.100.151
INFO Uploading /data/build/guestroom-infrared/.pioenvs/guestroom-infrared/firmware.bin (574368 bytes)
INFO Compressed to 393100 bytes
ERROR ESP requests password, but no password given!
```

I'm not sure where to go from here.
```

mr2toyou, 2024-06-20:
Having the same ota issue.   Following

mkosmo, 2024-06-21:
I'm using the following, per [the doc](https://next.esphome.io/components/ota):

```
ota:
  - platform: esphome
    password: !secret ota_password
```

And getting the same error "ERROR ESP requests password, but no password given!"

madbrain76, 2024-06-21:
@mkosmo,
My 2 AR-01 devices were using key: previously.

Interestingly, I have a 3rd AR-01 device which wasn't using key at all. I was able to migrate that one by adding "  -platform: esphome" . I don't know why that device is different.

madbrain76, 2024-06-21:
Looks like it's just a silly indentation problem on my side.

Using the syntax you provided, with a 2 extra spaces, it's building correctly on my device right now.
I believe it got well past the point of "no password given" .  I'll edit this post again when they are both complete.

I still wonder why the 3rd one doesn't have a password :)

mkosmo, 2024-06-21:
I fixed mine.  I reread the documentation again, focusing on the changelog instead and realized I was creating an object it didn't expect.  This worked:

```
ota:
  platform: esphome
  password: !secret ota_password
```

It seems the documentation is inconsistent right now.