esphome/issues #5945: error: 'class esphome::http_request::HttpRequestArduino' has no member named 'get_string'

DaveOke, 2024-06-21:
### The problem

Lambda @ https://esphome.io/cookbook/http_request_sensor.html fails with: error: 'class esphome::http_request::HttpRequestArduino' has no member named 'get_string'

Happned after upgrading to 2024.6.1

Actual entry in yaml:
interval:
    - interval: 60s
      then:
        - http_request.get:
            url: http://10.0.1.8:1880/api/humidity
            on_response:
              then:
                - lambda: |-
                    json::parse_json(id(http_request_id).get_string(), [](JsonObject root) {
                        id(humidity).publish_state(root["value"]);
                    });      

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2025.6.3

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

lambda

### Example YAML snippet

```yaml
http_request:
  useragent: esphome/device
  id: http_request_id
  verify_ssl: false

interval:
    - interval: 60s
      then:
        - http_request.get:
            url: http://10.0.1.8:1880/api/humidity
            on_response:
              then:
                - lambda: |-
                    json::parse_json(id(http_request_id).get_string(), [](JsonObject root) {
                        id(humidity).publish_state(root["value"]);
                    });
```

### Anything in the logs that might be useful for us?

```txt
/config/esphome/Co2.yaml: In lambda function:
/config/esphome/Co2.yaml:57:41: error: 'class esphome::http_request::HttpRequestArduino' has no member named 'get_string'
   57 |                     json::parse_json(id(http_request_id).get_string(), [](JsonObject root) {
      |                                         ^~~~~~~~~~
```

pksoft585, 2024-06-21:
Explanation is in documentation there: [https://esphome.io/components/http_request.html#get-values-from-a-json-body-response](url):

```yaml
- http_request.get:
    url: http://10.0.1.8:1880/api/humidity
    capture_response: true
    on_response:
      then:
        - lambda: |-
            json::parse_json(body, [](JsonObject root) -> bool {
                id(humidity).publish_state(root["value"]);
                return true;
            });```

andrewjswan, 2024-06-22:
> ```yaml
> capture_response: true
> ```

In ESP IDF i get:
```
HTTP_CLIENT: Body received in fetch header state
```
in Arduino i get or `empty body` or stuck on get

pksoft585, 2024-06-22:
I'm using Arduino framework without problem - but on ESP32 (WT32-ETH01):

```yaml
esp32:
  board: esp-wrover-kit
  framework:
    type: arduino
...

http_request:
  useragent: esphome/device
  timeout: 10s
  id: http_req
  follow_redirects: False
  redirect_limit: 1
  verify_ssl: false

...

              - http_request.get:
                  headers:
                    content-Type: application/json
                  url: !secret url_ble
                  capture_response: true
                  on_response:
                    - if:
                        condition:
                          lambda: |-
                            return response->status_code == 200;
                        then:
                          - logger.log: "Get value from BLE temperature..."
                          - lambda: |-
                              json::parse_json(body, [](JsonObject root) -> bool {
                                  id(ble_temperature).publish_state(root["value"]);
                                  return true;
                              });
                        else:
                          - logger.log: "Get value from BLE temperature - error!!!"

...
```

KirkKirk, 2024-06-23:
Have a similar situation and couldn't figure it out. 
Can someone please help me?

worked fine like this: 
```
http_request:
  useragent: esphome/device
  id: http_request_id
  verify_ssl: false
  timeout: 10s

captive_portal:

interval:
  - interval: 10s
    then:
      - http_request.get: 
          url: http://192.168.1.158/sensor/pool_temp
          capture_response: true
          on_response:
            then:
              - lambda: |-
                  json::parse_json(id(http_request_id).get_string(), [](JsonObject root) {
                      id(humidity).publish_state(root["value"]);
                      return true;
                  });
```

but doesn't work like this: 

```
http_request:
  useragent: esphome/device
  id: http_request_id
  verify_ssl: false
  timeout: 10s

captive_portal:

interval:
  - interval: 10s
    then:
      - http_request.get: 
          url: http://192.168.1.158/sensor/pool_temp
          capture_response: true
          on_response:
            then:
              - lambda: |-
                  json::parse_json(body, [](JsonObject root) -> bool {
                      id(pool_temperature_remote).publish_state(root["value"]);
                      return true;
                  });
```

after installation, I'm getting these errors:
```
[21:54:44][W][http_request.arduino:058]: Using HTTP on Arduino version >= 3.1 is **very** slow. Consider setting framework version to 3.0.2 in your YAML, or use HTTPS
[21:54:44][E][http_request.arduino:112]: HTTP Request failed; URL: http://192.168.1.158/sensor/pool_temp; Code: 404
[21:54:44][E][component:164]: Component http_request set Error flag: unspecified
[21:54:45][D][sensor:093]: 'Compensation': Sending state nan  with 5 decimals of accuracy
[21:54:45][E][component:176]: Component http_request cleared Error flag
```

andrewjswan, 2024-06-23:
capture_response: false - All Ok
capture_response: true - [15:39:20][D][http_request.arduino:119]: Content-Length: -1 and Esp32 stuck

andrewjswan, 2024-06-24:
Body not received
```
      - http_request.get:
          url: !lambda |-
            return ((std::string) "https://api.svitlobot.in.ua/channelPing?channel_key=AABBCCDD").c_str();
          headers:
            Accept: "*/*"
            Accept-Language: "en"
            Cache-control: "no-cache"
            Content-Type: "text/plain"
          capture_response: true
          on_response:
            then:
              - lambda: |-
                  ESP_LOGD("SvitloBot", "Response: %d (%ums): [%d]: %s", response->status_code, response->duration_ms, body.length(), body.c_str());
```

IDF return:
```
[20:02:50][D][esp-idf:000]: I (54061) HTTP_CLIENT: Body received in fetch header state, 0x3ffcf341, 32
[20:02:50][D][SvitloBot:093]: Response: 200 (1145ms): [0]: 
```

pksoft585, 2024-06-24:
And it can't be a problem that the response from the server is text/plain? From the service description it looks like the response should be json... (speculation).

andrewjswan, 2024-06-24:
> And it can't be a problem that the response from the server is text/plain?

The result is an ordinary line, but no matter what header you specify there, or not specify it at all, nothing changes.

andrewjswan, 2024-06-25:
If server not return Content Length, or Content Length  = -1 Then we try to allocate memory with size -1

https://github.com/esphome/esphome/blob/fb9844463b53d3aee6fd5e67bd5f71a7568c6a98/esphome/components/http_request/http_request.h#L136-L141

It seems to me that it is at this moment that ESP freezes.

andrewjswan, 2024-06-25:
Also, when reading, if Content Length is negative, we get a negative buffer length.

https://github.com/esphome/esphome/blob/fb9844463b53d3aee6fd5e67bd5f71a7568c6a98/esphome/components/http_request/http_request_arduino.cpp#L137-L145

andrewjswan, 2024-06-25:
2024.6.2 Everything is the same. But the 2024.5.5 option with `get_string` worked without problems and returned the result.

andrewjswan, 2024-06-25:
This request https://github.com/esphome/issues/issues/5945#issuecomment-2187027867 return:
- Arduino: Content-Length: -1 and Esp freeze
- ESP-IDF: Content-Length: 0 and no body
```
curl.exe -v -k https://api.svitlobot.in.ua/channelPing?channel_key=AABBCCDD                                                                                                     *   Trying 104.21.73.54:443...
* Connected to api.svitlobot.in.ua (104.21.73.54) port 443 (#0)                                                                                                                                                    * ALPN, offering h2                                                                                                                                                                                           21:29* ALPN, offering http/1.1
* error setting certificate verify locations, continuing anyway:
*  CAfile: /tank/d$/CMD/Scheduler/Utils/ssl/certs/ca-bundle.crt
*  CApath: none
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: CN=svitlobot.in.ua
*  start date: Jun  1 08:53:59 2024 GMT
*  expire date: Aug 30 08:53:58 2024 GMT
*  issuer: C=US; O=Google Trust Services LLC; CN=GTS CA 1P5
*  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* Using Stream ID: 1 (easy handle 0x1b6369684b0)
> GET /channelPing?channel_key=AABBCCDD HTTP/2
> Host: api.svitlobot.in.ua
> user-agent: curl/7.73.0
> accept: */*
>
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* old SSL session ID is stale, removing                                                                                                                                                                            * Connection state changed (MAX_CONCURRENT_STREAMS == 100)!                                                                                                                                                        < HTTP/2 200                                                                                                                                                                                                       < date: Tue, 25 Jun 2024 18:28:37 GMT                                                                                                                                                                              < content-type: text/html; charset=utf-8                                                                                                                                                                           < access-control-allow-origin: *                                                                                                                                                                                   < cf-cache-status: DYNAMIC                                                                                                                                                                                         < report-to: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v4?s=DU0YrskCR5HU3bL9l0Hsj%2F9OQHXR4ozg2rHUUS722Px7jTu3tWFe%2BkvFsKlynq8yChSArgP%2BYedlVG4juRgRsS7USWHfblgusqBF9XnApIlwJDKW7yEhjgyCTd5nVq2sx%2FvRHuaA"}],"group":"cf-nel","max_age":604800}                                                                                                                                                                < nel: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}                                                                                                                                                < server: cloudflare                                                                                                                                                                                               < cf-ray: 899704b49db73625-FRA                                                                                                                                                                                     < alt-svc: h3=":443"; ma=86400                                                                                                                                                                                     <                                                                                                                                                                                                                  Channel "AABBCCDD" is not found! Insert correct key from chat bot!* Connection #0 to host api.svitlobot.in.ua left intact 
```

jesserockz, 2024-06-26:
One thing that was overlooked with the rewrite was servers sending `chunked` responses with no known `Content-Length` header. This will need more code added to handle these kind of responses.

closed 2024-06-26 by jesserockz

andrewjswan, 2024-06-26:
> I am closing this issue as `get_string` was removed with the rewrite and it is a separate that the chunked responses need to be handled.

Should I open a new Issue? Or is the current one enough and should I wait for a FIX in 2024.6.3?

andrewjswan, 2024-06-26:
https://github.com/esphome/esphome/pull/6953

ekaftan, 2024-09-13:
I spent a couple of hours troubleshooting this problem. I got an empty body from a http_request.post.

Server was not sending Content-Length and thus response body was empty.

I fixed the server side to include Content-Length and it now works.

I got his two lines on the log console:

HTTP_CLIENT: Body received in fetch header state, 0x3fcc8a4b, 31
Could not allocate memory for JSON document! Requested 0 bytes

HLFCode, 2024-11-23:
> One thing that was overlooked with the rewrite was servers sending `chunked` responses with no known `Content-Length` header. This will need more code added to handle these kind of responses.
> 

Please see https://github.com/esphome/issues/issues/5949#issuecomment-2495513699