esphome/issues #5947: ESP32 resseting each 3-20 minutes

ProComCZ, 2024-06-21:
### The problem

ESP32 is resetting each 3-20 minues with reson - "esphome Timer Group 1 Watch Dog Reset Digital Core" or "Software Reset CPU" after upgrade to 2024.6.1 . 2024.5.5 was OK

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.3

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
substitutions:
  friendly_name: Console Fan

esphome:
  name: console-fan

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXXXXXXXX"

ota:
  platform: esphome
  password: "XXXXXXXXXXXXX"

time:
  - platform: homeassistant
    id: my_time

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Console-Fan Fallback Hotspot"
    password: "XXXXXXXXX"

# DEBUG
captive_portal:
debug:
  update_interval: 20s   
# DEBUG
    
    # Enable webserver
web_server:
  port: 80

# UART pro PZEM
uart:
  rx_pin: GPIO22
  tx_pin: GPIO23
  baud_rate: 9600
  
# čidla DALLAS config GPIO
#dallas:
#    pin: GPIO32
#    update_interval: 30s

one_wire:
  - platform: gpio
    pin: GPIO32
#    update_interval: 30s
    
text_sensor:

  # Send IP Address
  - platform: wifi_info
    ip_address:
      name: $friendly_name IP Address
    
  # Send Uptime in raw seconds
  - platform: template
    name: $friendly_name Uptime
    id: uptime_human
    icon: mdi:clock-start
    
  # Senzor z HA který mi do ESP posílá informaci o aktuálním přebytečném výkonu
  - platform: homeassistant
    name: "Spirála 1 (W)"
    entity_id: input_number.power_spirala_l1
    id: power_spirala_L1
    
  - platform: homeassistant
    name: "Spirála 2 (W)"
    entity_id: input_number.power_spirala_l2
    id: power_spirala_L2
    
  - platform: homeassistant
    name: "Spirála 3 (W)"
    entity_id: input_number.power_spirala_l3
    id: power_spirala_L3
  
  - platform: debug
    reset_reason:
      name: "Fan Reset Reason"
    
sensor:    
  - platform: dallas_temp
    address: 0x4805169207f8ff28
    name: "bojler_1_up"
    id: bojler_1_up
    on_value:
      if:
        condition:
          not:
            api.connected:
        then:
          - output.turn_off: SSR_relay1
          - output.turn_off: SSR_relay2
          - output.turn_off: SSR_relay3
          - logger.log: "RELÉ VYPNUTA Z DŮVODU ZTRÁTY SPOJENÍ S HA"
    filters:
      - filter_out: NAN
          
  - platform: dallas_temp
    address: 0x2300000838003828
    name: "bojler_2_down"
    id: bojler_2_down
    filters:
      - filter_out: NAN
  - platform: dallas_temp
    address: 0x49031689c744ff28
    name: "bojler_3_tuv"
    id: bojler_3_tuv
    filters:
      - filter_out: NAN
  - platform: dallas_temp
    address: 0xcd03168a0d5cff28
    name: "bojler_4_vystup_tepla"
    id: bojler_4_vystup_tepla 
    filters:
      - filter_out: NAN
  - platform: dallas_temp
    address: 0x290416928ebeff28
    name: "rozvadec_SSR_temp"
    id: rozvadec_SSR_temp 
    filters:
      - filter_out: NAN
  - platform: dallas_temp
    address: 0xc101204e90190528
    name: "garaz_temp"
    id: garaz_temp
    filters:
      - filter_out: NAN
  - platform: dallas_temp
    address: 0x2601204e88196528
    name: "bojler_kryt_temp"
    id: bojler_kryt_temp 
    filters:
      - filter_out: NAN
 
    
    
  - platform: pzemac
    power:
      name: "Spirala1"
      id: spirala1
    energy:
      name: "Energy1"
    voltage:
      name: "Voltage 1"
    address: 0x1
    update_interval: 20s
    
  - platform: pzemac
    power:
      name: "Spirala2"
      id: spirala2
    energy:
      name: "Energy2"
    voltage:
      name: "Voltage 2"    
    address: 0x2
    update_interval: 20s
    
  - platform: pzemac
    power:
      name: "Spirala3"
      id: spirala3
    energy:
      name: "Energy3"
    voltage:
      name: "Voltage 3"
    address: 0x3
    update_interval: 20s
    
  # Send WiFi signal strength & uptime to HA   
  - platform: wifi_signal
    name: $friendly_name WiFi Strength
    update_interval: 61s
    
  # This is a bit of overkill. It sends a human readable 
  # uptime string
  # 1h 41m 32s instead of 6092 seconds
  - platform: uptime
    name: $friendly_name Uptime
    id: uptime_sensor
    update_interval: 61s
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            # Custom C++ code to generate the result
            state: !lambda |-
              int seconds = round(id(uptime_sensor).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? to_string(days) + "d " : "") +
                (hours ? to_string(hours) + "h " : "") +
                (minutes ? to_string(minutes) + "m " : "") +
                (to_string(seconds) + "s")
              ).c_str();
# DEBUG
  - platform: debug
    free:
      name: "Heap Free"
    block:
      name: "Heap Max Block"
    loop_time:
      name: "Loop Time"              
# DEBUG

output:
  # ledc is the name of the pwm output system on an esp32
  - platform: ledc
    id: SSR_relay1
    pin: GPIO25
    frequency: "1220 Hz" 
    min_power: 0%
    max_power: 100%
    
  - platform: ledc
    id: SSR_relay2
    pin: GPIO26 
    frequency: "1220 Hz" 
    min_power: 0%
    max_power: 100%
    
  - platform: ledc
    id: SSR_relay3
    pin: GPIO27 
    frequency: "1220 Hz" 
    min_power: 0%
    max_power: 100%

# Good for debugging, you can manually set the fan speed.
fan:
  - platform: speed
    output: SSR_relay1
    name: "SSR_relay1"
    on_turn_on:
    - logger.log: "SSR_relay1 is ON!"
    on_turn_off:
    - logger.log: "SSR_relay1 is OFF!"
    
  - platform: speed
    output: SSR_relay2
    name: "SSR_relay2"
    on_turn_on:
    - logger.log: "SSR_relay1 is ON!"
    on_turn_off:
    - logger.log: "SSR_relay1 is OFF!"

  - platform: speed
    output: SSR_relay3
    name: "SSR_relay3"
    on_turn_on:
    - logger.log: "SSR_relay1 is ON!"
    on_turn_off:
    - logger.log: "SSR_relay1 is OFF!"

switch:
  # Expose an ESP32 restart button to HA
  - platform: restart
    name: "Console Fan ESP32 Restart"

  - platform: gpio
    pin: GPIO33
    name: "GoodWeFan"
    id: GoodWeFan
    
script:
  - id: connection_watchdog
    mode: queued
    then:
     if:
      condition:
        not:
          api.connected:
      then:
      - logger.log: "WATCHDOG PRUUUUUUUUUUUUUSERRRRRRRRRRRRRR"
      - delay: 60s
      - script.execute: connection_watchdog
```

emefff, 2024-06-21:
Same here, seems to be the reason for https://github.com/esphome/issues/issues/5940 also....

betaphi, 2024-06-21:
Same exact thing. Seems to be related to the new `one_wire`/`dallas_temp` implementation.

Dominator86, 2024-06-21:
I'm certain that one_wire/dallas_temp is the culprit since switching to dallasng fixed the resets for me.
Board is az-delivery-devkit-v4 and framework is esp-idf if it matters.

lucasimons, 2024-06-21:
> Same here, seems to be the reason for #5940 also....
> 
> other common stuff: one_wire dallas_temp on esp32dev.

I have 3 esp32 dev UNO boards (linke in the pic) , i have 10 dallas sensor for each board all are unstable yesterday they are resetting now are unavailaible.
[image]

Miilani, 2024-06-21:
Same issue here, I removed the "one wire" lines from my configuration and it stopped restarting

Makin-Things, 2024-06-22:
Same issue for me. @ssieb  have you seen this issue?

ellull, 2024-06-22:
I think this is a duplicate of #5939.

ssieb, 2024-06-22:
I have seen this one and the other one.  However, I don't see any way this would happen and I can't reproduce it myself.  I haven't seen any serial logs of it happening either.

Danit2, 2024-06-22:
Same here. 
only the esp‘s with the new one_wire component have this problem. 
my esp‘s ar all restarting every 2 to 30 minutes. 
[image]

georgea62, 2024-06-22:
After upgraded to 2024.6.0 & 1, my Dallas device won't detected.
Going Bach to version 2024.5.x everything is fine.

I tried the following and started working again.

```
external_components:  
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ gpio ]
    refresh: 1min
```

but later I found is reseting every few minutes.
here is part of the serial log:

`
[11:35:31][D][dallas.temp.sensor:054]: 'Temperature': Got Temperature=22.8°C
[11:35:31][D][sensor:094]: 'Temperature': Sending state 22.75000 °C with 1 decimals of accuracy
[11:35:32][D][sensor:094]: 'WiFi Signal dB': Sending state -39.00000 dBm with 0 decimals of accuracy
[11:35:32][D][sensor:094]: 'WiFi Signal Percent': Sending state 100.00000 % with 0 decimals of accuracy
[11:35:32][D][sensor:094]: 'Uptime Sensor': Sending state 414.12500 s with 0 decimals of accuracy
[11:35:34][D][sensor:094]: 'Random': Sending state 300.00000  with 1 decimals of accuracy
[11:36:01]Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 
[11:36:01]
[11:36:01]Core  1 register dump:
[11:36:01]PC      : 0x40097470  PS      : 0x00060735  A0      : 0x80096892  A1      : 0x3ffcde90  
[11:36:01]A2      : 0x3ffbf0c8  A3      : 0xffffffff  A4      : 0x0006c36b  A5      : 0x00060723  
[11:36:01]A6      : 0x00060723  A7      : 0x00000001  A8      : 0x8009676e  A9      : 0xffffffff  
[11:36:01]A10     : 0x00000003  A11     : 0x00060723  A12     : 0x00060723  A13     : 0x00006b00  
[11:36:01]A14     : 0x007bf0c8  A15     : 0x003fffff  SAR     : 0x0000000e  EXCCAUSE: 0x00000006  
[11:36:01]EXCVADDR: 0x00000000  LBEG    : 0x40084a8d  LEND    : 0x40084a95  LCOUNT  : 0x00000027  
[11:36:01]
[11:36:01]
[11:36:01]Backtrace:0x4009746d:0x3ffcde900x4009688f:0x3ffcded0 0x40081421:0x3ffcdef0 0x400eeb56:0x3ffcdf10 0x4008143d:0x3ffcdf30 0x400815d7:0x3ffcdf50 0x40081603:0x3ffcdf70 0x400901a9:0x3ffcdf90 0x400d9fc2:0x3ffcdfb0 0x400da15d:0x3ffcdfe0 0x400ef617:0x3ffce020 0x400ece3d:0x3ffce070 0x400efb3e:0x3ffce0a0 0x400fc035:0x3ffce0c0 
[11:36:01]
[11:36:01]
[11:36:01]Core  0 register dump:
[11:36:01]PC      : 0x4020fd22  PS      : 0x00060235  A0      : 0x80106800  A1      : 0x3ffbd700  
[11:36:01]A2      : 0x00000000  A3      : 0x00000030  A4      : 0x80085548  A5      : 0x3ffbea4c  
[11:36:01]A6      : 0x00000000  A7      : 0x00060023  A8      : 0x801062ca  A9      : 0x3ffbd6d0  
[11:36:02]A10     : 0x00000000  A11     : 0x80000001  A12     : 0x80097648  A13     : 0x3ffe1a80  
[11:36:02]A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001a  EXCCAUSE: 0x00000006  
[11:36:02]EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
[11:36:02]
[11:36:02]
[11:36:02]Backtrace:0x4020fd1f:0x3ffbd7000x401067fd:0x3ffbd720 0x400957ac:0x3ffbd740 
[11:36:02]
[11:36:02]
[11:36:02]
[11:36:02]
[11:36:02]ELF file SHA256: 0000000000000000
[11:36:02]
[11:36:02]Rebooting...
[11:36:02]ets Jul 29 2019 12:21:46
[11:36:02]
[11:36:02]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[11:36:02]configsip: 0, SPIWP:0xee
[11:36:02]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[11:36:02]mode:DIO, clock div:2
[11:36:02]load:0x3fff0030,len:1184
[11:36:02]load:0x40078000,len:13132
[11:36:02]load:0x40080400,len:3036
[11:36:02]entry 0x400805e4
[11:36:02][I][logger:156]: Log initialized
[11:36:02][C][safe_mode:079]: There have been 0 suspected unsuccessful boot attempts
[11:36:02][D][esp32.preferences:114]: Saving 1 preferences to flash...
[11:36:02][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[11:36:02][I][app:029]: Running through setup()...
[11:36:02][C][gpio.one_wire:011]: Setting up 1-wire bus...
[11:36:02][D][number:012]: 'My_Number': Sending state 0.000000
[11:36:02][C][light:035]: Setting up light 'Lamp control PWM'...
[11:36:02][D][light:036]: 'Lamp control PWM' Setting:
[11:36:02][D][light:041]:   Color mode: 
[11:36:02][D][light:085]:   Transition length: 1.0s
[11:36:02][C][output.switch:011]: Setting up Output Switch 'Buzzer'...
[11:36:02][D][switch:016]: 'Buzzer' Turning OFF.
[11:36:02][D][switch:055]: 'Buzzer': Sending state OFF
[11:36:02][C][dallas.temp.sensor:081]: setting up Dallas temperature sensor...
[11:36:02][D][text_sensor:064]: 'Firmware Version': Sending state '2024.6.1 Jun 21 2024, 23:33:20'
[11:36:02][C][esp32_ble:
`

Lauriz7, 2024-06-23:
can confirm issue with one_wire dallas sensor component causing frequent resets.

ssieb, 2024-06-24:
Try again.  I've updated my branch with a potential fix.

georgea62, 2024-06-24:
> Try again. I've updated my branch with a potential fix.

So far is working great.
Thank you!!!!!

Miilani, 2024-06-24:
> Try again. I've updated my branch with a potential fix.

How I update?

georgea62, 2024-06-24:
> > Try again. I've updated my branch with a potential fix.
> 
> How I update?

add this:

```
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: onewire
    components: [ gpio ]
    refresh: 1min

```

Miilani, 2024-06-24:
Apparently it worked, I'll monitor it until tomorrow and come back here to say.

closed 2024-06-24 by kbx81