esphome/issues #5948: !remove not working if id is constructed from substitution

Masterz69, 2024-06-21:
### The problem

`!remove` do not "see" component which name is not static like `sensor_x`, but have substitution inside: `sensor_${id}`.
Error about 'source not found' shown then.

### Which version of ESPHome has the issue?

2024.4.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
----- main file -----

packages: !include { file: xxx.yaml, vars: { id: 1 } }
sensor:
  - id: !remove sensor_1

----- xxx.yaml -----

sensor:
 - platform: template
   id: sensor_${id}
```

jpeletier, 2025-10-28:
Fixed by https://github.com/esphome/esphome/pull/11203
If you can please close the issue, thanks!

Masterz69, 2025-10-28:
> Fixed by [esphome/esphome#11203](https://github.com/esphome/esphome/pull/11203) If you can please close the issue, thanks!

Confirm, it working. Thank You very much.