esphome/issues #5949: http_request.get broken(?)

DragonfireJake, 2024-06-21:
### The problem

`http_request.get` seems to be somewhat broken (at least for me) especially in combination with `max_response_buffer_size`

using the function without `max_response_buffer_size` causes the esp to hang, and then it just spams 
`[02:38:44][E][http_request.arduino:132]: Stream pointer vanished!`

printing out the `body` via
```
char *ptr;
ptr = (char*)body.c_str();
ESP_LOGD("custom", "ptr: %s\n", ptr);
```
shows that there are some weird characters in the beginning:
```
[12:59:25][D][custom:349]: ptr: a8

{"code":"XXX","product":{"brands":"XXX","code":"XXX","product_name":"XXX"},"status":1,"status_verbose":"product found"}
```

after the get call the body should only be composed of the json at the end, no idea where the "a8" and the line brak comes from

Also `max_response_buffer_size` does not seem to work as described in the documentation, for me, the number defined for `max_response_buffer_size` defines the number of characters the `body` instead of the size of it in kB as mentioned in the dock

If I do this:
`max_response_buffer_size: 24`

Instead of what I showed above, the log will show
```
[02:55:05][D][custom:349]: ptr: a8

{"code":"XXXXXXXXXXX
```

If I put in 28 instead of 24 it will show 4 more characters and so on.
Increasing the size to over around 200 will result in a crash of the esp.

(I am not a C++ dev and this is my first time opening an issue on github, sorry if I messed anything up)

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.3

### What platform are you using?

ESP32

### Board

wemos_d1_mini32

### Component causing the issue

http_request

### Example YAML snippet

```yaml
substitutions:
  friendly_name: "Barcode-Scanner-Custom"

  # Query ID for opentindb (https://opengtindb.org/api.php)
[... 408 lines cut ...]
[13:26:16][D][text_sensor:064]: 'Barcode-Scanner-Custom Product': Sending state ''
[13:26:16][D][getEanData:215]: Scanned EAN: \xxx#\xxx?

[13:26:16][D][text_sensor:064]: 'Barcode-Scanner-Custom EAN': Sending state 'XXXXXXXXXXXXX'
[13:26:16][D][openFood:335]: API: https://world.openfoodfacts.org/api/v2/product/XXXXXXXXXXXXX?fields=brands,product_name,code

[13:26:17][D][http_request.arduino:119]: Content-Length: -1
[13:27:32][E][http_request.arduino:132]: Stream pointer vanished!
[13:27:32][E][http_request.arduino:132]: Stream pointer vanished!
[13:27:32][E][http_request.arduino:132]: Stream pointer vanished!
[13:27:32][E][http_request.arduino:132]: Stream pointer vanished!
[13:27:32][E][http_request.arduino:132]: Stream pointer vanished!
[13:27:32][E][http_request.arduino:132]: Stream pointer vanished!
[13:27:32][E][http_request.arduino:132]: Stream pointer vanished!
[13:27:32][E][http_request.arduino:132]: Stream pointer vanished!
[13:27:32][E][http_request.arduino:132]: Stream pointer vanished!
[13:27:32][E][http_request.arduino:132]: Stream pointer vanished!
[13:27:32][E][http_request.arduino:132]: Stream pointer vanished!
[13:27:32][E][http_request.arduino:132]: Stream pointer vanished!
and this goes on infinitley until the device crashes
```

### Additional information

I have two `http_request.get` functions that do different things. Only if both of them have the `max_response_buffer_size` defined the system won't run into the `[02:38:44][E][http_request.arduino:132]: Stream pointer vanished!`-Error upon executing them

Chris112, 2024-06-22:
I too have have a very similiar project with the same URL lookup and tried swapping from the arduino framework to esp-idf and using SSL with no luck. I've rolled esphome back until this is fixed 👀

andrewjswan, 2024-06-24:
https://github.com/esphome/issues/issues/5945#issuecomment-2187027867

MisterRadish, 2024-06-25:
1) Until 2024.6.1 the get_string method worked fine for me on multiple ESP8266 making http requests
Upgrading to 2024.6.1 forced the use of the capture_response: true and use the the "body" variable as get_string was no longer supported. I rolled out to 1 ESP8266 that communicates directly with another ESP8266 running ESPHome and imediately ran into the memory leak issue so I waited until 2024.6.2  #([5955](https://github.com/esphome/issues/issues/5955)) and the fix before continuing my roll out ro other ESPs. 
2) Following the 2024.6.2 fix and when rolling out to other ESP8266 devices I have run into the "[E][http_request.arduino:132]: Stream pointer vanished!" error and the ESP8266 locks up. I have tried setting the max_response_buffer_size to a value for all HTTP requests as mentioned above but this did not solve the issue. 

In summary : The inital ESP (1) is ESPHome to ESPHome and works on 2024.6.2 while the other ESPs (2) do not. The only difference is that the ESPs in (2) are using HTTP request to access a tasmota plug rather than another ESPHome device. Both devices provide a JSON response which caused no issue in 2024.5.5.

MisterRadish, 2024-06-26:
Adding reference to this [comment](https://github.com/esphome/issues/issues/5945#issuecomment-2190324938) which states.

> One thing that was overlooked with the rewrite was servers sending chunked responses with no known Content-Length header. This will need more code added to handle these kind of responses.

andrewjswan, 2024-06-26:
https://github.com/esphome/esphome/pull/6953

andrewjswan, 2024-06-28:
PR is not ready yet.:(

XavierBerger, 2024-06-30:
**I reproduce the issue** `[E][http_request.arduino:132]: Stream pointer vanished!` with my own http server (written in python) using esphome 2024.6.4 **and I managed to make it disappear.**

When I run http_request.get from ESP8285 and when  `Content-Lenght` is not missing, the error is present.

When I add `Content-Lenght`, the error disappears.

Here is the code of esp32
```yaml
[...]
           - http_request.get:
                url: http://${power_meter_ip_address}/emeter/${emeter_index}
                capture_response: true
                on_response:
                  then:
                    - lambda: |-
                        json::parse_json(body, [](JsonObject root) -> bool {
                          id(real_power).publish_state(root["power"].as< float >());
                          return true;
                        });
[...]
```

Here is the server code to reproduce the issue (simulating a Shelly-EM):

```python
import json
from http.server import BaseHTTPRequestHandler, HTTPServer

class SimpleHTTPRequestHandler(BaseHTTPRequestHandler):
    def do_GET(self):
        response = {
            "power": 123,
            "reactive": 56,
            "voltage": 230,
            "is_valid": True,
            "total": 1526,
            "total_returned": 2635
        }
        response_text = json.dumps(response)
        response_bytes = response_text.encode()
        self.send_response(200)
        self.send_header('Content-type', 'application/json')
        self.send_header('Content-Length', str(len(response_bytes)))
        self.end_headers()
        self.wfile.write(response_bytes)

def run(server_class=HTTPServer, handler_class=SimpleHTTPRequestHandler, ip='0.0.0.0', port=8000):
    server_address = (ip, port)
    httpd = server_class(server_address, handler_class)
    print(f'Starting httpd server on {ip}:{port}')
    httpd.serve_forever()

if __name__ == '__main__':
    run()
```

MisterRadish, 2024-07-04:
Any idea when this issue will be addressed? Happy to help test any fix.

MisterRadish, 2024-07-19:
This issue does not seem to be mentioned in the 2024.7.0 release but I tested anyway and can confirm it has not been fixed yet.

andrewjswan, 2024-07-19:
> it has not been fixed yet
@jesserockz said not long ago: https://github.com/esphome/esphome/pull/6953#issuecomment-2209516085
That's why we're waiting. :)

tomatensaus, 2024-07-19:
I get exactly the same issue using the arduino framework. 
This breaks my rest service integration, using an ESP32. 
Fixed the "get_string()" and "on_response:" syntax that was changed.
I suspect this issue also has something to do with the version of the arduino framework.

I tried switching to the IDF framework to test (like this)
```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
```

but the error makes no sense (using Content-Type: application/json
) and posting JSON. 
Error:
```
11:40:54	[W]	[web_server_idf:070][1;31m[httpd][0;33m	
Only application/x-www-form-urlencoded supported for POST request
11:40:57	[I]	[esp-idf:000]	
[0;32mI (29401) HTTP_CLIENT: Body received in fetch header state, 0x3ffd68f6, 14
11:40:57	[E]	[json:084]	
Could not allocate memory for JSON document! Requested 0 bytes, free heap: 110592
```

I cannot see how application/json would not be a valid header for IDF.
I am also rolling back to the last working version. Adding my feedback that this is indeed a big issue

Martin97y, 2024-08-10:
Hello, any solutions for this yet? Im using esp8266mod and im getting same error. [http_request.arduino:132]: Stream pointer vanished!

XeRiZeR, 2024-09-05:
Anyone come up with a solution for this one?

lnxsrt, 2024-09-07:
Bummer that such a core piece of functionality has been INOP for over 3 months now. This is what I am doing as a work around for now. Works for me, but YMMV.

```
external_components:
  - source: github://esphome/esphome@2024.5.5
    components: [ http_request, json ]
```

philpownall, 2024-09-07:
I too was using 2024.5.5, but it only worked for every second request for me: I found this in a discussion on Discord:

```
external_components:
  # makes http_request work with content-length: -1
  - source: github://pr#6953
    components: [ http_request ]
```
Here's the Discord reference:
```
https://discord.com/channels/429907082951524364/1277367067351449662/1277380571865153628
```

Martin97y, 2024-09-08:
> I too was using 2024.5.5, but it only worked for every second request for me: I found this in a discussion on Discord:
>
> ```
> external_components:
>   # makes http_request work with content-length: -1 [...]

And where can i change it?

andrewjswan, 2024-09-08:
There it is also not finished and does not always work as desired, I tested it for about a month, in the end I have abandoned this solution for now.
https://github.com/esphome/esphome/pull/6953#issuecomment-2242341960

Martin97y, 2024-09-08:
> Bummer that such a core piece of functionality has been INOP for over 3 months now. This is what I am doing as a work around for now. Works for me, but YMMV.
> 
> ```
> external_components:
>   - source: github://esphome/esphome@2024.5.5
>     components: [ http_request, json ]
> ```

yo where i can set this?

philpownall, 2024-09-08:
To use a PR as above, you include it at the top of your .yaml file for the node and recompile/install.  Use http_request.get as you would normally.  

As andrewjswan noted, it is use at your own risk, and the PR is not likely to be included in the official release.  It keeps my node working for now...

tomatensaus, 2024-09-16:
I did some testing on the IDF framework and concluded http_request is also broken.
I prepared a minimal snipped to aid re-testing this issue.

I was also not able to run a URL with a port eg. http://192.168.2.16:8000/rest-api  ("The 8000 is interpreted as part of the yaml as key: value entries"

```
esphome:
  name: ipv432

esp32:
  board: esp32dev
  #board: nodemcu-32s
  framework:
    type: esp-idf
    #type: arduino
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
wifi:
  power_save_mode: none
  use_address: ${inverter_network_name}
  ap:
    ssid: "${friendly_name}Hotspot"

captive_portal:

 #debug:
logger:
  level: DEBUG

api:
ota:
  - platform: esphome
    id: esphome_ota
    password: !secret inverter_ota_password
  - platform: http_request

web_server:
  port: 80

http_request:
  # id: http_request_data
  timeout: 15s
  follow_redirects: false
  verify_ssl: false

button:
  - platform: template
    name: "JSONREST"
    id: json_check
    on_press:
      then:
        - http_request.post:
            url: http://192.168.2.16/restapi
            headers:
              Content-Type: application/json
            capture_response: true
            json: |-
              root["esphome"] = ESPHOME_VERSION;
              root["req"] = "test";
            on_response:
              then:
                - lambda: |-
                    json::parse_json(body, [](JsonObject root) -> bool {
                    id(rest_result).publish_state(root["total_returned"]);
                    return true;
                    });

  - platform: restart
    id: reboot

text_sensor:
  - platform: template
    name: "result"
    id: rest_result
```

Running with this webserver (running as root to access port 80 as I could not get the url to accept a port) to test:

```
import json
from http.server import BaseHTTPRequestHandler, HTTPServer

class SimpleHTTPRequestHandler(BaseHTTPRequestHandler):
    def do_POST(self):
        response = {
            "power": 123,
            "reactive": 56,
            "voltage": 230,
            "is_valid": True,
            "total": 1526,
            "total_returned": 2635,
            "boring" : "some boring string with no real value",
            "boring2" : "some boring string with no real value2",
        }
        response_text = json.dumps(response)
        response_bytes = response_text.encode()
        self.send_response(200)
        self.send_header('Content-type', 'application/json')
        # Comment the following line to reproduce the error
        #self.send_header('Content-Length', str(len(response_bytes)))
        self.end_headers()
        self.wfile.write(response_bytes)

def run(server_class=HTTPServer, handler_class=SimpleHTTPRequestHandler, ip='0.0.0.0', port=80):
    server_address = (ip, port)
    httpd = server_class(server_address, handler_class)
    print(f'Starting httpd server on {ip}:{port}')
    httpd.serve_forever()

if __name__ == '__main__':
    run()
```

HLFCode, 2024-10-25:
I can confirm it definitely occurs when the server doesn't include a `Content-Length` header.
Possibly also when header `Transfer-Encoding: chunked` is included
As a fix, if you have a fixed length response you can set `max_response_buffer_size` to 4 characters greater than the expected received body size and filter out the first 4 characters.
Anything bigger will produce the `Stream pointer vanished!` crash so it's a bit risky.
Going back to 2024.5.5 was the only reliable fix for me (not a long-term solution obviously).
If I curl the server which causes the problem I get the following (json body omitted)
```
HTTP/1.1 201 Created
Date: Fri, 25 Oct 2024 07:36:16 GMT
Server: Apache/2
Allow: POST, OPTIONS
Transfer-Encoding: chunked
Content-Type: application/json

```

Is anybody working on this?
Any ideas where to start looking for the code containing the bug?

tomatensaus, 2024-11-01:
> Any ideas where to start looking for the code containing the bug

The code is likely in the http_request component. I was hoping for a fix, but it seems like the devs have bigger issues to attend to. 
I am planning to hack together a fix for the ESP-IDF code (as I am switching my projects over). Will share my patch if I manage to make it work.

HLFCode, 2024-11-23:
I decided to investigate this for myself.
Got a solution.
Not familiar with git so please bear with me.
I did a pull request on the current esphome dev version and edited my [http_request.h](https://github.com/HLFCode/esphome/blob/dev/esphome/components/http_request/http_request.h) with a working solution for the `Stream pointer vanished!` issue and `Content_Header=-1` issue

To use it I put the following in my yaml

```
external_components:
  - source: github://hlfcode/esphome@dev
    components: [ http_request ]

```
All the http_request parameters as normal

It works for me. Please be aware I might refine it/break it so use with care (I don't understand git/push/pull/commit!)
With this I can have extended timeouts and set the watchdog timer to longer than the http timeout so no more reboots with slow servers!

Not sure how to progress refining it or how to get it into esphome...
How do I freeze my dev version before fiddling with new version?

HLFCode, 2024-11-23:
Had a look a branches and pull requests.
I've now made the original branch `main `so the ESPHome code should be
```
external_components:
  - source: github://hlfcode/esphome@main
    components: [ http_request ]

```
If It proves to be long term stable and I don't get any adverse comments I'll create a pull request

philpownall, 2024-11-23:
Mike,

I tried it out with IPGeolocation API at
https://api.ipgeolocation.io/astronomy

With the previous fix:

external_components:
  # makes http_request work with content-length: -1
  - source: github://pr#6953
    components: [ http_request ]

and

      - http_request.get:
          capture_response: true
          # max_response_buffer_size: 1024
          headers:
            Content-Type: application/json

I get this response:

[14:00:01][W][esp-idf:000]: I (81226) HTTP_CLIENT: Body received in fetch
header state, 0x3fcaf957, 249

[14:00:01][W][component:237]: Component time took a long time for an
operation (1358 ms).
[14:00:01][W][component:238]: Components should block for at most 30 ms.

... and the data is fetched correctly.

But with:

external_components:
  - source: ***@***.***
    components: [ http_request ]

and:

      - http_request.get:
          capture_response: true
          # max_response_buffer_size: 1024
          headers:
            Content-Type: application/json

I get this response:

[13:50:01][W][esp-idf:000]: I (156144) HTTP_CLIENT: Body received in fetch
header state, 0x3fcb7813, 249

[13:50:01][E][json:084]: Could not allocate memory for JSON document!
Requested 0 bytes, free heap: 143360
[13:50:01][W][component:237]: Component time took a long time for an
operation (1374 ms).
[13:50:01][W][component:238]: Components should block for at most 30 ms.

And the data is not fetched.  If there's something I missed, please let me
know...

Regards,
Phil

HLFCode, 2024-11-23:
Sorry, not tested on idf, only arduino.
I'll have a think.

HLFCode, 2024-11-24:
Just tested ESP-IDF and I also get

`[E][json:xxx]: Could not allocate memory for JSON document! Requested 0 bytes, free heap: xxxxxx
`
I'll look into it

vasily9k, 2024-11-27:
> I decided to investigate this for myself. Got a solution. Not familiar with git so please bear with me. I created a fork from the current esphome dev version and edited my [http_request.h](https://github.com/HLFCode/esphome/blob/dev/esphome/components/http_request/http_request.h) with a working solution for the `Stream pointer vanished!` issue and `Content_Header=-1` issue
> 
> To use it I put the following in my yaml
> 
> ```
> external_components:
>   - source: github://hlfcode/esphome@dev
>     components: [ http_request ]
> ```
> 
> All the http_request parameters as normal
> 
> It works for me. Please be aware I might refine it/break it so use with care (I don't understand git/push/pull/commit!) With this I can have extended timeouts and set the watchdog timer to longer than the http timeout so no more reboots with slow servers!
> 
> Not sure how to progress refining it or how to get it into esphome... How do I freeze my dev version before fiddling with new version?

Thanks Mike for your decision. The ESP32 stopped looping (with 'Stream pointer vanished!')  after receiving chunked responses.
As for my project, 
after some research of your code, I had to delete not two bytes like yours, but 4 leading bytes (N, N, 0x0D, 0x0A)  and 7 trailing bytes (0x0D, 0x0A, '0', 0x0D, 0x0A, 0x0D, 0x0A) to get the response body as it was sent by **_my server_**.

```
external_components:
  - source: github://vasily9k/esphome@http_request
    components: [ http_request ]

```

HLFCode, 2024-11-27:
@vasily9k  Glad to hear you've made progress.
I agree that the raw returned bytes start N, N, 0x0D, 0x0A and end with some more 0x0D, 0x0A (cr/lf).
I can't see where the cr/lfs could be inserted in the response by the Esphome code and as they are normal ascii line ends I've presumed they were actually transmitted by the server.

The server I'm using sends json as a response and the esphome json decoder removes the cr/lfs by default without problem.
I notice you state the end has some cr/lfs after the trailing 0.
The bytes to c_str conversion should take the 0 as the string terminator so you shouldn't see it or anything after it.

Is your server sending an ascii 0 (0x60) rather than a numeric 0 (0x0) by mistake?

I'm inclined not to prune the response further as some responses might intentionally start/end with cr/lf and unwanted cr/lfs could be removed by the esphome on_response handling.

I'm having a look at the ESP32-IDF platform and on first pass isn't as 'simple' to fix...

HLFCode, 2024-11-27:
Arduino:
My main repository handles (for me) chunked responses. There might be problematic extra white space characters in the reponse body.
Does anybody have a theory where/how whitespace characters are injected into chunked responses?

ESP32-IDF:
My main repository handles (for me) chunked responses.
There are no injected white space characters in the rsponse body

My knowledge does not extend far enough to understand if chunked responses need multiple reads to ensure all chunks are read.
If someone knows of a problematic server please let me know.

```
external_components:
  - source: github://hlfcode/esphome@dev
    components: [ http_request ]

```

lanrat, 2024-11-27:
I'm pretty sure the extra CR/LF's as well as a size octet are part of the `chunked` encoding, and are meant to act as field separators between the chunks. [source](https://en.wikipedia.org/wiki/Chunked_transfer_encoding#Format).

If these are the extra bytes, then it should be the job of the HTTP client library to remove them.

HLFCode, 2024-11-28:
Thanks @lanrat that was a useful observation and link.
Clearly the extra bytes need to be removed.

I'm now not sure if the esphome code should handle this or not.
As I see it:
The idf platform uses Espressif's [esp_http_client ](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/esp_http_client.html)library and this caters for chunked data.
The Ardino platform uses Arduino's [HttpClient ](https://docs.arduino.cc/libraries/httpclient/)library which does not currently cater for chunked data.

I believe the idf platform is catered for in my code but even though the Arduno platform now doesn't crash the esp (stream pointer vanished), it has extra white space characters in the body supplied to the esphome user.

Should the esphome code handle the "chunks" or should the HttpClient library?
It should be clear I'm right up at the edge of my understanding of/ ability with this issue...

I currently have a[ pull request](https://github.com/esphome/esphome/pull/7884) in with my code (my first), but I missed [this pull request](https://github.com/esphome/issues/issues/6392) which also handles the idf platform (partly - misses redirection handling) but doesn't address the Arduino platform.

As far as I see it my code is better than the current esphome code but is still missing a complete Arduino solution.

Should I withdraw my pr until this is resolved?

HLFCode, 2024-11-28:
After a bit of research it seems others have not used the HttpClient library for their chunked handling but gone their own way
According to the logic [here](https://eugeniopace.org/general/http/arduino/2022/05/17/An-opinionanted-http-client-for-arduino.html#:~:text=HTTPResponse%20*%20processChunkedResponse()), the process is something like

- Detect Chunked encoding
- loop while client available
- ..read the chunk length by reading until cr is detected and decoding it
- ..read one byte to discard lf
-  ..read 'length' bytes into the response buffer
-  ..read 2 bytes to discard cr/lf
-  ..terminate the buffer with 0x0 in case it's the last chunk
- end loop
- discard any remaining data

I don't see any reason why the HttpClient couldn't be used this way...

MagMeister, 2024-12-03:
Thank you for your work, I'm making some progress now.
I needed to use @vasily9k 's version since I got a parsing error with @HLFCode 's version.

Still characters at the beginning and at the end:
[image]

Fixed:
[image]

However, in any case it's still a pain in the ass to get http**s** working for the ESP8266 architecture...

vasily9k, 2024-12-03:
@MagMeister
I continued my research for different sized response bodies and found (thanks @lanrat) that the number of leading characters depends on the size of the response body.
    As result: response body has redundant 4-6 leading bytes (L1, L2, L3*, L4*, 0x0D, 0x0A)
   and 7 trailing bytes (0x0D, 0x0A, 0x30, 0x0D, 0x0A, 0x0D, 0x0A)
    L1-L4 are from 2 to 4 characters indicating the length of the content in HEX.

The component has been updated (the crutch has become a little neater) and works.

```
external_components:
  - source: github://vasily9k/esphome@http_request
    components: [ http_request ]
```
PS
Perhaps the following information will be useful to someone: the maximum length of the response body that arrives untrimmed is about 4300 bytes (depends on the size of the HTTP headers). This does not depend on the value of max_response_buffer_size, I set it to 10000

HLFCode, 2024-12-05:
I've done a lot of digging and discovered several issues.
Chunked data has to be read in pieces as the length of the data is not known until you start reading it.
You can't just "top and tail" (trim) a single chunk and expect it to work with multiple chunks.
It can also come in successive transmissions (chunks) of differing lengths.
Some servers send the data with the header thus revealing the overall data length. Others do not.
In the former case the ESP-IDF platform warns you about this at runtime.

I've extensively [reworked my code](https://github.com/HLFCode/esphome) and I believe it now works on ESP-IDF and Arduino platforms. I've tried varying chunk lenghts, delays between chunks and non-chunked
The ESP-IDF platform handles chunked data internally but the Arduino does not.
My code now does not arbitrarily top and tail the data but systematically reads through every chunk received extracting the data as it goes. For example a chunk header can be either 4 or 5 bytes depending on the chunk length.
I have yet to test it on ESP8266...
I have not tested it on very large chunk sizes - I suspect it might fail if the chunk size is larger than the read buffer...

I also wrote a [Python http server](https://github.com/HLFCode/http_server) to test it. It can send chunked or non-chunked data.
As before the code can be used with

```
external_components:
  - source: github://hlfcode/esphome@main
    components: [ http_request ]

```
I'd be grateful for feedback on whether or not it works for you and if not what symptoms you see.

HLFCode, 2024-12-06:
@vasily9k You were more right than I was. The chunk header can be any length depending on how many bytes it takes to store the chunk length as ascii.
So for a chunk size of, say, 15 the header would be `F, cr, lf` but for 255 it would be `F, F, cr, lf`, 256 would be `1, 0, 0, cr, lf` etc
I've updated yet again to allow for up to 6 bytes for the header size data and I've broken up the reads if the chunk size is larger than the receive buffer (taken as 512).
The overall read it still limited by `MAX_SIZE` (1000 I think). This is unchanged from the base ESPHome http_request

I really don't understand github I now have to use
```
external_components:
  - source: github://hlfcode/esphome/esphome@main
    components: [ http_request ]

```
To get esphome to load the correct version
Totally flummoxed by pull requests!

MagMeister, 2024-12-06:
@HLFCode

Works great on my ESP32 **lolin_c3_mini** board for every http(s) request, nice job!

On my ESP8266 **d1_mini** board I'm getting these warnings:
_[W]	[http_request.arduino:065]
Using HTTP on Arduino version >= 3.1 is **very** slow. Consider setting framework version to 3.0.2 in your YAML, or use HTTPS_
But using https, the board keeps rebooting... Maybe it just doesn't have enough process power to deal with https requests?

HLFCode, 2024-12-06:
Thanks @MagMeister 
The `very slow` message is advice built in to the http_component and I have no idea what `very `means...
Re rebooting: I've had that too on an ESP8266 - even before my code update.
I think you need more memory than the ESP8266 has -I noticed the free heap went very low before the reboots.
Another memory eater for me on the ESP8266 was the web server component...

miceno, 2025-01-03:
I am experiencing the same issue on using `http_response.get`, but I think the root cause in my case is related to the way the server sends the response. It is NOT chunk mode.

I was having an "similar" issue recently using the AsyncHTTPRequest_Generic library on Arduino IDE, due to an HTTP server that was non-compliant, it sends LF instead of CR+LF at the end of header lines, and thus, it messes the rest of the parsing of the headers and body. 

My server is not sending any `Content-Length` header on the response.

You have further details [here](https://github.com/miceno/AsyncHTTPRequest_Generic/pull/1).

So I think the HTTP client library is handling the CR+LF badly and it runs out of memory. I tried to review the `http_response` code from ESPHome, but I didn't manage to identify the issue. Just pointing this since it looks like the additional chars that shows are the CR+LF.

miceno, 2025-01-15:
Hi all!

I found a similar issue on this component (an infinite loop on processing the response), and I have a workaround, see here #6682 .

briis, 2025-02-10:
@HLFCode 
> I'd be grateful for feedback on whether or not it works for you and if not what symptoms you see.

This fixed my problem with `http_request.get` on data that was returned in chunked mode. THANKS.

briis, 2025-03-08:
After updating to 2025.2.2 it stopped working again. Not sure what causes this, but nothing changed in the data source or in my code, I am just back to the same error. 😩

briis, 2025-05-22:
Even with the new version of http_request that just came in ESPHome 2025.5, it still does not handle Chuncked Response headers. Your rewrite works great, however as of the above version I start receiving these warnings:

````
WARNING Using `update.UPDATE_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `update.update_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: http_request
````
There is still plenty of time, but it does not seem that they will ever fix that specific problem with the headers directly in ESPHome.

HLFCode, 2025-05-22:
@briis Thanks for the tipoff. I haven't updated to 2025.5 yet.
Hopefully it should be simple to make the changes to avoid the deprecation warnings

HLFCode, 2025-05-29:
OK, I got it to fail with the config below with a compile error which I don't get with the current ESPHome code

```
src/esphome/components/online_image/online_image.cpp: In member function 'virtual void esphome::online_image::OnlineImage::update()':
src/esphome/components/online_image/online_image.cpp:128:44: error: cannot convert 'std::__cxx11::basic_string<char>' to 'const char*' in assignment
   accept_header.value = accept_mime_type + ",*/*;q=0.8";
                                            ^~~~~~~~~~~~
*** [.pioenvs/test-http/src/esphome/components/online_image/online_image.cpp.o] Error 1

```
Not sure where I should start looking as my code does not change data types.

Config:
```
esphome:
  name: test-http
  friendly_name: Test HTTP

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

external_components:
  - source: github://hlfcode/esphome@main
    components: [ http_request ]

# Enable Home Assistant API
api:
  encryption:
    key: "D25eJT/23nBJ3743gyldYRXj4gEB/oYZPzdxWzWuQ8I="
  reboot_timeout: 0s
    

ota:
  - platform: esphome
    password: "1423e63134ebb40e16f750cd143b9bbd"

wifi:
  ssid: !secret office_wifi_ssid
  password: !secret office_wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test-Http Fallback Hotspot"
    password: "fL3CUcFxcRMc"

captive_portal:
    
http_request:
  id: http_test
  verify_ssl: false

online_image:
  - url: "https://example.com/example.png"
    format: png
    id: my_online_image
    type: RGB

display:
  - platform: addressable_light
    addressable_light_id: test_light
    height: 100
    width: 100

light:
  - platform: fastled_clockless
    id: test_light
    chipset: WS2811
    pin: GPIO02
    num_leds: 60
    rgb_order: BRG
    name: "FastLED WS2811 Light"

```

ArcticXWolf, 2025-05-29:
Ah okay, I was just about to post you my config:
```
esphome:
  name: example

# ESPHOME has problems with chunked HTTP responses:
# https://github.com/esphome/issues/issues/5949
external_components:
  - source: github://hlfcode/esphome/esphome@main
    components: [ http_request ]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap: {}
  on_connect:
    - delay: 16ms
    - script.execute: example_script

captive_portal:

ota:
  platform: esphome

api:

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  level: verbose

i2c:

http_request:
  verify_ssl: false
  timeout: 10s

online_image:
  - id: example_image
    format: bmp
    type: BINARY
    url: https://example.com/test.bmp # random test image
    resize: 800x480
    on_download_finished:
      - delay: 16ms
      - component.update: example_display
      - logger.log:
          format: "online_image downloaded"

script:
  - id: example_script
    then:
      - logger.log: "Triggering download"
      - http_request.get:
          url: https://google.com
          capture_response: true
          max_response_buffer_size: "4096"
          on_error:
            - logger.log:
                format: "HTTP Error"
          on_response:
            - logger.log:
                format: "HTTP Response: %s"
                args: [body.c_str()]
            # I would do something with the HTTP response here (changing the online_image url)
            - delay: 16ms  # lets the http_request component free up the reserved memory
            - component.update: example_image

# online_image needs a display, so this is mine, inkplate 10
pca6416a:
  - id: pca6416a_hub
    address: 0x20
    # Primary expander for display control and additional I/O
  - id: pca6416a_hub2
    address: 0x21
    # Secondary expander for additional I/O

display:
  - platform: inkplate6
    id: example_display
    greyscale: true
    partial_updating: true
    update_interval: never
    model: inkplate_10

    ckv_pin: 32
    sph_pin: 33
    gmod_pin:
      pca6416a: pca6416a_hub
      number: 1
    gpio0_enable_pin:
      pca6416a: pca6416a_hub
      number: 8
    oe_pin:
      pca6416a: pca6416a_hub
      number: 0
    spv_pin:
      pca6416a: pca6416a_hub
      number: 2
    powerup_pin:
      pca6416a: pca6416a_hub
      number: 4
    wakeup_pin:
      pca6416a: pca6416a_hub
      number: 3
    vcom_pin:
      pca6416a: pca6416a_hub
      number: 5
```

ArcticXWolf, 2025-05-29:
Ah, I think I found it. It is just a version incompatibility. This [commit here](https://github.com/esphome/esphome/commit/8b7aa4c1102f2102d38a2d189e678b6761ee7d51) changed the Header struct in http_requests to strings from char. And your fix is based on the older implementation. So when the new online_image expects std::string, but your older http_requests gives a const char, then it cannot convert it.

HLFCode, 2025-06-03:
@ArcticXWolf Great - that looks like something I can try and learn by.
I presume you did this all by "command line" and not in VSCode or Github directly?

I'm away on holiday for a few days so 'll try it when I get back
Many thanks

ArcticXWolf, 2025-06-04:
Yeah, I use git on the command line to have full control about what it does. For basic tasks the GUI tools (github web and vscode) work well, but everything that takes more than simple commits, merges, etc I use the CLI.

HLFCode, 2025-06-07:
@ArcticXWolf Hit a brick wall at your step 3 `git pull hlfcode`
Got

```
>git pull hlfcode
The authenticity of host 'github.com (20.26.156.215)' can't be established.
ED25519 key fingerprint is SHA256:xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx <- masked
This key is not known by any other names.
Are you sure you want to continue connecting (yes/no/[fingerprint])? y
Please type 'yes', 'no' or the fingerprint: yes
Warning: Permanently added 'github.com' (ED25519) to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

```
I did a search but got nowhere..
Any ideas?

[I'm using Windows btw]

HLFCode, 2025-06-07:
@ArcticXWolf Thanks for all the help.
Feeling a bit down about it atm.
I'm not proud about who gets the credit for the code - I just want it (or anything like it) incorporated into esphome.

Feel free to copy/use my code, meanwhile I'll have a look at you references, try to start small and build up to a repository the size/complexity of esphome.

HLFCode, 2025-06-07:
Really struggling with github
The [commit](https://github.com/esphome/esphome/commit/8b7aa4c1102f2102d38a2d189e678b6761ee7d51) shows the new structure using `const char` in http_request.h but if I go to the [dev branch](https://github.com/esphome/esphome/blob/dev/esphome/components/http_request/http_request.h) for the same file it shows the old code/structure using `std::string`

Where is the [commit ](https://github.com/esphome/esphome/commit/8b7aa4c1102f2102d38a2d189e678b6761ee7d51)stored and why isn't it in the dev branch of esphome?

How can I get a clean copy of esphome with the committed code?

ArcticXWolf, 2025-06-07:
I think you read that diff the wrong way around:
The commit 8b7aa4c has as the diff:
```
struct Header {
-  const char *name;
-  const char *value;
+ std::string name;
+ std::string value;
};
```
This means the commit removed the const chars and replaced them with strings.
And this is what you see as the current file on the dev branch:

```
struct Header {
  std::string name;
  std::string value;
};
```

All lines with `-` are removed in a commit and all lines with `+` are added.

ArcticXWolf, 2025-06-07:
I dont think you need to delete your fork, but it can be a way to solve it. Just before you do that, a small tip: You can [take a look here](https://github.com/esphome/esphome/compare/dev...HLFCode:esphome:main) before you delete your fork. It shows you the differences between your branch and the dev branch, i.e. it shows the final result of your work. You can even get a text version for download [by adding .diff to that URL](https://github.com/esphome/esphome/compare/dev...HLFCode:esphome:main.diff).

So download that before you delete your fork (after you delete it that site isnt available anymore) and then use that to redo everything in your new fork 🙂

HLFCode, 2025-06-08:
Sorry @ArcticXWolf had a bad day yesterday and thanks for your help.

I had a look at the diff and I think the differences are OK in that they represent my intent and do not include the `Header` as a struct of `char *`.
My fork has the Header as `std:string`s like the current esphome dev branch.
So my http_request.h has
```
struct Header {
  std::string name;
  std::string value;
};

```
Which I believe is correct.

Compiling yaml with an http_request compiles fine (not yet tested functionality):
```
      - http_request.post:
          url: "http://192.168.1.5:8000"
          capture_response: true
          headers:
            Content-Type: application/json
            X-Pin: 1
            X-Sensor: 1234;
          json: |-
            root["software_version"] = "abcd";
          on_response:
            then:
              - logger.log:
                  format: "status code %s: %s"
                  args: [response->status_code, body.c_str()]

```

I tried compiling with an online_image and still got the error you get:
```
src/esphome/components/online_image/online_image.cpp: In member function 'virtual void esphome::online_image::OnlineImage::update()':
src/esphome/components/online_image/online_image.cpp:128:44: error: cannot convert 'std::__cxx11::basic_string<char>' to 'const char*' in assignment
   accept_header.value = accept_mime_type + ",*/*;q=0.8";
                                            ^~~~~~~~~~~~

```
This is weird on two levels:

- Why is it trying to convert to const char*
- Why is it reporting line 128 when it's on line 134

Is my esphome compiler picking up an earlier version of online_image somehow or is online_image not picking up on my fork?
My yaml points to my fork like this
```
external_components:
  - source: github://hlfcode/esphome@main
    components: [ http_request ]

```

FWIW I'm using esphome V2025.5.2 and HA is not offering me an update on this.

HLFCode, 2025-06-08:
I'm definitely not picking up something right with the compiler.
If I remove my `external components` in the yaml and use the esphome current fork entirely I'm told the headers are now request_headers
```
INFO ESPHome 2025.5.2
INFO Reading configuration /config/esphome/test-http.yaml...
Failed config

button.template: [source /config/esphome/test-http.yaml:50]
  platform: template
  id: test_button
  on_press: 
    - http_request.post: 
        url: http://192.168.1.43:8000
        capture_response: True
        
        The 'headers' options has been renamed to 'request_headers'.
        headers: 
          Content-Type: application/json
          X-Pin: 1
          X-Sensor: 1234;
        json: root["software_version"] = "abcd";
        on_response: 
          then: 
            - logger.log: 
                format: status code %s: %s
                args: 
                  - response->status_code
                  - body.c_str()

```
Yet when I look at my `__init__.py` on [github](https://github.com/HLFCode/esphome/blob/main/esphome/components/http_request/__init__.py) it has the same CONSTs as the esphome fork but the above error implies my component code is out of date (uses `headers` and not `request_headers`) 

I feel I'm missing something really obvious

ArcticXWolf, 2025-06-08:
I dont know exactly what is going wrong, but it might be that you have an outdated cached version of your repo that esphome uses.
What I mean by that is that if you use `external_components` then ESPhome downloads a copy of that repo (your repo) to the `.esphome/external_components` directory. And my suspicion is that this cache is still your old version and not what you have here on Github.

Maybe try deleting/moving the `.esphome` folder and try again to trigger a fresh download.

ArcticXWolf, 2025-06-08:
Hmm, this is weird, because as far as I can see the release branch is included in the dev branch history:

[image]

You can see here that at commit 367017b35, the `origin/release` (which is the esphome release branch on my pc) was merged into the `origin/dev` branch and since then the release branch hasnt had any new commits.

HLFCode, 2025-06-08:
Not sure what to say, the dev and release have different `__init__.py`s in their http_request component.
dev has added references to constants which the release does not have.

[6 more replies]