esphome/issues #5952: Component esphome.ota is marked FAILED

cburghardt, 2024-06-22:
### The problem

I upgraded from ESP home 2024.5 to 2024.6, adopted the config to the new OTA
The update itself worked but the log shows **Component esphome.ota is marked FAILED**

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.4

### What platform are you using?

ESP32-IDF

### Example YAML snippet

```yaml
substitutions:
  name: esp32-bluetooth-proxy-c581e8
  friendly_name: Bluetooth Proxy
packages:
  esphome.bluetooth-proxy: github://esphome/firmware/bluetooth-proxy/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  
api:
  encryption:
    key: 

esp32:
  board: esp32dev
  framework:
    type: esp-idf

ota:
  - platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Anything in the logs that might be useful for us?

```txt
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.178.143 using esphome API
INFO Successfully connected to esp32-bluetooth-proxy-c581e8 @ 192.168.178.143 in 11.192s
INFO Successful handshake with esp32-bluetooth-proxy-c581e8 @ 192.168.178.143 in 0.187s
[12:09:49][I][app:100]: ESPHome version 2024.6.1 compiled on Jun 22 2024, 11:58:13
[12:09:49][I][app:102]: Project esphome.bluetooth-proxy version 1.0
[12:09:49][C][wifi:599]: WiFi:
[12:09:49][C][wifi:427]:   Local MAC: 30:83:98:C5:81:E8
[12:09:49][C][wifi:432]:   SSID: [redacted]
[12:09:49][C][wifi:435]:   IP Address: 192.168.178.143
[12:09:49][C][wifi:439]:   BSSID: [redacted]
[12:09:49][C][wifi:440]:   Hostname: 'esp32-bluetooth-proxy-c581e8'
[12:09:49][C][wifi:442]:   Signal strength: -56 dB ▂▄▆█
[12:09:49][C][wifi:446]:   Channel: 6
[12:09:49][C][wifi:447]:   Subnet: 255.255.255.0
[12:09:50][C][wifi:448]:   Gateway: 192.168.178.1
[12:09:50][C][wifi:449]:   DNS1: 192.168.178.109
[12:09:50][C][wifi:450]:   DNS2: 0.0.0.0
[12:09:50][C][logger:185]: Logger:
[12:09:50][C][logger:186]:   Level: DEBUG
[12:09:50][C][logger:188]:   Log Baud Rate: 115200
[12:09:50][C][logger:189]:   Hardware UART: UART0
[12:09:50][C][bluetooth_proxy:088]: Bluetooth Proxy:
[12:09:50][C][bluetooth_proxy:089]:   Active: YES
[12:09:50][C][safe_mode.button:024]: Safe Mode Button 'Safe Mode Boot'
[12:09:50][C][safe_mode.button:024]:   Icon: 'mdi:restart-alert'
[12:09:50][C][esp32_ble:383]: ESP32 BLE:
[12:09:50][C][esp32_ble:385]:   MAC address: 30:83:98:C5:81:EA
[12:09:50][C][esp32_ble:386]:   IO Capability: none
[12:09:50][C][esp32_ble_tracker:653]: BLE Tracker:
[12:09:50][C][esp32_ble_tracker:654]:   Scan Duration: 300 s
[12:09:50][C][esp32_ble_tracker:655]:   Scan Interval: 320.0 ms
[12:09:50][C][esp32_ble_tracker:656]:   Scan Window: 30.0 ms
[12:09:50][C][esp32_ble_tracker:657]:   Scan Type: ACTIVE
[12:09:50][C][esp32_ble_tracker:658]:   Continuous Scanning: True
[12:09:50][C][mdns:115]: mDNS:
[12:09:50][C][mdns:116]:   Hostname: esp32-bluetooth-proxy-c581e8
[12:09:50][C][esphome.ota:073]: Over-The-Air updates:
[12:09:50][C][esphome.ota:074]:   Address: esp32-bluetooth-proxy-c581e8.local:3232
[12:09:50][C][esphome.ota:075]:   Version: 2
[12:09:50][C][esphome.ota:073]: Over-The-Air updates:
[12:09:50][C][esphome.ota:074]:   Address: esp32-bluetooth-proxy-c581e8.local:3232
[12:09:50][C][esphome.ota:075]:   Version: 2
[12:09:50][E][component:082]:   Component esphome.ota is marked FAILED
[12:09:50][C][safe_mode:018]: Safe Mode:
[12:09:50][C][safe_mode:020]:   Boot considered successful after 60 seconds
[12:09:50][C][safe_mode:021]:   Invoke after 10 boot attempts
[12:09:50][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[12:09:50][C][api:139]: API Server:
[12:09:50][C][api:140]:   Address: esp32-bluetooth-proxy-c581e8.local:6053
[12:09:50][C][api:142]:   Using noise encryption: YES
[12:09:50][C][improv_serial:032]: Improv Serial:
```

falconindy, 2024-06-22:
I'm seeing this as well for a device of mine, but I should also mention that it's benign -- I can still remotely update the device. The problem is like you've got a duplicated ota: section in your YAML. One is explicitly in your own config, and the other is in remote package, as of https://github.com/esphome/firmware/commit/87a7b4b54ecd6588b7942417230bfdb8fdbad113.

afarago, 2024-06-23:
I am seeing a similar issue (ESP32, IDF)

ota is also "duplicated", one in my base config (via package), and one (via package) in a bluetooth include.
When I remove one of the ota elements it works fine.

falconindy, 2024-06-24:
Definitely works, but definitely not convenient, since sometimes one of the ota sections comes from a vendor and won't have an id, or even a consistent id. This makes it hard to reuse upstream configs and/or package your own base overlay that has an ota section with things like a password.

gpambrozio, 2024-06-25:
I was seeing this and was not able to update some of my devices using ota (ota update failed with connection reset). In my case I had a button that allowed me to boot the device in safe mode so activating that allowed be to do an ota update.

jesserockz, 2024-06-26:
https://github.com/esphome/esphome/pull/6984 fixes this by stopping two instances of the esphome ota platform in config with the same port. 

It is not a final fix, but it should help by making it clearer to users before installing.

closed 2024-06-26 by jesserockz