esphome/issues #5955: 2024.6.1 http_request GET memory leak

rckstrh, 2024-06-23:
### The problem

After updating to 2024.6.1 and changing http_request and json configurations accordingly in a bigger project I am working on, I have seen that the ESP8266 I am using is reporting out of memory errors after a short while and finally crashes due to that. I have played around to find the root cause and have prepared a minimal configuration (without json) that shows that when using http_request and setting the capture_response variable to true the module leaks memory. For that I have prepared a simple timer that tries to GET a 512 byte txt file and having the debug sensor module enabled showing free heap space. 

When having capture_response set to false the free heap size stays constant but when having capture_response set to true the free heap size steadily declines until the microcontroller runs out of memory and crashes. This also happens when there is "nothing to be done to the response" e.g. when commenting out the "on_response" block or commenting out the debug output in the on_response lambda. 

I have attached the minimal config I was using to reliably reproduce the memory leak. As mentioned, you can try and comment out the logging in the lambda in line 41 or just completely comment out the on_response block in line 37-41 -> when setting capture_response: true in line 36 it will leak memory.

I know there are already other issues (#5949, #5951) open regarding http_request but in this issue I try to provide you with a minimal example and my findings on that matter and my assumption that it is caused by a memory leak in http_request when using capture_response / handling of the response body. I have not found a way to somehow free dangling memory allocated internally by the http_request component from within the on_response lambda.

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

http_request

### Example YAML snippet

```yaml
esphome:
  name: httprequestmemleakexample
  friendly_name: HttpRequestMemLeakExample

esp8266:
  board: d1_mini

# Enable Home Assistant API
api:
  encryption:
    key: "OFAtXlMrl9spHbpJnP4YKnU9GClHbaUJPPdT9K/j8gE="

ota:
  - platform: esphome
    password: "debugonly"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:
    
http_request:
  useragent: esphome/device
  id: http_liquid
  timeout: 10s
  verify_ssl: false

time:
  - platform: sntp
    on_time: 
      - seconds: /1
        then:
          - http_request.get:
              url: "http://krautomation.tech/debug/debug_512.txt"
              capture_response: true # setting this to false and the memory does not leak
              on_response:
                then:
                  - lambda: |-
                      // try and comment the following line out; it still leaks memory when capture_response is set to true
                      ESP_LOGD("http_request_response", "body len: %d", body.length());

debug:
  update_interval: 5s

sensor:
  - platform: debug
    free:
      name: "Heap Free"

# Logger must be at least debug (default)
logger:
  level: debug
```

gabest11, 2024-06-23:
https://github.com/esphome/esphome/blob/1e05bcaa614040617ef94340a54ca3f22871c4ac/esphome/components/http_request/http_request.h#L140

I think 
‎ExternalRAMAllocator::allocate is not paired with 
‎ExternalRAMAllocator::deallocate...

MisterRadish, 2024-06-23:
I can confirm that I appear to be getting the same issue after upgrading to 2024.6.1 on multpile ESP8266 and board esp01_1m

further info:
I was using id(http_request_id).get_string() up to this point to retrieve the response as described in https://esphome.io/cookbook/http_request_sensor.html without issue but as the get_string() method no longer seems to be supported I  therefore migrated all my boards to use "capture_response: true" with the created "body" variable and now have the reboot issue as described.

closed 2024-06-24 by jesserockz