esphome/issues #5958: RP2040 SI7021 DHT not measuring after upgrade to 2024.6.1

arnasDan, 2024-06-23:
### The problem

I upgraded ESPHome from 2024.2 to 2024.6 for a Pico W with the Sonoff SI7021 sensor. Measurement always fails after the upgrade; after a rollback to 2024.2, it works fine again.
It seems likely to be related to https://github.com/esphome/esphome/pull/6604, which altered the timings used for this sensor. It appears that the old configuration is valid for the RP2040 as well (not just the ESP8266, as implemented).

### Which version of ESPHome has the issue?

2024.6.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

RP2040

### Board

rpipicow

### Component causing the issue

dht

### Anything in the logs that might be useful for us?

```txt
[21:03:16][W][dht:169]: Requesting data from DHT failed!
[21:03:16][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
```

emefff, 2024-06-27:
Hello,
did you try to address the SI7021 via HTU21D component? Maybe that works:
https://esphome.io/components/sensor/htu21d.html

arnasDan, 2024-06-27:
The HTU21D component interfaces with the SI7021 chip directly and uses I²C, while the Sonoff device that I have only exposes a one wire signal, so that's unfortunately not an option unless I modify the board :)