esphome/issues #5964: Home Assistant crashes while compiling with 4GB RAM using ESPhome 2024.6.2

darth-hp, 2024-06-25:
### The problem

I installed 2024.6.2 and tried to update a ESP32 board and then I got a disconnect not realizing that HA crashed in the background. I tried three times - all the same. Increasing memory to 5GB changed the game and install went fine.

Below is the log output I saw while the issue arised:

```
Processing jarvis-01 (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.4
|-- NeoPixelBus @ 2.7.3
Compiling .pioenvs/jarvis-01/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/jarvis-01/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/jarvis-01/src/esphome/components/api/api_pb2.cpp.o
```

### Which version of ESPHome has the issue?

2024.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.2 HAOS 12.4

### What platform are you using?

ESP32

### Board

ESP-WROOM-32

### Component causing the issue

unknown

### Example YAML snippet

```yaml
substitutions:
  name: "jarvis-01"
  friendly_name: Jarvis-01

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  on_boot:
    priority: 200
    then:
      - wait_until:
          condition:
            wifi.connected:
          timeout: 1s

esp32:
  board: esp32dev
  framework:
    type: arduino
#    type: esp-idf

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0

<<: !include .common.yaml

#micro_wake_word:
#  model: hey_jarvis

i2s_audio:
  - id: i2s_in
    i2s_lrclk_pin: GPIO26 #WS 
    i2s_bclk_pin: GPIO25 #SCK

microphone:
  - platform: i2s_audio
    adc_type: external
    pdm: false
    id: mic_i2s
    channel: right
    bits_per_sample: 32bit
    i2s_audio_id: i2s_in
    i2s_din_pin: GPIO33  #SD Pin from the INMP441 Microphone

voice_assistant:
  microphone: mic_i2s
  id: va
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  use_wake_word: false
  #media_player: media_player_speaker
  
  on_wake_word_detected: 
    - light.turn_on:
        id: led_light
  
  on_listening: 
    - light.turn_on:
        id: led_light
        effect: "Scan Effect"
        red: 63%
        green: 13%
        blue: 93%

#  on_stt_vad_start:
#    - light.turn_on:
#        id: led_light
#        effect: "Pulse"
#        red: 0%
#        green: 100%
#        blue: 100%
  
  on_stt_end:
    - light.turn_on:
        id: led_light
        effect: "None"
        red: 0%
        green: 100%
        blue: 0%

  on_error: 
    - light.turn_on:
        id: led_light
        effect: "None"
    - if:
        condition:
          switch.is_on: use_wake_word
        then:
          - switch.turn_off: use_wake_word
          - delay: 1sec 
          - switch.turn_on: use_wake_word      

  on_client_connected:
    - if:
        condition:
          switch.is_on: use_wake_word
        then:
          - voice_assistant.start_continuous:

  on_client_disconnected:
    - if:
        condition:
          switch.is_on: use_wake_word
        then:
          - voice_assistant.stop:
 
  on_end:
    - light.turn_off:
        id: led_light

binary_sensor:
  - platform: status
    name: API Connection
    id: api_connection
    filters:
      - delayed_on: 1s
    on_press:
      - if:
          condition:
            switch.is_on: use_wake_word
          then:
            - voice_assistant.start_continuous:
    on_release:
      - if:
          condition:
            switch.is_on: use_wake_word
          then:
            - voice_assistant.stop:

switch:
  - platform: template
    name: Use wake word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - lambda: id(va).set_use_wake_word(true);
      - if:
          condition:
            not:
              - voice_assistant.is_running
          then:
            - voice_assistant.start_continuous
    
    on_turn_off:
      - voice_assistant.stop
      - lambda: id(va).set_use_wake_word(false);

light:
  - platform: neopixelbus
    id: led_light
    type: grb
    pin: GPIO32      # DIN pin of the LED Strip
    num_leds: 24      # change the Number of LEDS according to your LED Strip.
    name: "Light"
    variant: ws2812x
    default_transition_length: 0.8s
    effects:
      - addressable_scan:
          name: "Scan Effect"
          move_interval: 100ms
          scan_width: 2
      - pulse:
          name: "Pulse"
          update_interval: 2s
```

jburgess777, 2024-06-27:
I have similar issues with my 2GB Odroid-n2 running out of memory when compiling Esphome. It had been working fine for the past year, but started crashing during updates in the past few weeks. I don’t know whether the memory required by the compiler has increased, or whether the free Ram has decreased.

Could there be a way to add a ‘—jobs’ option to ‘pio run’ to reduce the number of concurrent compiler processes? https://docs.platformio.org/en/latest/core/userguide/cmd_run.html#options

darth-hp, 2024-07-04:
[image]

jburgess777, 2024-07-05:
I found a workaround by limiting the number of concurrent compiler processes to 2. The default is to use all CPUs, which was 6 in my case. By running fewer compilers, the peak memory usage is reduced.

https://community.home-assistant.io/t/compilation-fails-xtensa-lx106-elf-g-fatal-error-killed-signal-terminated-program-cc1plus/463184/18

ssieb, 2024-07-08:
Yes, that is the workaround.  Compiling takes a lot of RAM.

closed 2024-07-08 by ssieb

darth-hp, 2024-07-18:
Interims fix:

Try for every affected component
```yaml
esphome:
  compile_process_limit: 1
```

or set compile_process_limit in the addon itself.

I have set it to 4 now.