esphome/issues #5965: OTA automations do not get triggered

DeanCording, 2024-06-25:
### The problem

OTA on_begin: and on_end:  automations do not get triggered during an OTA update using the ESPHome platform.

### Which version of ESPHome has the issue?

2024.6.0  onwards

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

ESP-12

### Component causing the issue

ota

### Example YAML snippet

```yaml
ota:
  - platform: esphome
    on_begin:
        then:
        - lambda: |-
            id(serial).write_str(" \n");
            id(serial).flush();
            id(serial).write_str("{\"reset\":1}\n");
            id(serial).flush();
            id(serial).write_str("Updating...\n");
            id(serial).flush();
    on_end:
        then:
        - lambda: |-
            id(serial).write_str(" \n");
            id(serial).flush();
            id(serial).write_str("{\"reset\":1}\n");
            id(serial).flush();
            id(serial).write_str("Rebooting...\n");
            id(serial).flush();
```

kbx81, 2024-06-25:
Based on your logs, this is likely because of #5952, not because of a problem with the automations.

DeanCording, 2024-06-26:
Yes, that was the issue.

closed 2024-06-26 by DeanCording