esphome/issues #5966: Wake word still does not respond after a while

itnassol, 2024-06-25:
### The problem

The response at first is perfect I can call the wake word and is very quick to respond. However if I don't use it for a while, and I have done some testing and it's about 15 minutes, it's like it has gone to sleep I then have to tap the speaker to "wake it up" it's then good for another 15 minutes.

### Which version of ESPHome has the issue?

2024.6.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.4

### What platform are you using?

ESP32

### Board

onju-voice

### Example YAML snippet

```yaml
substitutions:
  name: "dr-ada"
  friendly_name: "DR Ada"
  wifi_ap_password: "password"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  min_version: 2023.10.1
  on_boot:
    then:
      - light.turn_on:
          id: top_led
          effect: slow_pulse
          red: 100%
          green: 60%
          blue: 0%
      - wait_until:
          condition:
            wifi.connected:
      - light.turn_on:
          id: top_led
          effect: pulse
          red: 0%
          green: 100%
          blue: 0%
      - wait_until:
          condition:
            api.connected:
      - light.turn_on:
          id: top_led
          effect: none
          red: 0%
          green: 100%
          blue: 0%
      - delay: 1s
      - script.execute: reset_led

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
[... 468 lines cut ...]
              return;
          }

switch:
  - platform: template
    name: Use Wake Word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    on_turn_on:
      - script.execute: turn_on_wake_word
    on_turn_off:
      - script.execute: turn_off_wake_word

  - platform: template
    name: Listen LR
    id: listen_lr
    optimistic: true
    on_turn_on:
      - switch.turn_off: use_wake_word
      - delay: 1s
      - voice_assistant.start_continuous
    on_turn_off:
      - switch.turn_on: use_wake_word
```

darki73, 2024-07-01:
From my observations, this is the issue with the Arduino framework.

Whenever i use the simplified configuration with `esp-idf` and just `speaker` + `microphone` + `voice_assistant` + `micro_wake_word`, it works every single time, even from 3 to 7 meters away from the speaker.

Turns out, and correct me if i am wrong, arduino framework is only capable of utilizing one core, and all logic runs on the main thread (hence why we are unable to use `micro_wake_word` with arduino framework).

Observing the traffic:
1. Satellite sends traffic to Home Assistant
2. Home Assistant sends traffic to Wake Word detection instance (i run it on a separate VM)
3. Gets the result back, usually with error that no wake word is detected
4. Either satellite or Home Assistant gets overwhelmed with information

Upon a closer look at the pipeline, it triggers wake word detection every 0.5 to 2 seconds whenever any source of voice/audio is around.

Sadly, while using ESP32-S3 module (i designed a custom PCB for that), you have to either rely on the arduino to behave (you can stop and start `voice_assistant` every 5 to 10 minutes, and it is still somewhat broken), or forget about the `media_player` component (i really wanted it as having a whole house audio system is kinda awesome) and just use your 55 USD (price per board + speaker + leds + mic + 3d printing an enclosure) as a simple voice assistant which works every single time no matter where you are in the room.

itnassol, 2024-07-01:
HI Ivan,

Just brilliant, although this is way above my head, it has been interesting taking a deeper delve into this, thank you for your time with this, at the moment I have found a few work arounds in order to make it a little more compatible with what I am doing and everything seems to be working. As I have 5 (at the moment) converted Google minis around the house with more to follow, it dawned on me to only trigger the wake word... A. When the room is occupied and B. After it has been dormant for a few minutes.

So, each speaker is essentially in sleep mode until someone is in the room, I do this using ESP presence, as it is an old Victorian house with thick walls setting the ESP presence in each separate room is fairly simple, it then sets the speaker to listen when someone walks in, this allows me to just say a command as soon as I walk in like.... "Lights on" for example. the it goes to wake word. If the wake word is not used for 15 minutes I reload the integration and it's fine again. The only exception to this is if the system is asking me something, for example when it starts to get dark, the wake word turns off and the system asks, do I want evening mode, if I say Yes please it trigger evening mode, I I say No thanks it just goes back to wake word. 

A bit Heath Robinson, but it works, and people are a little surpirsed when the system asks me if i want somethinf done, lol...

darki73, 2024-07-01:
@itnassol you might want to give a shot to `https://github.com/gnumpi/esphome_audio` with the `esp-idf` framework, with the following options:
```yaml
framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_ESP32_S3_BOX_BOARD: "y"
      COMPILER_OPTIMIZATION_SIZE: "y"

      CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM: "16"
      CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM: "512"
      CONFIG_TCPIP_RECVMBOX_SIZE: "512"
      CONFIG_TCP_SND_BUF_DEFAULT: "65535"
      CONFIG_TCP_WND_DEFAULT: "512000"
      CONFIG_TCP_RECVMBOX_SIZE: "512"
```

and this as a settings for your `i2s_audio`:
```yaml
---
external_components:
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: dev-next
    components:
      - adf_pipeline
      - i2s_audio
    refresh: 0s

i2s_audio:
  - id: i2s_in
    i2s_lrclk_pin: GPIO7
    i2s_bclk_pin: GPIO16
  - id: i2s_out
    i2s_lrclk_pin: GPIO8
    i2s_bclk_pin: GPIO18

adf_pipeline:
  - platform: i2s_audio
    id: adf_i2s_in
    type: audio_in
    i2s_audio_id: i2s_in
    i2s_din_pin: GPIO15
    pdm: false
    channel: left
    sample_rate: 16000
    bits_per_sample: 32bit
  - platform: i2s_audio
    id: adf_i2s_out
    type: audio_out
    i2s_audio_id: i2s_out
    i2s_dout_pin: GPIO17
    adf_alc: true
    alc_max: .5

microphone:
  - platform: adf_pipeline
    id: i2s_mic
    gain_log2: 3
    keep_pipeline_alive: false
    pipeline:
      - adf_i2s_in
      - self

media_player:
  - platform: adf_pipeline
    id: i2s_player
    name: "${device_friendly_name} Media Player"
    keep_pipeline_alive: false
    internal: false
    pipeline:
      - self
      - adf_i2s_out
```

I managed to get it working just enough for all the speakers being able to hear me no matter where i am in the house.

Just a side note, i am using MAX98357A as an external AMP and INMP441 (waiting for other mics to be delivered) so you might need to tweak some settings.

With the `esp-idf` i not have almost no issues (the device reboots sometimes due to ADF issues, but this is just the way it is, gnumpi did really amazing job with his library, but as far as i understand, he is a sole developer, so this is why it can be a hit or miss for some).

itnassol, 2024-07-02:
Thank you, This is all exciting stuff, I will give it a go today.