esphome/issues #5969: Climate Integration won't set below 17C

simondrake, 2024-06-25:
### The problem

I have an ESP dongle attached to my Bosch Air Conditioning unit, flashed with ESP Home and the following configuration:

```yaml
esphome:
  name: "air-conditioner-office"
  platform: ESP8266
  board: esp12e
  name_add_mac_suffix: true

wifi:
  ssid: <ssid>
  password: <password>
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.20.177
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.20.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: <ssid>
    password: <password>

captive_portal:

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: <key>

ota:
  password: <password>

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 9600
  
climate:
  - platform: midea
    name: Office Midea Climate  # Use a unique name.
    period: 1s                  # Optional
    timeout: 2s                 # Optional
    num_attempts: 3             # Optional
    autoconf: true              # Autoconfigure most options.
    beeper: true                # Beep on commands.
    visual:                     
      min_temperature: 16 °C    
      max_temperature: 30 °C    
      temperature_step: 0.5 °C  
    supported_modes:            # Optional. All capabilities in this section may be detected by autoconf.
      - HEAT_COOL
      - COOL
      - HEAT
    supported_swing_modes:
      - VERTICAL                # This capability is always used.

web_server:
  port: 80
```

The problem is, when trying to change the temperature of the unit (which **can** go to 16C) the UI resets back to 17C. The following screenshot shows that the slider has been moved to 16C, but the Target remains at 17C:

[image]

If I then refresh the page, the slider has moved back to the 17C position. 

**Note:** The Air Conditioning unit **does** get set to 16C correctly, but for some reason the integration doesn't accept it.

This then means, if I wanted the target temperature to be 16C, I would need to turn it on and then set it to 16C each time. Not a massive problem, by any means, but I am still interested in whether I've done something wrong. 

### Which version of ESPHome has the issue?

2024.5.5

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A - although this has knock on effects on HA also displaying the wrong temperature, the issue is ESPHome itself.

### What platform are you using?

ESP8266

### Component causing the issue

Climate

### Example YAML snippet

```yaml
Provided above
```

ssieb, 2024-06-25:
The midea component is quite sure that 17 is the minimum.  I don't know why that was done.  Maybe your device is an exception?

simondrake, 2024-06-26:
Isn't the point that the minimum is configurable via `min_temperature`?

ssieb, 2024-06-26:
When the climate component can support whatever range, then yes, you can adjust the visual range.  But this component can't set temperature below 17 or above 30 and forces the visual range to match that.  I'm guessing that's the usual range of the supported devices.

ssieb, 2024-06-26:
It's a limitation of the midea component.  If the underlying library and devices can support lower temperatures, then you could make a PR.

simondrake, 2024-06-26:
Thanks for the help/answers