esphome/issues #5975: Multiple ways to build the same target were specified for: adc.o

p1otr, 2024-06-26:
### The problem

Since few ESPHome versions (or espidf/platformio updates? I didn't change anything relatet do it) I cannot build my ESP32 firmwares anymore (ESP8266 works fine).

It fails with this message:
```
*** Multiple ways to build the same target were specified for: /tmp/esphome/.esphome/build/lampa-salon/.pioenvs/lampa-salon/adc.o  (from ['/home/piotr/.platformio/packages/framework-espidf/components/driver/adc.c'] and from ['/home/piotr/.platformio/packages/framework-espidf/components/driver/esp32/adc.c'])
File "/home/piotr/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 684, in compile_source_files
```

Removing ~/.platformio and ./.esphome and regenerating all needed files doesn't help

### Which version of ESPHome has the issue?

2024.6.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

esp32doit-devkit-v1

### Component causing the issue

adc

### Example YAML snippet

```yaml
esphome:
  name: some_name

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
```

werdnum, 2024-06-27:
I was able to reproduce this in ESPHome dashboard (running on linux-aarch64), but not on my Intel Mac. Perhaps an issue specific to compiling on ARM?

I see https://github.com/esphome/esphome/pull/6330 in the changelog, I wonder if this has anything to do with it?

p1otr, 2024-06-27:
Not ARM specific as I'm on amd64 (on Debian)

p1otr, 2024-06-28:
FTR: error message is generated by platformio's scons-local-4.5.2 in my case

p1otr, 2024-06-28:
The problem is only with esp-idf - removing framework settings (and thus defaulting to platformio/framework-arduinoespressif32 doesn't trigger it)

p1otr, 2024-07-07:
with ESPHome 2024.6.6 / framework-espidf @ 3.40407.240606 (4.4.7) the file with a conflict changed to:
.platformio/packages/framework-espidf/components/efuse/esp32/esp_efuse_fields.c
.platformio/packages/framework-espidf/components/efuse/src/esp_efuse_fields.c

p1otr, 2024-07-23:
It looks like symlinked ~/.platformio is needed to trigger it. I updated the issue with a script to reproduce

GrumpyMeow, 2024-08-10:
I get the same issue here...
The error is about having 2 files "adc.c" files, of which one seems specific for esp32:
```
from ['/var/lib/private/esphome/.platformio/packages/framework-espidf/components/driver/adc.c'] and 
from ['/var/lib/private/esphome/.platformio/packages/framework-espidf/components/driver/esp32/adc.c']
```
I have this error on 2 of my esphome-devices which specificly use audio.

airampg, 2025-01-15:
Hi all, 
I get the same issue, though it is not the adc that it points to:

`*** Multiple ways to build the same target were specified for: /data/build/bed-sensor-airam/.pioenvs/bed-sensor-airam/esp_efuse_fields.c.o  (from ['/data/cache/platformio/packages/framework-espidf/components/efuse/esp32/esp_efuse_fields.c'] and from ['/data/cache/platformio/packages/framework-espidf/components/efuse/src/esp_efuse_fields.c'])
File "/data/cache/platformio/platforms/espressif32/builder/frameworks/espidf.py", line 770, in compile_source_files`

This code, in particular, I've compiled and uploaded to the ESP several times before. I just went to change a minor thing in the code (timings for the ADC to a value I've already used before), and now it fails.

compiling on intel 64 bits...

PKM85, 2025-02-04:
Hi there, 
I had the same issue yesterday after changing just changing the running time of a time based cover in ESP Home. 

But I tried **"Clean Build Files"** in the 3 Dot Menu in the ESP Home Dashboard on that device and I don't know if its just a coincidence, but that solved the problem for me!

Hope it does for you as well...

promufa, 2025-03-19:
> Hi there, I had the same issue yesterday after changing just changing the running time of a time based cover in ESP Home.
>
> But I tried **"Clean Build Files"** in the 3 Dot Menu in the ESP Home [...]

No clue how this happened but I had 3 Esp32's (1x C3, 2x Wroom32) with the same issues when trying to recompile with minimal changes to eg sensor thresholds or BLE scan interval. Cleaning build files resolved the issue for all 3 of them