esphome/issues #5980: Not possible to configure password for S3 Box Voice Assistant in 2024.6.0 and above

rdbahm, 2024-06-27:
### The problem

Following the changes to OTA in ESPHome 2024.6.0, I am no longer able to have an OTA password configured.

I use the S3 Box Voice Assistant firmware. I have adopted the configuration, and have added a handful of customizations to suit my needs. Like all of my ESPHome devices, I set an OTA password to prevent unauthorized firmware uploads on these devices. Following the release of 2026.6.0, it's no longer possible for me to compile firmware - the command instead returns various errors (it seems to have recently changed in 2024.6.3).

The trouble I'm running into is that I don't seem to be able to _override_ the default OTA configuration provided by the S3 Box configuration. I believe that if an ID were specified in the esphome-provided configuration, I would be able to use the "extend" functionality to add my own OTA configuration, sidestepping this issue.

This issue occurs regardless of if I specify an ID in my own OTA configuration, as well as independent of "version" in the OTA config.

### Which version of ESPHome has the issue?

2024.6.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6

### What platform are you using?

ESP32

### Board

S3 Box

### Component causing the issue

ota.esphome

### Example YAML snippet

```yaml
wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
    priority: 10

api:
  encryption:
    key: !secret encryption_psk

ota:
  - platform: esphome
    password: !secret ota_pw

packages:
  esphome.voice-assistant: github://esphome/firmware/voice-assistant/esp32-s3-box-3.yaml@main
esphome:
  name: example
```

### Anything in the logs that might be useful for us?

```txt
Failed config

ota.esphome: [source /data/packages/c46f54c1/voice-assistant/esp32-s3-box-3.yaml:84]
  
  Only one instance of the esphome ota platform is allowed per port. Note that this error may result from OTA specified in packages.
  platform: esphome
  version: 2
  port: 3232
```

ssieb, 2024-06-27:
This will be fixed in 2024.6.4 with https://github.com/esphome/esphome/pull/7001

ssieb, 2024-06-27:
But also, `id`s will be added in the configs soon.

closed 2024-06-27 by ssieb

ssieb, 2024-06-27:
https://github.com/esphome/firmware/pull/215