esphome/issues #5982: `src/esphome/components/binary_sensor/automation.cpp' not found,

comet424, 2024-06-27:
### The problem

esphome wont compile anymore
used to work  now wont compile since last update i guess

### Which version of ESPHome has the issue?

2024.6.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.4

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: esp32-mic-speaker
  friendly_name: esp32-mic-speaker
  on_boot:
     - priority: -100
       then:
         - wait_until: api.connected
         - delay: 1s
         - if:
             condition:
               switch.is_on: use_wake_word
             then:
               - voice_assistant.start_continuous:

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended

# Enable logging
logger:

# Enable Home Assistant API
api:
ota: 
  platform: esphome
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
#  manual_ip:
#    static_ip: 192.168.20.220
#    gateway: 192.168.0.1
#    subnet: 255.255.255.0  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Mic-Speaker"
    password: "9vYvAFzzPjuc"
web_server:

#sensor:
#  - platform: homeassistant
#    id: big_speaker
#    entity_id: media_player.speaker_3_esphome_i2s_media_player

i2s_audio:
  i2s_lrclk_pin: GPIO5    #WS/FS
  i2s_bclk_pin: GPIO21    #SCK

microphone:
  - platform: i2s_audio
    id: mic
    adc_type: external
    i2s_din_pin: GPIO26    #SD/DATA
    pdm: false

#speaker:
#  - platform: i2s_audio
#    id: big_speaker
#    dac_type: external
#    i2s_dout_pin: GPIO25
#    mode: mono

voice_assistant:
  microphone: mic
  use_wake_word: true
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
 # speaker: big_speaker
  id: assist
  on_tts_end:
     - homeassistant.service:
         service: media_player.play_media
         data:
           entity_id: media_player.speaker_1_esphome_i2s_media_player
           media_content_id: !lambda 'return x;'
           media_content_type: music
           announce: "true"

switch:
  - platform: template
    name: Use wake word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - lambda: id(assist).set_use_wake_word(true);
      - if:
          condition:
            not:
              - voice_assistant.is_running
          then:
            - voice_assistant.start_continuous
    on_turn_off:
      - voice_assistant.stop
      - lambda: id(assist).set_use_wake_word(false);
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.6.3
INFO Reading configuration /config/esphome/esp32-mic-speaker.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-mic-speaker (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- ArduinoJson @ 6.18.5
*** [.pioenvs/esp32-mic-speaker/src/esphome/components/binary_sensor/automation.o] Source `src/esphome/components/binary_sensor/automation.cpp' not found, needed by target `.pioenvs/esp32-mic-speaker/src/esphome/components/binary_sensor/automation.o'.
========================== [FAILED] Took 1.54 seconds ==========================
```

ssieb, 2024-06-27:
Try a clean build.

comet424, 2024-06-27:
@ssieb ok thanks thats ;let me compile it... clean build i take it deletes  it all and starts fresh?

ssieb, 2024-06-27:
Yes, this is a common issue with esp-idf.

closed 2024-06-27 by ssieb

comet424, 2024-06-27:
ah ok so if if i have issues just do the clean build then...  i figured if i use just esp32-wrooms  i be ok  faster and better then the 8266 ones  but  learn something new everyday lol