esphome/issues #5983: Strange behavior with DS18B20

deanfourie1, 2024-06-28:
### The problem

When I setup the one_wire component on its own, it finds my DS18B20 sensor and displays the ID,  However, when I setup the DS18B20 device as a sensor, with the address specified, I get the error no sensors found.

With the following code,

```
one_wire:
  - platform: gpio
    pin: GPIO14
```

[image]

And with the following code,

```
one_wire:
  - platform: gpio
    pin: GPIO14

sensor:
  - platform: dallas_temp
    address: 0x950121128170a628
    name: "ServerRoomTemp"
```

[image]

### Which version of ESPHome has the issue?

2024.6.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.4

### What platform are you using?

ESP32

### Board

Kinkony KC868-A8

### Component causing the issue

dallas_temp

deanfourie1, 2024-06-29:
Additionally, from the log.

```
[01:01:38][W][dallas.temp.sensor:139]: 'CoreRackTemp' - Scratch pad checksum invalid!
[01:01:38][W][component:157]: Component dallas_temp.sensor set Warning flag: scratch pad checksum invalid
[01:01:38][D][sensor:094]: 'CoreRackTemp': Sending state nan °C with 1 decimals of accuracy
```

ssieb, 2024-06-29:
That is very strange.  Does removing the sensor definition bring it back?

deanfourie1, 2024-06-29:
Yea, if I remove the dallas_temp sensor, the sensor is found and displays the address, to which I then add to my sensor and then again, no sensors found with the above error.

ssieb, 2024-06-29:
Try adding `setup_priority: -100` to the sensor and see what happens.

deanfourie1, 2024-06-30:
Wow, thank you so much. 

That worked.

ssieb, 2024-06-30:
I think I have an idea of what went wrong.  Try putting `setup_priority: 1000` on the `one_wire:` component instead.  Looks like I forgot to set the priority.

deanfourie1, 2024-06-30:
Yup that works too, I removed from the sensor and added the above to the one_wire component.

But its weird that it works on 3 of the other identical boards just not this one, all on the latest version.

ssieb, 2024-06-30:
ok, even stranger is that the one_wire component actually does have that priority already.  I looked in the wrong file.  I didn't think I would have missed that...  :shrug:

deanfourie1, 2024-07-01:
Yea thats strange, but for now this is resolved for me haha, sensor is working perfectly so im happy for this to be closed.

Thanks again for the support, love your work! Love ESPhome.

closed 2024-07-01 by deanfourie1