esphome/issues #5988: PLEASE stop releasing updates every day.  or every other day.  Schedule non-critical updates at regular intervals!!!!!!

douginoz, 2024-06-29:
### The problem

I'm not alone when I say that I'm tired of seeing near-daily updates popping up in HA for esphome.  Critical bug fixes obviously need to get out quickly, not could you please use a more regulated scheduled release approach?  There's no reason to be constantly releasing new versions every day, or every couple of days.  

I don't mean to belittle the effort or insult the devs.  But please try to use a best-practice approach to software releases.

### Which version of ESPHome has the issue?

all

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

all

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
<null>
```

MallocArray, 2024-06-29:
Respectfully disagree. If things are broken, I don't want to wait for weeks until the next scheduled cadence to get the fix

ssieb, 2024-06-29:
Point releases are only for critical bug fixes.  You don't have to update if you don't want to.

closed 2024-06-29 by ssieb