esphome/issues #5990: DNS settings incorrect / overwritten when container restarts - breaks OTA

andreas-bulling, 2024-06-29:
### The problem

I have had issues with ESPHome for quite some time. Namely that OTA didn't work because `.local` device hostnames couldn't be found/resolved. Some devices worked for some reason, most didn't.

It took me a long time until I figured out what was going on: 

`exec -it addon_5c53de3b_esphome bash` in the Advanced SSH & Web Terminal and `cat /etc/resolv.conf ` showed an automatically generated file with a completely wrong nameserver (no idea where this is coming from and which IP is used here):

```
# Generated by Docker Engine.
# This file can be edited; Docker Engine will not make further changes once it
# has been modified.

nameserver 172.30.32.3
search local.hass.io
options timeout:10

# Based on host file: '/etc/resolv.conf'
# Overrides: [nameservers search options]
```

I can edit the file and fix the nameserver / search entry. But after a restart the container is restarted and the file is reset.

Result: I have to manually fix `resolv.conf` manually every couple of days.

I am pretty certain that this was not necessary last year, i.e. with an older version of ESPHome/HA

### Which version of ESPHome has the issue?

2024.6.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

gabest11, 2024-06-30:
This is more like a docker issue. Try it in host mode or specify your dns settings. https://docs.docker.com/compose/compose-file/05-services/#networks

andreas-bulling, 2024-06-30:
I have no idea how/where to configure host mode/dns settings. afaik the containers are all managed/started by HA...

closed 2024-11-22 by github-actions[bot]