esphome/issues #5991: rp2040_pio_led_strip no longer working with sk6812

lsellens, 2024-06-30:
### The problem

The rp2040_pio_led_strip can no longer control a sk6812 strip after 2024.6.0b1 it works fine with 2024.5.5. I scoped the pin output and I am getting data but my scope is really antiquated so using it to figure out whats going on with timing would be difficult. I switched the config to use ws2812b since the timings are very similar and surprisingly it does control the strip with that setting although there is a lot of flashing and very un-smooth transitions even just from on to off.

### Which version of ESPHome has the issue?

2024.6.0b1 and everything after

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

all

### What platform are you using?

RP2040

### Board

pico w

### Component causing the issue

rp2040_pio_led_strip

### Example YAML snippet

```yaml
light:
  - platform: rp2040_pio_led_strip
    rgb_order: GRB
    chipset: SK6812
    is_rgbw: True
    pin: GPIO15
    num_leds: 20
    pio: 0
    name: "Sk6812 test Light"
    id: led_strip0
```

### Anything in the logs that might be useful for us?

```txt
nothing abnormal in the logs at all. The light never actually turns on.

INFO ESPHome 2024.6.4
INFO Reading configuration /config/esphome/pico3.yaml...
INFO Generating C++ source...
INFO Generating PIO assembly code
INFO Compiling app...
Processing pico3 (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.30702.0 (3.7.2) 
 - tool-pioasm-rp2040-earlephilhower @ 5.100300.230216 (10.3.0) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
Dependency Graph
|-- WiFi @ 1.0.0
|-- LEAmDNS @ 1.2
|-- Updater @ 1.0
|-- noise-c @ 0.1.4
|-- MD5Builder @ 1.0.0
|-- lwIP-Ethernet @ 1
|-- lwIP_CYW43 @ 1
|-- SPI @ 1.0
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
RAM:   [===       ]  28.2% (used 73904 bytes from 262144 bytes)
Flash: [=====     ]  52.5% (used 547904 bytes from 1044480 bytes)
Building .pioenvs/pico3/firmware.bin.signed
========================= [SUCCESS] Took 5.44 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.0.217
INFO Uploading /data/build/pico3/.pioenvs/pico3/firmware.bin (559936 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 9.18 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.0.217 using esphome API
INFO Successfully connected to pico3 @ 192.168.0.217 in 19.423s
INFO Successful handshake with pico3 @ 192.168.0.217 in 0.192s
[14:12:48][I][app:100]: ESPHome version 2024.6.4 compiled on Jun 30 2024, 14:11:08
[14:12:48][C][wifi:599]: WiFi:
[14:12:48][C][wifi:427]:   Local MAC: D8:3A:DD:27:B4:BE
[14:12:48][C][wifi:432]:   SSID: [redacted]
[14:12:48][C][wifi:435]:   IP Address: 192.168.0.217
[14:12:48][C][wifi:438]:   BSSID: [redacted]
[14:12:48][C][wifi:440]:   Hostname: 'pico3'
[14:12:48][C][wifi:442]:   Signal strength: -49 dB ▂▄▆█
[14:12:48][C][wifi:446]:   Channel: 10
[14:12:48][C][wifi:447]:   Subnet: 255.255.255.0
[14:12:48][C][wifi:448]:   Gateway: 192.168.0.1
[14:12:48][C][wifi:449]:   DNS1: 192.168.0.9
[14:12:48][C][wifi:450]:   DNS2: 0.0.0.0
[14:12:48][C][logger:185]: Logger:
[14:12:48][C][logger:186]:   Level: DEBUG
[14:12:48][C][logger:188]:   Log Baud Rate: 115200
[14:12:48][C][logger:189]:   Hardware UART: USB_CDC
[14:12:48][C][logger:193]:   Level for 'component': DEBUG
[14:12:48][C][rp2040_pio_led_strip:176]: RP2040 PIO LED Strip Light Output:
[14:12:48][C][rp2040_pio_led_strip:177]:   Pin: GPIO15
[14:12:48][C][rp2040_pio_led_strip:178]:   Number of LEDs: 20
[14:12:48][C][rp2040_pio_led_strip:179]:   RGBW: YES
[14:12:48][C][rp2040_pio_led_strip:180]:   RGB Order: GRB
[14:12:48][C][rp2040_pio_led_strip:181]:   Max Refresh Rate: 0.000000 Hz
[14:12:48][C][light:103]: Light 'Sk6812 test Light'
[14:12:48][C][light:105]:   Default Transition Length: 1.0s
[14:12:48][C][light:106]:   Gamma Correct: 2.80
[14:12:48][C][shutdown.button:018]: Shutdown Button 'Shutdown'
[14:12:48][C][shutdown.button:018]:   Icon: 'mdi:power'
[14:12:48][C][mdns:115]: mDNS:
[14:12:48][C][mdns:116]:   Hostname: pico3
[14:12:48][C][esphome.ota:073]: Over-The-Air updates:
[14:12:48][C][esphome.ota:074]:   Address: pico3.local:2040
[14:12:48][C][esphome.ota:075]:   Version: 2
[14:12:48][C][esphome.ota:078]:   Password configured
[14:12:48][C][safe_mode:018]: Safe Mode:
[14:12:48][C][safe_mode:019]:   Boot considered successful after 60 seconds
[14:12:48][C][safe_mode:021]:   Invoke after 10 boot attempts
[14:12:48][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[14:12:48][C][api:139]: API Server:
[14:12:48][C][api:140]:   Address: pico3.local:6053
[14:12:48][C][api:142]:   Using noise encryption: YES
[14:12:55][D][light:036]: 'Sk6812 test Light' Setting:
[14:12:55][D][light:041]:   Color mode: White
[14:12:55][D][light:047]:   State: ON
[14:12:55][D][light:085]:   Transition length: 1.0s
[14:12:57][D][light:036]: 'Sk6812 test Light' Setting:
[14:12:58][D][light:047]:   State: OFF
[14:12:58][D][light:085]:   Transition length: 1.0s
[14:13:00][D][light:036]: 'Sk6812 test Light' Setting:
[14:13:00][D][light:041]:   Color mode: RGBW
[14:13:00][D][light:047]:   State: ON
[14:13:00][D][light:055]:   Color brightness: 100%
[14:13:00][D][light:058]:   Red: 100%, Green: 25%, Blue: 14%
[14:13:00][D][light:063]:   White: 0%
[14:13:00][D][light:085]:   Transition length: 1.0s
[14:13:02][D][light:036]: 'Sk6812 test Light' Setting:
[14:13:02][D][light:055]:   Color brightness: 100%
[14:13:02][D][light:058]:   Red: 33%, Green: 31%, Blue: 100%
[14:13:02][D][light:063]:   White: 0%
[14:13:02][D][light:085]:   Transition length: 1.0s
[14:13:03][D][light:036]: 'Sk6812 test Light' Setting:
[14:13:03][D][light:055]:   Color brightness: 100%
[14:13:03][D][light:058]:   Red: 22%, Green: 100%, Blue: 39%
[14:13:03][D][light:063]:   White: 0%
[14:13:03][D][light:085]:   Transition length: 1.0s
[14:13:04][D][light:036]: 'Sk6812 test Light' Setting:
[14:13:04][D][light:047]:   State: OFF
[14:13:04][D][light:085]:   Transition length: 1.0s
```

lsellens, 2024-06-30:
So I dug out my logic analyzer and wrote a decoder for sigrok to decode the sk6812 properly since the ws281x one didn't seem to work right with the sk6812 timing. I cant find a problem. The data being sent to the strip when running 2024.5.5(which does work) and the data I get when running 2024.6.4(does not work) are both practically identical. I can' find any timing issues and have no clue why its not working. Someone mush smarter then me is going to have to look at it. Let me know if anyone is interested I can provide captures from pulseview if you need.

michaeltanner70, 2024-07-01:
Same issue with ESP32-S3 with onboard RGB-LED

jesserockz, 2024-07-03:
@michaeltanner70 That is an entirely different component and set of code and will be unrelated to this issue. Please mae your own issue for that.

lsellens, 2024-11-01:
This is still an issue even with the most recent version. Looking at the code the math adds up the timing should be correct but what the pico produces is not. I dont know enough about pio to figure out what is causing the timing issue but using a logic analyzer and trial by error this will give the correct timing
"SK6812": LEDStripTimings(20, 54, 38, 38),
I havn't tested any of the strip configurations.

pseud0sphere, 2025-01-16:
I tried @lsellens' timing, converted to microseconds using the 57.5us per cycle in the rp2040_pio_led_strip/light.py file, which is (with some rounding):

````
    bit0_high: 0.35us
    bit0_low: 0.94us
    bit1_high: 0.66us
    bit1_low: 0.66us
````

This seems to work for SK6812.

> Ttrial by error this will give the correct timing "SK6812": LEDStripTimings(20, 54, 38, 38)

closed 2025-06-09 by jesserockz