esphome/issues #5993: [capture_response] is an invalid option for [http_request.get]

10MINT, 2024-07-01:
### The problem

I was trying the [get json example from the docs](https://esphome.io/components/http_request.html#get-values-from-a-json-body-response) together with on boot. The compiler doesn't accept the capture_response keyword. Without it, the compiler rightfully complains that there is no body in the lambda

### Which version of ESPHome has the issue?

2024.5.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

http request

### Example YAML snippet

```yaml
esphome:
  name: xxxx
  on_boot:
    priority: 249
    then:
    - http_request.get:
        url: xxxx
        capture_response: true
        on_response:
          then:
            - lambda: |-
                json::parse_json(body, [](JsonObject root) -> bool {
                    ESP_LOGD("on_boot", "The value is: %f", root["value"]);
                    // do stuff
                    return true;
                });
```

### Anything in the logs that might be useful for us?

```txt
esphome: None
  name: xxxx
  on_boot: 
    priority: 249
    then: 
      - http_request.get: 
          url: xxxx

          [capture_response] is an invalid option for [http_request.get]. Did you mean [on_response]?
          capture_response: True
            then:
              - lambda: |-
                  json::parse_json(body, [](JsonObject root) -> bool {
                      ESP_LOGD("on_boot", "The value is: %f", root["value"]);
                      // do stuff
                      return true;
                  });
```

MisterRadish, 2024-07-04:
I think you need to be on version 5.6.0 or higher to use the capture_response option

10MINT, 2024-07-08:
That seems to be correct, at least on 2024.6.6 it's working. Generally, is there any way to find out the version a feature was introduced in? It would be nice to include it in the docs in which version a component or parameter was introduced

guillempages, 2024-08-01:
the http_request component was heavily refactored on the 2024.6 release, and IIRC this was also explicitly mentioned in the breaking changes section of the release notes.

That said, I understand that not everybody is looking at the release notes for every release or the breaking changes done there :-(

I'm not aware of a (meaningful) way of knowing when a feature or parameter was added, besides looking at the documentation repository and searching for merged PRs and trying to find the relevant one 😞
I don't think that it would be feasible to ask every contributor to add the version of the parameter in the docs when they create the docs (it's already difficult enough to actually document the feature). Especially since when creating the PR and the documentation, it is not known when this will actually be reviewed or merged.
And for doing it automatically, I'm by far not an expert in the framweork being used for documentation, but I don't think this is doable out-of-the-box.
In any case this would be a generic featur request then.

Using "latest&greatest" fixes the issue, so I don't think that we need to keep this open, do we? 😉

closed 2024-08-01 by guillempages

10MINT, 2024-08-01:
Some constructive criticism: I agree that most people don't read all of the release notes for every release, but I like that banner you have on the website, which says that there has been a new release since you last visited the site. I haven't seen that anywhere else, that's really nice! As for the documentation: A lot of tools have their own documentation pages for each minor version, e.g. 
- Numpy: https://numpy.org/doc/stable/index.html
  - They even have the version a parameter was introduced in, see e.g. https://numpy.org/doc/stable/reference/generated/numpy.hstack.html#numpy-hstack
- Bootstrap: https://getbootstrap.com/docs/5.3/getting-started/introduction/
- WeeWX: https://www.weewx.com/docs.html
- React: https://17.reactjs.org/

I don't know what framework you're using for the documentation, so I don't know what's possible. 

Btw, I also got the pre 2024.6 version working
```yaml
esphome:
  name: xxxx
  on_boot:
    priority: 249
    then:
    - http_request.get:
        url: xxxx
        verify_ssl: false
        on_response:
          then:
            - lambda: |-
                json::parse_json(id(http_request_id).get_string(), [](JsonObject root) -> bool {
                    ESP_LOGD("on_boot", "The value is: %f", root["value"]);
                    // do stuff
                    return true;
                });
```

Both configs need the https://esphome.io/components/http_request.html component

guillempages, 2024-08-01:
@nagyrobi  You are by far more experienced on the documentation framework; what do you think about the comments above; to somehow (automatically) make it visible when something was added? Would this be feasible?