esphome/issues #5996: Sprinkler pause does not close valve after shutdown command

goedh452, 2024-07-01:
### The problem

When calling sprinkler.pause the valves are not always closed. This consistently happens in the following sequence:

1. Start sprinkler with sprinkler.start_full_cycle
2. Pause sprinkler with sprinkler.pause
3. The sprinkler is paused and the valve is closed
4. Stop full cycle with sprinkler.shutdown
5. Start sprinkler with sprinkler.start_full_cycle
6. Pause sprinkler with sprinkler.pause
7. The program is not paused and the valve is not closed

This only happens when using the services and not when the switch of the controller itself is used.

I'm using this because I created a bit more logic for controller my sprinkler system. For narrowing down the issue, I created a set of buttons.

The log shows that the second time Pause is pressed, the button press is registered, but does not have any effect on the controller.

### Which version of ESPHome has the issue?

2024.6.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.6

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

sprinkler

### Example YAML snippet

```yaml
### BUTTONS ###
button:
  - platform: template
    id: sprinkler_start
    name: "Start"
    on_press:
      then:
        - sprinkler.start_full_cycle: ${controller_name}

  - platform: template
    id: sprinkler_pauzeer
    name: "Pause"
    on_press:
      then:
        - sprinkler.pause: ${controller_name}

  - platform: template
    id: sprinkler_hervat
    name: "Resume"
    on_press:
      then:
        - sprinkler.resume: ${controller_name}

  - platform: template
    id: sprinkler_shutdown
    name: "Shutdown"
    on_press:
      then:
        - sprinkler.shutdown: ${controller_name}
```

### Anything in the logs that might be useful for us?

```txt
[20:23:12][D][button:010]: 'Start' Pressed.
[20:23:12][D][sprinkler:1427]: CYCLE is starting valve 0 for 15 seconds, cycle 1 of 1
[20:23:12][D][switch:012]: 'Valve zone 1' Turning ON.
[20:23:12][D][switch:055]: 'Valve zone 1': Sending state ON
[20:23:12][D][switch:055]: 'Irrigatie achtertuin': Sending state ON
[20:23:12][D][switch:055]: 'Zone 1: Gras midden': Sending state ON
[20:23:15][D][button:010]: 'Pause' Pressed.
[20:23:15][D][switch:016]: 'Valve zone 1' Turning OFF.
[20:23:15][D][switch:055]: 'Valve zone 1': Sending state OFF
[20:23:15][D][sprinkler:959]: Paused valve 0 with 12 seconds remaining
[20:23:17][D][button:010]: 'Shutdown' Pressed.
[20:23:19][D][sprinkler:1449]: All valves stopped, including pumps
[20:23:21][D][button:010]: 'Start' Pressed.
[20:23:21][D][sprinkler:1427]: CYCLE is starting valve 0 for 15 seconds, cycle 1 of 1
[20:23:21][D][switch:012]: 'Valve zone 1' Turning ON.
[20:23:21][D][switch:055]: 'Valve zone 1': Sending state ON
[20:23:21][D][switch:055]: 'Irrigatie achtertuin': Sending state ON
[20:23:21][D][switch:055]: 'Zone 1: Gras midden': Sending state ON
[20:23:23][D][button:010]: 'Pause' Pressed. --> CONTROLLER IS NOT PAUSED
[20:23:26][D][button:010]: 'Shutdown' Pressed.
[20:23:28][D][sprinkler:1449]: All valves stopped, including pumps
```

kbx81, 2024-08-16:
It would be helpful to see your `sprinkler` config, as well.

goedh452, 2024-08-18:
Sure, here is the config of the controller:

```
### SUBSTITUTIONS ###
substitutions:
  ### Controller en zones
  controller_name: irrigatie_achtertuin
  zone_1_valve_id: valve_zone_1
  zone_1_valve_name: Valve zone 1
  zone_1_name: Gras midden
  zone_2_valve_id: valve_zone_2
  zone_2_valve_name: Valve zone 2
  zone_2_name: Gras achter
  zone_3_valve_id: valve_zone_3
  zone_3_valve_name:  Valve zone 3
  zone_3_name: Plantenbakken
  zone_4_valve_id: valve_zone_4
  zone_4_valve_name: Valve zone 4
  zone_4_name: Border

### SPRINKLER CONTROLLER ###
sprinkler:
  - id: ${controller_name}
    main_switch: "Irrigatie achtertuin"
    auto_advance_switch: "Auto Advance"
    multiplier_number: "Multiplier"
    valve_overlap: 1s
    valves:
      - valve_switch: "Zone 1: ${zone_1_name}"
        enable_switch: "Cycle zone 1: ${zone_1_name}"
        run_duration_number: "Timer: ${zone_1_name}"
        valve_switch_id: ${zone_1_valve_id}
      - valve_switch: "Zone 2: ${zone_2_name}"
        enable_switch: "Cycle zone 2: ${zone_2_name}"
        run_duration_number: "Timer: ${zone_2_name}"
        valve_switch_id: ${zone_2_valve_id}
      - valve_switch: "Zone 3: ${zone_3_name}"
        enable_switch: "Cycle zone 3: ${zone_3_name}"
        run_duration_number: "Timer: ${zone_3_name}"
        valve_switch_id: ${zone_3_valve_id}
      - valve_switch: "Zone 4: ${zone_4_name}"
        enable_switch: "Cycle zone 4: ${zone_4_name}"
        run_duration_number: "Timer: ${zone_4_name}"
        valve_switch_id: ${zone_4_valve_id}
```