esphome/issues #5998: after YAML code is edited, old binary is downloaded

I-hate-2FA, 2024-07-02:
### The problem

when compiling for the first time everything works fine, but when you change the yaml code and want to compile again the previous old binary file is downloaded, i have same issue for all my devices it is not related to custom sensor, and after i left esphome for few hours i can download new binary normally again, both download manually and plug into this pc have issue, i cannot test ota because i use without home assistant, device is in different subnet
i tried clean build file, restart esphome addon, both didnt work, only way to fix it is just left it for a while
idk what could be causing this, if you can help i am happy to provide further information
it had been a awsome project

### Which version of ESPHome has the issue?

2024.6.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.4

### What platform are you using?

ESP32

### Board

does not matter (esp32dev)

### Component causing the issue

compiler

### Example YAML snippet

```yaml
esphome:
  name: esphome-trash-can
  friendly_name: esphome-trash-can
  includes:
    - ADC_IR.h

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 0s  

ota:
  - platform: esphome

wifi:
  ssid: ASUS
  password: 0492733932

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Trash-Can"
  reboot_timeout: 0s

esp32_improv:
  authorizer: none

web_server:
  port: 80
  version: 2
  ota: true   
  local: true 
   
sensor:   
  - platform: custom
    lambda: |-
      auto myiradc = new adcir();
      App.register_component(myiradc);
      return {myiradc};
    sensors:
      name: "ir sensor"
      id: "analogir"
      unit_of_measurement: "a"
      filters:
        - quantile:
            window_size: 5
            send_every: 1
            send_first_at: 1
            quantile: .8
      internal: True      

binary_sensor:
  - platform: template
    name: "binary IR"
    id: "binaryir"
    lambda: |-
      if (id(analogir).state > 30) {
        return true;
      } else {
        return false;
      }          
    on_press:
      then:
        - output.turn_on: ledg
    on_release:
      then:
        - output.turn_off: ledg    
  - platform: copy
    source_id: "binaryir"
    name: "timer"
    id: "timer"
    filters:
      - delayed_off: 5s 
    on_press:
      then:
        - output.turn_on: ledr
    on_release:
      then:
        - output.turn_off: ledr     

output:
  - platform: gpio
    pin: GPIO13
    id: ledg
  - platform: gpio
    pin: GPIO15
    id: ledr
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.6.4
INFO Reading configuration /config/esphome/esphome-trash-can.yaml...
WARNING GPIO15 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-trash-can (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
RAM:   [==        ]  16.1% (used 52760 bytes from 327680 bytes)
Flash: [========= ]  86.8% (used 1592513 bytes from 1835008 bytes)
========================= [SUCCESS] Took 14.26 seconds =========================
INFO Successfully compiled program.
```

### Additional information

[image]
file size is always same as old one

Spartan-II-117, 2024-07-24:
This seems to be a browser cache issue. Are you connected through a proxy?
Could you try clearing browser cache between downloads?

I-hate-2FA, 2024-07-24:
i tried everything, i am connected through cloudflare

Spartan-II-117, 2024-07-24:
Ok, it sounds like cloudflare is coaching your file.

I-hate-2FA, 2024-08-02:
i reopened the issue, because i purged all cache in cloudflare but same problem is still there