esphome/issues #6000: Remote_transmitter not working?

deanfourie1, 2024-07-03:
### The problem

I cannot get the remote_transmitter integration to work.

When attempting to send a code with a switch,

```
switch:
  - platform: template
    name: Raw Code Power Button
    turn_on_action:
      - remote_transmitter.transmit_raw:
          code: [4088, -1542, 1019, -510, 513, -1019, 510, -509, 511, -510, 1020,
                 -1020, 1022, -1019, 510, -509, 511, -510, 511, -509, 511, -510,
                 1020, -1019, 510, -511, 1020, -510, 512, -508, 510, -1020, 1022]
```

I would expect to see "Sending RF code" or something in the logs, however, I simply just see

```
[19:15:01][D][switch:012]: 'Raw Code Power Button' Turning ON.
[19:15:03][D][switch:016]: 'Raw Code Power Button' Turning OFF.
[19:15:04][D][switch:012]: 'Raw Code Power Button' Turning ON.
[19:15:05][D][switch:016]: 'Raw Code Power Button' Turning OFF.
```

Also, checked my received and nothing is received.

### Which version of ESPHome has the issue?

2024.6.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.4

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

remote_transmitter

### Example YAML snippet

```yaml
remote_transmitter:
  pin: 27
  carrier_duty_percent: 100%

switch:
  - platform: template
    name: Raw Code Power Button
    turn_on_action:
      - remote_transmitter.transmit_raw:
          code: [4088, -1542, 1019, -510, 513, -1019, 510, -509, 511, -510, 1020,
                 -1020, 1022, -1019, 510, -509, 511, -510, 511, -509, 511, -510,
                 1020, -1019, 510, -511, 1020, -510, 512, -508, 510, -1020, 1022]
```

### Additional information

Hardware is the basic ASK 433 transmitter board, 3 pin VCC, GND and DATA.

https://randomnerdtutorials.com/rf-433mhz-transmitter-receiver-module-with-arduino/

ssieb, 2024-07-03:
There is no logging for transmits unless something fails.

kayooneteu, 2024-08-22:
I have the same bug. remote _transmitter stopped working after upgrade. For me last working version was compiled with 2024.6.6 .
My Yaml
```yaml
esp8266:
  board: esp01_1m
remote_transmitter:
  pin: GPIO4
  carrier_duty_percent: 100%
button:
  - platform: template
    id: roleta_all_up_id
    name: "0.1 all"
    icon: "mdi:transfer-up"
    internal: true
    on_press:
    - remote_transmitter.transmit_dooya:
        id: 0x00C9DFD0
        channel: 239
        button: 14
        check: 14
  - platform: template
    id: roleta_all_stop_id
    name: "0.2 all"
    icon: "mdi:stop"
    internal: true
    on_press:
    - remote_transmitter.transmit_dooya:
        id: 0x00C9DFD0
        channel: 239
        button: 10
        check: 10
...
```
With 2024.7.x and 2024.8.0 it's not working. No errors on output (log) from esp device connected to Home Assistant .
With 2024.8.0 I see warning during compilation/installation:
[image]

randybb, 2024-08-22:
@kayooneteu that warning is totally unrelated https://github.com/esphome/issues/issues/6179