esphome/issues #6001: esp32c6 (arduino framework): Error: This board doesn't support arduino framework!

muzzy124, 2024-07-03:
### The problem

I'm trying to compile a midea climate dongle based on esp32c6 devboard.
But it keeps showing me:

INFO ESPHome 2024.6.4
INFO Reading configuration /config/esp32c6-midea.yaml...
WARNING This board is unknown. Make sure the chosen chip component is correct.
WARNING This board is unknown. Make sure the chosen chip component is correct.
INFO Generating C++ source...
INFO Compiling app...
Processing esp32c6-samsung-ac (board: esp32-c6-devkitc-1; framework: arduino; platform: platformio/espressif32@6.7.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C6 160MHz, 512KB RAM, 8MB Flash
 - framework-arduinoespressif32 @ 3.0.2+sha.65204d9 
 - framework-arduinoespressif32-libs @ 5.1.0+sha.bd2b9390ef 
 - toolchain-riscv32-esp @ 13.2.0+20230928
Error: This board doesn't support arduino framework!
========================== [FAILED] Took 0.43 seconds ==========================

### Which version of ESPHome has the issue?

2024.6.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

esp32c6

### Example YAML snippet

```yaml
esphome:
  name: esp32c6-samsung-ac
  friendly_name: esp32c6-samsung-ac
  platformio_options:
    platform_packages:
      - platformio/toolchain-riscv32-esp @ 13.2.0+20230928 
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.2
      - framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.2/esp32-arduino-libs-3.0.2.zip

esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  framework:
    type: arduino
    platform_version: 6.7.0

# Enable logging
logger:
    # level: DEBUG
    # baud_rate: 0

# Enable Home Assistant API
api:
***

ota:
***

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

# UART0 configuration for AUX air conditioner communication
uart:
  tx_pin: GPIO16
  rx_pin: GPIO17
  baud_rate: 9600

climate:
  - platform: midea
    name: Samsung AC bedroom         # Use a unique name.
```

Breefield, 2024-07-10:
I am also experiencing this with the `esp32-c6-devkitc-1`, my platformio.ini looks like:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:adafruit_feather_m0]
platform = atmelsam
board = adafruit_feather_m0
framework = arduino

[env:esp32-c6-devkitc-1]
platform = espressif32
board = esp32-c6-devkitc-1
framework = arduino
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git

```

ssieb, 2024-07-10:
The C6 is not supported yet, so don't file issues about it.
That's a message from platformio that the board definition doesn't indicate support for Arduino.  There's nothing that esphome can do about that.   Maybe try picking a different board name.

closed 2024-07-10 by ssieb

Breefield, 2024-07-10:
Oh gosh, I'm so sorry—I thought I was in the repo: https://github.com/platformio/platform-espressif32
Yes, this is definitely not ya'lls responsibility. Thanks for pruning the issues :)