esphome/issues #6003: Senville ac gets stuck in dry mode. No other modes work

jonatino, 2024-07-04:
### The problem

When using esphome midea everything normally works fine. But for me there is a bug if you set to dry mode and leave it on for a few minutes then try to change it to any other mode, it won’t work. From that point on you can only change it to dry mode or off. I’ve tried restarting, even cutting power to the ac completely and still would only go dry or off. The only way I could fix it was by using the remote to set to cool mode. Once I set to cool mode with remote, I could control the unit as normal. It happens every time for me. I have a Senville unit which is just a rebranded midea ac.

please help fix this issue. Thank you! 🙏 

### Which version of ESPHome has the issue?

2022.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7

### What platform are you using?

ESP8266

### Board

https://smartlight.me/smart-home-devices/wifi-devices/wifi-dongle-air-conditioners-midea-idea-electrolux-for-home-assistant

### Anything in the logs that might be useful for us?

```txt
I can see the commands for different modes coming in but the ac just seems to ignore them. No errors or odd logs.
```

### Additional information

Not sure what the platform is, what I linked my device.

wroadd, 2024-07-18:
I can confirm the same on MDV and Midea devices , which have the same dongle.
No error message.

jonatino, 2024-11-18:
Not stale.... still a bug

jonatino, 2025-05-15:
I've figured out how to reproduce it. I noticed if the unit is in COOL mode with **fan mode AUTO**, I can switch to Dry mode and back to other modes just fine. It breaks when the unit is in non dry mode with manual fan speed set (eg cool mode with fan set to **high**). I think this breaks it because Dry mode does not allow for fan speeds to be configured. 

To reproduce:
1. Set AC to **cool** mode, set fan mode to **high**
2. Switch AC mode to **dry** mode
3. Once it switches, try to change it back to **cool** mode (or any other mode)

You wont be able to change the mode as outlined in the original issue.

Video of reproduction:
https://github.com/user-attachments/assets/6c28a8a2-27a3-4f2a-be80-26e14c35f3a7

Video proof it works fine with **cool** mode set to fan mode **auto**:
https://github.com/user-attachments/assets/7ccba724-cb30-4077-850e-dbccb1bbe18d

jonatino, 2025-05-15:
@jesserockz @dudanov given my last comment do you guys have any idea as to whether this repo has the bug or if its the underlying uart control repo here https://github.com/dudanov/MideaUART is the one with the issue?

bsell93, 2025-06-20:
I’m experiencing this as well using esp8266 and mrcool mini split. Any update here? The reproduction is accurate.

dkowis, 2026-10-06:
So I wonder if DRY mode should always set the fan to AUTO, so that you can leave the mode, and possibly disallow fan mode changes. 

I suspect that's the reason, yeah. 

So I tested that. I can move in and out of DRY mode with a Mr Cool Minisplit, as long as I set the fan speed to AUTO first. 

I think a working patch would be that, just like HEAT/COOL, the fan needs to be set to AUTO at the same time, when turned to DRY.

Yes, the behavior @rodrigojfernandez fixed here: https://github.com/midea-lan/midea-local/pull/422/changes is exactly what we need in the ESP Home land. Force the FAN to AUTO when leaving DRY mode.

cybervulture, 2026-11-15:
I got a workaround for this, this issue tracker seems to be deprecated so I doubt we'll get an actual fix but this seems to be working right now.

Using the ESPHome Climate module [climate.on_control Trigger](https://esphome.io/components/climate/#climate-on_control-trigger) I added an automation that checks if the mode setting is not "DRY", basically did you just change modes from "DRY" to something else. If the so then set the fan to "AUTO". You could probably add a check to first see if the old mode is dry but I'm lazy and this works, I really don't care if changing between the other modes forces the fan back to auto.

Screenshot from ESPHome Builder in Home Assistant:
[image]

The yaml for my HVAC unit, see the `on_control` block at the bottom:
```yaml
climate:
  - platform: midea
    name: ${friendly_name}                            # The value of "friendly_name" will be used here
    id: midea_climate                           
    period: 1s                                        # Optional
    timeout: 2s                                       # Optional
    num_attempts: 3                                   # Optional
    autoconf: false                                   # Autoconfigure most options.
    beeper: true                                      # Beep on commands.
    visual:                                           # Optional. Example of visual settings override.
      min_temperature: "${visual_min_temperature}"    # min: 17
      max_temperature: "${visual_max_temperature}"    # max: 30
      temperature_step: "${visual_temperature_step}"  # min: 0.5
    supported_modes:                                  # All capabilities in this section detected by autoconf.
      - FAN_ONLY                                      # This capability is always used.
      - HEAT_COOL
      - COOL
      - HEAT
      - DRY
    custom_fan_modes:
      - SILENT
      - TURBO
    supported_presets:          # All capabilities in this section detected by autoconf.
      - ECO
      - BOOST
      - SLEEP                   # This capability is always used.
    custom_presets:             # All capabilities in this section detected by autoconf.
      - FREEZE_PROTECTION
    supported_swing_modes:
      - VERTICAL                # This capability is always used.
      # - HORIZONTAL
      # - BOTH
    outdoor_temperature:        # Optional. Outdoor temperature sensor (may display incorrect values after long inactivity).
      name: ${friendly_name} Outdoor Temperature

    # THIS IS THE FIX
    on_control:
      then:
        - logger.log: Dry Mode Override
        - lambda: |
            // Automatically force fan mode to AUTO when any mode other than DRY is requested
            if (x.get_mode() != CLIMATE_MODE_DRY) {
              x.set_fan_mode(CLIMATE_FAN_AUTO);
            }
```