esphome/issues #6004: Initial value for global float variable not working

KlausHans, 2024-07-04:
### The problem

I have a global float variable i want to initialize with 0 and restore it:
```
globals:
  - id: g_water_daily
    type: float
    restore_value: true
    initial_value: "0.0"
```
On boot, i set the restored value to a template sensor:
```
esphome:
  ...
  on_boot:
    then:
      - sensor.template.publish:  # restore today
          id: today
          state: !lambda 'return id(g_water_daily);'  
sensor:
  - platform: template
    name: "Heute"
    id: today
    icon: "mdi:water-plus"
    unit_of_measurement: "L"
    device_class: "water"
```
This doesn't work. The template sensor is always unknown after startup:
[image]

I also have a global int-variable that restores pulse counts, it works flawlessly. That's why i think the problem is the float type.

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.1

### What platform are you using?

ESP32

### Board

Generic Esp32

### Component causing the issue

globals

### Example YAML snippet

```yaml
esphome:
  name: garten-wasserzahler
  friendly_name: Garten Wasserzähler
  on_boot:
    then:
      - pulse_meter.set_total_pulses: # restore pulses
          id: pulse_meter_id
          value: !lambda 'return id(g_total_pulses);'
      - sensor.template.publish:  # restore today
          id: today
          state: !lambda 'return id(g_water_daily);'  

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  - platform: esphome
    password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Garten-Wasserzahler"

captive_portal:

web_server:
  port: 80

preferences:
  flash_write_interval: 3h

globals:
  - id: g_total_pulses
    type: int
    restore_value: true
    initial_value: "1125000"
  - id: g_water_daily
    type: float
    restore_value: true
    initial_value: "0.0"

sensor:
  - platform: pulse_meter
    id: pulse_meter_id
    internal_filter: 100ms
    pin:
      number: 13 # D13
      inverted: true
      mode:
        input: true
        pullup: true # Anschluss zwischen Pin und GND
    name: "Durchfluss"
    icon: "mdi:water"
    unit_of_measurement: "L/h"
    state_class: "measurement"
    device_class: "volume_flow_rate"
    timeout: 20s
    accuracy_decimals: 0
    filters:
      - multiply: 30
    total:
      name: "Gesamt"
      id: gesamt_id
      icon: "mdi:water-circle"
      unit_of_measurement: "L"
      state_class: "total_increasing"
      device_class: "water"
      accuracy_decimals: 1
      filters:
        - multiply: 0.5
      on_raw_value: # ignore filters
        then:
          - globals.set:
              id: g_total_pulses
              value: !lambda 'return id(gesamt_id).raw_state;'
          - globals.set: # increase global today by 0.5 L per pulse
              id: g_water_daily
              value: !lambda 'return id(today).state + 0.5;'
          - sensor.template.publish: # increase today by 0.5 L per pulse
              id: today
              state: !lambda 'return id(today).state + 0.5;'
  - platform: template
    name: "Heute"
    id: today
    icon: "mdi:water-plus"
    unit_of_measurement: "L"
    device_class: "water"

time:
  - platform: homeassistant
    id: homeassistant_time
    on_time:
      # Midnight
      - seconds: 0
        minutes: 0
        hours: 0
        then:
          - sensor.template.publish: # reset today
              id: today
              state: 0
          - globals.set: # reset global today
              id: g_water_daily
              value: "0.0"
```

uli-rpi, 2024-07-04:
try "Factory Reset Button"

KlausHans, 2024-07-04:
Thank you for the suggestion, nice idea.

It didn't changed anything, but in the log there was an error while resetting:
```
[21:07:31][I][factory_reset.button:013]: Resetting to factory defaults...
[21:07:31][D][esp32.preferences:176]: Cleaning up preferences in flash...
[21:07:31][I][app:132]: Rebooting safely...
[21:07:31][D][esp32.preferences:114]: Saving 1 preferences to flash...
[21:07:31][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 0 written, 1 failed
[21:07:31][E][esp32.preferences:146]: Error saving 1 preferences to flash. Last error=ESP_ERR_NVS_INVALID_HANDLE for key=233825507
```
Is this a clue where the problem is?

uli-rpi, 2024-07-04:
Yes I had the same problem, and using factory reset button fixed it. Please refer to the "note" in the documentation

KlausHans, 2024-07-04:
I did the factory reset at least two times. As i said, it didn't changes anything regarding the problem i posted. The template sensor, that should be initialized by the global float variable, that should be initialized with 0, is still unknown (int variable init still works).
[image]

I don't understand your hint to the note. There is nothing of importance in there for me, it only warns for lost data.

ssieb, 2024-07-04:
If you factory reset, it can't save the preference at reboot because you've deleted it.  You can ignore that error.
Watch the serial logs from boot to see what's happening with the template sensor.

uli-rpi, 2024-07-04:
Sorry I was imprecise, I only did the factory reset once manually, after that the problem was solved for me

KlausHans, 2024-07-04:
> If you factory reset, it can't save the preference at reboot because you've deleted it. You can ignore that error. Watch the serial logs from boot to see what's happening with the template sensor.

What exactly do you mean with serial log? I've added a restart button and this is the log via the esphome ui:
```
[22:54:44][D][button:010]: 'Restart' Pressed.
[22:54:44][I][restart.button:012]: Restarting device...
[22:54:44][I][app:132]: Rebooting safely...
[22:54:44][D][esp32.preferences:114]: Saving 1 preferences to flash...
[22:54:44][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
INFO Processing expected disconnect from ESPHome API for garten-wasserzahler @ 192.168.66.123
WARNING Disconnected from API
INFO Successfully connected to garten-wasserzahler @ 192.168.66.123 in 1.044s
INFO Successful handshake with garten-wasserzahler @ 192.168.66.123 in 0.177s
[22:54:51][C][restart.button:017]: Restart Button 'Restart'
[22:54:51][C][pulse_meter:108]: Pulse Meter 'Durchfluss'
[22:54:51][C][pulse_meter:108]:   Device Class: 'volume_flow_rate'
[22:54:51][C][pulse_meter:108]:   State Class: 'measurement'
[22:54:51][C][pulse_meter:108]:   Unit of Measurement: 'L/h'
[22:54:51][C][pulse_meter:108]:   Accuracy Decimals: 0
[22:54:51][C][pulse_meter:108]:   Icon: 'mdi:water'
[22:54:51][C][pulse_meter:109]:   Pin: GPIO13
[22:54:51][C][pulse_meter:111]:   Filtering rising edges less than 100000 µs apart
[22:54:51][C][pulse_meter:116]:   Assuming 0 pulses/min after not receiving a pulse for 20s
[22:54:51][C][homeassistant.time:010]: Home Assistant Time:
[22:54:51][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[22:54:51][C][captive_portal:088]: Captive Portal:
[22:54:51][C][web_server:173]: Web Server:
[22:54:51][C][web_server:174]:   Address: garten-wasserzahler.local:80
[22:54:51][C][mdns:115]: mDNS:
[22:54:51][C][mdns:116]:   Hostname: garten-wasserzahler
[22:54:51][C][esphome.ota:073]: Over-The-Air updates:
[22:54:51][C][esphome.ota:074]:   Address: garten-wasserzahler.local:3232
[22:54:51][C][esphome.ota:075]:   Version: 2
[22:54:51][C][esphome.ota:078]:   Password configured
[22:54:51][C][safe_mode:018]: Safe Mode:
[22:54:51][C][safe_mode:020]:   Boot considered successful after 60 seconds
[22:54:51][C][safe_mode:021]:   Invoke after 10 boot attempts
[22:54:51][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[22:54:51][C][api:139]: API Server:
[22:54:51][C][api:140]:   Address: garten-wasserzahler.local:6053
[22:54:51][C][api:142]:   Using noise encryption: YES
[22:55:05][D][pulse_meter:095]: No pulse detected for 20s, assuming 0 pulses/min
[22:55:05][D][sensor:094]: 'Durchfluss': Sending state 0.00000 L/h with 0 decimals of accuracy
[22:55:45][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[22:55:45][D][esp32.preferences:114]: Saving 1 preferences to flash...
[22:55:45][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
```
Is that the right log? I don't see anything helpful.
Or do you mean to connect via usb and check with a serial connection/terminal?

ssieb, 2024-07-04:
> Or do you mean to connect via usb and check with a serial connection/terminal?

yes

KlausHans, 2024-07-11:
Hm, i don't know why, but it works now as expected. As far as i am aware i didn't changed anything in the yaml. Strange.
[image]

KlausHans, 2024-07-11:
Eh, i was to fast. After reset of the device the value is not restored.
[image]
Will try usb serial if i have the time for it.