esphome/issues #6005: Digest authentication using Chrome

geotibi, 2024-07-04:
### The problem

When I'm trying to connect to my ESP device using Google Chrome I'm getting "HTTP ERROR 401" without requesting me any credentials. This is not happening in Firefox where the pop up for credential appear and after I insert them, everything is ok. After research I found out that the issue is inside Chrome because it does not support Digest authentication by default. If I'm editing the registry (basic,ntlm,negotiate) add add this Digest authentication then everything is working. The issue is that I'm using a business laptop and my company overwrite the registry from time to time so the configuration is lost. Is it any posibility to change the authentication methond for web server in ESPHome?

[image]

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

web_server

### Example YAML snippet

```yaml
web_server:
  port: 80
  auth:
    username: !secret web_server_username
    password: !secret web_server_password
  version: 3
```

geotibi, 2024-11-03:
Nobody is facing this issue?

JAKAMI99, 2025-05-09:
Chrome implemented Digest in the meantime. Can be closed