esphome/issues #6007: as3935 does not handle div_ratio correctly. Type

illigtr, 2024-07-04:
### The problem

Although the GUI verifies the div_ratio to be one of these values: 0,32,64,128... using 32 will never cause the as3935 register to be updated. This is because the code in both dev branch and release branch have this typo, see case 22, which should be case 32:

```yaml

void AS3935Component::write_div_ratio(uint8_t div_ratio) {
  ESP_LOGV(TAG, "Setting div ratio to %d", div_ratio);
  switch (div_ratio) {
    case 16:
      this->write_register(INT_MASK_ANT, ((1 << 7) | (1 << 6)), 0, 6);
      break;
    case 22:
      this->write_register(INT_MASK_ANT, ((1 << 7) | (1 << 6)), 1, 6);
      break;
    case 64:
      this->write_register(INT_MASK_ANT, ((1 << 7) | (1 << 6)), 1, 7);
      break;
    case 128:
      this->write_register(INT_MASK_ANT, ((1 << 7) | (1 << 6)), 3, 6);
      break;
    default:
      return;
  }
}
````

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.0

### What platform are you using?

ESP32

### Board

esp32-cam

### Component causing the issue

as3935

illigtr, 2024-07-05:
Additionally, if you look at case 64, I believe the line should read this instead:
```YAML
this->write_register(INT_MASK_ANT, ((1 << 7) | (1 << 6)), 2, 6);
```

DM2NT, 2024-10-12:
Hello, is there any way to fix this?

I've been sitting here for days trying to get the AS3935 to work. When calibrating the chips, I only get 28kHz from the IRQ pin and not 500kHz. It also makes no difference whether I set div_ratio to 16 or 32.