esphome/issues #6013: [samples] is an invalid option for [sensor.adc].

Ventosinos, 2024-07-05:
### The problem

Hi everybody.

I think that I found a bug on the ESP-S2. I know that saying that can sound a little pretentious for my side, but I have no other explanation. If I am wrong, please tell me so I can learn. Thank you so much.

# Problem:

On an ESP-S2, I can add one adc component (as shown here https://esphome.io/components/sensor/adc.html). But when I try to set up the "samples" configuration variable, ESPhome tells me that [samples] is an invalid option for [sensor.adc].

I am using an ESP-S2 mini, the one with the purple board. I set the board type as a "lolin_s2_mini"
[image]

### Which version of ESPHome has the issue?

2024.3.2 

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.4.3

### What platform are you using?

ESP32

### Board

lolin_s2_mini

### Component causing the issue

adc

### Example YAML snippet

```yaml
esphome:
  name: nevera
  friendly_name: nevera

esp32:
  board: lolin_s2_mini
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "-"

ota:
  password: "-"

wifi:
  networks:
  - ssid: !secret wifi_ssid2
    password: !secret wifi_password2
  manual_ip:
    static_ip: 192.168.1.19
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
    dns2: 8.8.4.4
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Nevera Fallback Hotspot"
    password: "-"

captive_portal:
    
web_server:
  port: 80

text_sensor:
  - platform: wifi_info
    ip_address:
      name: Direccion IP
    ssid:
      name: Nome da rede
    scan_results:
      name: Redes Dispoñibles

sensor:
  - platform: adc
    pin: GPIO7
    name: "Tension ADC1"
    attenuation: 11db
    id: ADC1
    update_interval: 1s
    samples: 10
```

### Anything in the logs that might be useful for us?

```txt
I don't have a log because I can't compile
```

### Additional information

[image]

Thank you all for your time. You are doing a great job with ESPhome!
(And sorry if I did something wrong 😥)

ssieb, 2024-07-06:
That option was added in the latest release and you're using a much older one.

closed 2024-07-06 by ssieb

Ventosinos, 2024-07-06:
Thank you @ssieb

I updated to 2024.6.6 and now it works perfectly

Sorry for bothering