esphome/issues #6017: Fail to boot with dual UART on S2 with esp-idf

swifty99, 2024-07-06:
### The problem

Since around 2024.04 two configured UARTs on a S2 Board will fail to boot.

It was working OK until 2024.03.
It is working with Arduino IDE
If either of the two UARTs is removed everything is fine

I really would like to stay at IDF as it did feel a lot more responsive on Home Assistant.

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

not related

### What platform are you using?

ESP32-IDF

### Board

S2-mini

### Component causing the issue

UART

### Example YAML snippet

```yaml
esphome:
  name: nouart

  on_boot:
    priority: 600
    then:
      - output.set_level:
          id: status_led
          level: 5%

external_components:
  - source: github://KinDR007/VictronMPPT-ESPHOME@main
  - source: github://syssi/esphome-jbd-bms@main

esp32:
  board: lolin_s2_mini  
  framework:
    #type: esp-idf
    type: arduino

wifi:
  networks:
  - ssid: "Wifier"
    password: !secret wifi_pw_home 

logger:
  level: DEBUG

api:

ota:
  platform: esphome

uart:
  - id: uart_jbd
    baud_rate: 9600
    tx_pin: GPIO33
    rx_pin: GPIO21

  - id: uart_victron
    baud_rate: 19200
    rx_pin: GPIO18 

victron:
  uart_id: uart_victron
  id: victron0

jbd_bms:
  - id: bms0
    uart_id: uart_jbd

output:
  - platform: slow_pwm
    pin: GPIO15
    id: status_led
    period: 2s
```

### Anything in the logs that might be useful for us?

```txt
unfortunately no. At least I found no way wit USB-C to grab the lines right at boot. After boot nothing ist transmitted on USBC, even if configured like:
     sdkconfig_options:
       CONFIG_ESP_CONSOLE_USB_CDC: y    
hardware log and network log is working if only one UART in use
```

ssieb, 2024-07-06:
Try disabling serial logging.

swifty99, 2024-07-07:
thx.

This is a good workaround.

```
logger:
  level: DEBUG
  baud_rate: 0
```

ssieb, 2024-07-07:
Leave this open.  I think it should still work with USB logging.

swifty99, 2024-07-17:
alright. And it did work a few months ago.