esphome/issues #6018: Getting Failed config  update.http_request when updating two devices

SGXander, 2024-07-06:
### The problem

Updated ESPHome from 2024.6.3 to 2024.6.6 and two of my 12 devices report the below message when attempting to update. The github version of the config package looks to have the error line requested but "clean build files" fails with the same error. Is there a way to manually clear the cache or force re-download of the package?

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.6.4

### What platform are you using?

ESP32

### Board

ESP32-S3-BOX-3

### Component causing the issue

ota

### Example YAML snippet

```yaml
substitutions:
  name: "r07-v01"
  id_prefix: "r07_v01"
  friendly_name: R07-V01
  comment: "Office Voice Assistant"

api:
  encryption:
    key: "<REDACTED>"

packages:
  voice_assistant: github://esphome/firmware/voice-assistant/esp32-s3-box-3.yaml@main
  device_common: !include .device_common.yaml
  wifi_supplement: !include .wifi_supplement.yaml
  restart_supplement: !include .restart_supplement.yaml
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/r07-v01.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
Failed config

update.http_request: [source /data/packages/c46f54c1/voice-assistant/esp32-s3-box-3.yaml:90]
  
  Component update.http_request requires 'platform: http_request' in component 'ota'.
  platform: http_request
  id: update_http_request
  name: Firmware
  source: https://firmware.esphome.io/voice-assistant/esp32-s3-box-3/manifest.json
```

Teagan42, 2024-07-09:
Add:

```
ota:
- platform: http_request
  id: ota_http_request
http_request:
```

beginner2047, 2024-07-10:
> Add:
> 
> ```
> ota:
> - platform: http_request
>   id: ota_http_request
> http_request:
> ```

I have same error after upgraded to 2024.6.6. No issue before upgrade to this version. I have added those lines but the following exception is shown

Failed config

ota.http_request: [source /config/esphome/bluetooth-proxy.yaml:17]
  platform: http_request
  
  ID ota_http_request redefined! Check ota->1->id.
  id: ota_http_request

If I remove the line "id: ota_http_request", the following exception is shown.

Failed config

update.http_request: [source /data/packages/c46f54c1/bluetooth-proxy/esp32-generic.yaml:33]
  
  Too many candidates found for 'ota_id' type 'http_request::OtaHttpRequestComponent' One is 'ota_http_request'.
  platform: http_request
  id: update_http_request
  name: Firmware
  source: https://firmware.esphome.io/bluetooth-proxy/esp32-generic/manifest.json
  disabled_by_default: False
  entity_category: config
  update_interval: 6h

only install if I remove all lines under ota:

beginner2047, 2024-07-10:
Solved. No need to update the ota: section. Only need to add 

http_request:

beginner2047, 2024-07-10:
Resolved.

If you have platform: esphome for ota, you must have "-" before platform

e.g.
Change from 
ota:
  platform: esphome
  password: xxxxxx
to
ota:
  - platform: esphome
    password: xxxxxx

When esphome introduce "platform: esphome", I have followed the guideline to use "-" as day1. However, after one or two version upgraded, I failed to install with "- platform: esphome". Many people ask to add "platform: esphome" without "-". It really solved the issue when I removed "-". Now, in version 2024.6.6, it requests "-" again.

modem7, 2024-07-15:
> Resolved.
>
> If you have platform: esphome for ota, you must have "-" before platform
>
> e.g. Change from ota: platform: esphome password: xxxxxx to ota:
>
> * platform: esphome
>   password: xxxxxx
>
> When [...]

It'd be clearer if you used a code block in your explanation. 

Do you mean something like this?:

```yaml
ota:
  - platform: esphome
    password: !secret ota_password
```

thrawnis, 2024-07-19:
I couldn't follow beginner2047's response but I did exactly what modem7 said and it worked.  Thanks!

closed 2024-10-11 by ssieb