esphome/issues #6019: firmware.ota.bin is missing for BK72XX

jackrosenthal, 2024-07-06:
### The problem

The new OTA over HTTP component specifies in the documentation that [`.esphome/build/project/.pioenvs/project/firmware.ota.bin` should be the file](https://esphome.io/components/ota/http_request#:~:text=ESPHome%20CLI%3A%20the%20directory) to provide over HTTP.

While it appears this file exists for ESP8266/ESP32, it is not generated for BK72XX platforms.  Perhaps more file paths need documented, or this file needs generated for all platforms?

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

BK72XX

### Board

n/a

### Component causing the issue

ota

jackrosenthal, 2024-07-07:
This may be moot at the moment, as `http_request` isn't supported on bk72xx yet anyway (esphome/feature-requests#2416), meaning the file can't even be downloaded yet.

closed 2024-11-18 by github-actions[bot]

jackrosenthal, 2025-01-15:
This is possibly related to #4997.  Also pinging to drop the stale label.

ssieb, 2025-01-16:
I think it's the same file for OTA like it is for 8266.