esphome/issues #6023: ESP32-S3-Box-3 - web installer is not prompting to add to wifi

BigBobbas, 2024-07-08:
### The problem

Several people have mentioned that flashing with the latest version of firmware from the web-installer is not prompting to add to wifi. I have tested and also experience this. installing from https://esphome.io/projects/index.html and https://www.home-assistant.io/voice_control/s3_box_voice_assistant/

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32-IDF

### Board

ESP32 S3 Box 3

ryanmeanstech, 2024-07-10:
I am having this issue with an esp32-s3-zero board as well:

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 4MB
  variant: esp32s3
  framework:
    type: arduino

gshlomi, 2024-10-02:
Same here.
Any chance of an ETA of a fix?

ryanmeanstech, 2024-10-02:
Try bumping your output_power up - after much research this fixed the issue for me...

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  reboot_timeout: 0s
  enable_on_boot: True
  fast_connect: True
  output_power: 8.5

BigBobbas, 2024-10-02:
> Try bumping your output_power up - after much research this fixed the issue for me...
> 
> wifi: ssid: !secret wifi_ssid password: !secret wifi_password reboot_timeout: 0s enable_on_boot: True fast_connect: True output_power: 8.5

this is an issue using the websinstaller and the serial inprov not working allowing to providion wifi  in some cases , not related to a 
wifi connection issue.

ryanmeanstech, 2024-10-02:
I know - it was throwing the serial inprov error for wifi until I bumped the power up. It was not a wifi connection issue for me either.

thaynes43, 2024-10-28:
> I know - it was throwing the serial inprov error for wifi until I bumped the power up. It was not a wifi connection issue for me either.

How do you increase the power when the web installer resets the device? I am having the same problem and this is about all the info I've found so far. I tried adopting the device with the "prepare for fresh install" firmware on it then flashing the output power you posted but the S3-Box3 web installers erase it before installing the new firmware. I am still hit without an option to input my WiFi password.

BigBobbas, 2024-10-28:
> How do you increase the power when the web installer resets the device? I am having the same problem and this is about all the info I've found so far. I tried adopting the device with the "prepare [...]

if you are not prompted to add your device to your wifi after flashing the s3 box 3 firmware. The device should broadcast an access point. you can connect to this, then in a browser enter 192.168.4.1 in the address bar. This should then open a web page for you to enter your wifi credentials. Once saved, wait for a few seconds and then if still connected , disconnect from the AP and you should find your device is connected to your network. This issue has nothing to do with the comments above regarding power and are unrelated.

ryanmeanstech, 2024-10-28:
> How do you increase the power when the web installer resets the device? I am having the same problem and this is about all the info I've found so far. I tried adopting the device with the "prepare [...]

I have hard coded my Wifi SSID and Secret into ESPHome - it is automatically connecting for me once I flash the firmware with the higher output power.

thaynes43, 2024-10-29:
> if you are not prompted to add your device to your wifi after flashing the s3 box 3 firmware. The device should broadcast an access point. you can connect to this, then in a browser enter [...]

Thanks for the tip on the AP. I saw that in the as an option in the YAML but it didn't click that it was broadcasting it. I ended up flashing from home assistant's guide and then going to https://web.esphome.io/?dashboard_install and connecting via USB. The menu then had an option to change the WiFi password which worked IF I held down the top button. I was very glad to finally try this thing out but I quickly realized the volume was super low so I set off on how to update that. That was a little tricky but I found some patch someone made and once adopted in esphome I was able to build firmeware and use that web installer one again to install it. It just did not want to install from my instance of epshome but at least it built it.