esphome/issues #6026: boot loop with ESP-IDF 5.2.x

vladm, 2024-07-08:
### The problem

Trying to get my esp32c6 running on esp-idf 5.2, but getting boot loop.

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.7.1

### What platform are you using?

ESP32-IDF

### Board

esp32-c6-devkitc-1

### Example YAML snippet

```yaml
esp32:
  board: esp32-c6-devkitc-1
  variant: esp32c6
  framework:
    type: esp-idf
    version: 5.2.2
    source: https://github.com/tasmota/esp-idf/releases/download/v5.2.2.240704/esp-idf-v5.2.2.zip
    platform_version: 6.7.0
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y
```

### Anything in the logs that might be useful for us?

```txt
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyACM1 with baud rate 115200
[11:55:35]I (98) boot: No factory image, trying OTA 0
[11:55:35]I (103) esp_image: segment 0: paddr=00010020 vaddr=420e0020 size=3fde8h (261608) map
[11:55:35]I (166) esp_image: segment 1: paddr=0004fe10 vaddr=40800000 size=00208h (   520) load
[11:55:36]I (167) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=d01f4h (852468) map
[11:55:36]I (347) esp_image: segment 3: paddr=0012021c vaddr=40800208 size=15f00h ( 89856) load
[11:55:36]I (369) esp_image: segment 4: paddr=00136124 vaddr=40816110 size=038d0h ( 14544) load
[11:55:36]I (379) boot: Loaded app from partition at offset 0x10000
[11:55:36]I (408) boot: Set actual ota_seq=1 in otadata[0]
[11:55:36]I (408) boot: Disabling RNG early entropy source...
[11:55:36]I (419) cpu_start: Unicore app
[11:55:36]W (428) clk: esp_perip_clk_init() has not been implemented yet
[11:55:36]I (434) cpu_start: Pro cpu start user code
[11:55:36]I (434) cpu_start: cpu freq: 160000000 Hz
[11:55:36]I (435) cpu_start: Application information:
[11:55:36]I (437) cpu_start: Project name:     esp32c6-2
[11:55:36]I (442) cpu_start: App version:      2024.6.6
[11:55:36]I (447) cpu_start: Compile time:     Jul  8 2024 11:54:54
[11:55:36]I (454) cpu_start: ELF file SHA256:  193926897...
[11:55:36]I (459) cpu_start: ESP-IDF:          5.2.2.240704
[11:55:36]I (464) cpu_start: Min chip rev:     v0.0
[11:55:36]I (469) cpu_start: Max chip rev:     v0.99
[11:55:36]I (474) cpu_start: Chip rev:         v0.0
[11:55:36]I (479) heap_init: Initializing. RAM available for dynamic allocation:
[11:55:36]I (486) heap_init: At 4081FFD0 len 0005C640 (369 KiB): RAM
[11:55:36]I (492) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
[... 51 lines cut ...]
[11:55:41]40829970: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[11:55:41]40829990: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[11:55:41]408299b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[11:55:41]408299d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[11:55:41]408299f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[11:55:41]40829a10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000010
[11:55:41]40829a30: 0x00000000 0x408298dc 0x00000000 0x00000000 0x00000048 0x420f5348 0x00000001 0x7fc00000
[11:55:41]40829a50: 0x420e1bb0 0x0001c200 0x40829a90 0x00000086 0x00000200 0x00000000 0x00000000 0x00000000
[11:55:41]40829a70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40829a00 0x408298dc 0x00000204
[11:55:41]40829a90: 0x3b305b1b 0x5b6d3633 0x655b5d44 0x692d7073 0x303a6664 0x3a5d3030 0x305b1b20 0x6d32333b
[11:55:41]40829ab0: 0x36282049 0x20293839 0x6f697067 0x5047203a 0x365b4f49 0x49207c5d 0x7475706e 0x203a6e45
[11:55:41]40829ad0: 0x4f207c31 0x75707475 0x3a6e4574 0x207c3120 0x6e65704f 0x69617244 0x31203a6e 0x7550207c
[11:55:41]40829af0: 0x70756c6c 0x7c31203a 0x6c755020 0x776f646c 0x30203a6e 0x6e49207c 0x303a7274 0x305b1b20
[11:55:41]40829b10: 0x5b1b0a6d 0x1b006d30 0x00000000 0xb33fffff 0x00000000 0x408362f8 0x40836794 0x40836910
[11:55:41]40829b30: 0x6d305b00 0x00000000 0x420464f0 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[11:55:41]40829b50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[11:55:41]40829b70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[11:55:41]40829b90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[11:55:41]
[11:55:41]
[11:55:41]
[11:55:41]ELF file SHA256: 193926897
[11:55:41]
[11:55:41]Rebooting...
```

vladm, 2024-07-08:
adding some info here - looks like some other components (text_sensor)  is causing a failure, excluding those solves the problem.

closed 2024-07-08 by vladm

ssieb, 2024-07-08:
Don't file issues for C6 boards as they aren't supported yet.  Only file an issue if you can reproduce the issue on a supported device.