esphome/issues #6027: LVGL Background image error

sdomotica, 2024-07-08:
### The problem

This is the error log
```
INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/touch2.yaml...
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing touch-panel (board: esp32-s3-devkitc-1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- lvgl @ 8.3.11
Compiling .pioenvs/touch-panel/src/main.o
/config/esphome/touch2.yaml: In lambda function:
/config/esphome/touch2.yaml:1300:37: error: 'lv_img_from' was not declared in this scope
/config/esphome/touch2.yaml:1300:37: note: suggested alternative: 'lv_img_t'
In file included from src/esphome.h:60,
                 from src/main.cpp:3:
src/esphome/components/lvgl/lvgl_esphome.h: At global scope:
src/esphome/components/lvgl/lvgl_esphome.h:38:19: warning: 'lv_color_t esphome::lvgl::lv_color_from(esphome::Color)' defined but not used [-Wunused-function]
 static lv_color_t lv_color_from(Color color) { return lv_color_make(color.red, color.green, color.blue); }
                   ^~~~~~~~~~~~~
*** [.pioenvs/touch-panel/src/main.o] Error 1
========================== [FAILED] Took 7.78 seconds ==========================

```

This is the code

```
image:
  - file: https://esphome.io/_images/logo.png
    id: boot_logo
    resize: 200x200
    type: RGB565

  - file: "fonts/001.png"
    id: wallsconce_180
    resize: 20x20
    
  - file: "fonts/carbon.jpg"
    id: wall_carbon
    resize: 200x200
    type: RGB565    

lvgl:
  displays:
    - display_id: my_display
  touchscreens:
    - touchscreen_id: my_touchscreen
  on_idle:
    timeout: 90s
    then:
      - logger.log: "LVGL is idle"
      - light.turn_off: backlight
      - lvgl.pause:

  disp_bg_image: wall_carbon
```

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.1

### What platform are you using?

ESP32-IDF

### Board

guition-4-480x480-esp32-s3-4848s040

### Component causing the issue

lvgl - disp_bg_image

### Example YAML snippet

```yaml
image:
  - file: https://esphome.io/_images/logo.png
    id: boot_logo
    resize: 200x200
    type: RGB565

  - file: "fonts/001.png"
    id: wallsconce_180
    resize: 20x20
    
  - file: "fonts/carbon.jpg"
    id: wall_carbon
    resize: 200x200
    type: RGB565    

lvgl:
  displays:
    - display_id: my_display
  touchscreens:
    - touchscreen_id: my_touchscreen
  on_idle:
    timeout: 90s
    then:
      - logger.log: "LVGL is idle"
      - light.turn_off: backlight
      - lvgl.pause:

  disp_bg_image: wall_carbon
```

HeMan, 2024-07-10:
I think you could get better response if you add this as a comment to https://github.com/esphome/esphome/pull/6363

sdomotica, 2024-07-10:
### The problem

This is the error log
```
INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/touch2.yaml...
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing touch-panel (board: esp32-s3-devkitc-1; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- lvgl @ 8.3.11
Compiling .pioenvs/touch-panel/src/main.o
/config/esphome/touch2.yaml: In lambda function:
/config/esphome/touch2.yaml:1300:37: error: 'lv_img_from' was not declared in this scope
/config/esphome/touch2.yaml:1300:37: note: suggested alternative: 'lv_img_t'
In file included from src/esphome.h:60,
                 from src/main.cpp:3:
src/esphome/components/lvgl/lvgl_esphome.h: At global scope:
src/esphome/components/lvgl/lvgl_esphome.h:38:19: warning: 'lv_color_t esphome::lvgl::lv_color_from(esphome::Color)' defined but not used [-Wunused-function]
 static lv_color_t lv_color_from(Color color) { return lv_color_make(color.red, color.green, color.blue); }
                   ^~~~~~~~~~~~~
*** [.pioenvs/touch-panel/src/main.o] Error 1
========================== [FAILED] Took 7.78 seconds ==========================

```

This is the code

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.1

### What platform are you using?

ESP32-IDF

### Board

guition-4-480x480-esp32-s3-4848s040

### Component causing the issue

lvgl - disp_bg_image

### Example YAML snippet

```yaml

external_components:
 - source: github://clydebarrow/esphome@lvgl_original
   refresh: 10min
   components: [ lvgl]

image:
  - file: https://esphome.io/_images/logo.png
    id: boot_logo
    resize: 200x200
    type: RGB565

  - file: "fonts/001.png"
    id: wallsconce_180
    resize: 20x20
    
  - file: "fonts/carbon.jpg"
    id: wall_carbon
    resize: 200x200
    type: RGB565    

lvgl:
  displays:
    - display_id: my_display
  touchscreens:
    - touchscreen_id: my_touchscreen
  on_idle:
    timeout: 90s
    then:
      - logger.log: "LVGL is idle"
      - light.turn_off: backlight
      - lvgl.pause:

  disp_bg_image: wall_carbon
```

### Anything in the logs that might be useful for us?

/config/esphome/touch2.yaml:1300:37: error: 'lv_img_from' was not declared in this scope

### Additional information

_No response_

clydebarrow, 2024-07-10:
I don't know why you are using the `lvgl_original` branch (which is now deleted anyway). Use the PR or the `lvgl` branch.

closed 2024-07-10 by clydebarrow

sdomotica, 2024-07-10:
@clydebarrow  sorry, you are right, I started form an Home Assistant topic, now I'm using as per indication this source.
Sorry again
- source: github://pr#6363