esphome/issues #6028: Conflicting sprintf Linter Requiremens Prevent Pull Request

TikiBill, 2024-07-08:
### The problem

When variables are defined as uint32_t and one tries to print the values with ESP_LOG*, the clang-tidy linter wants the format string to use %u while the python ones want %lu. Obviously these cannot both be satisfied at the same time, which prevents a pull request which logs unit32_t values from passing the lint tests.

This is in relation to https://github.com/esphome/esphome/pull/7055 and if you look at the history, I switched between the two a couple times. Example images below. As a work-around, I may change to using plain-old ints since I don't even need 16 bits of range and the code does bounds checking.

### Which version of ESPHome has the issue?

Latest from dev

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

N/A

### Component causing the issue

N/A

### Additional information

clang-tidy:
[image]

test_build_components:
[image]

ssieb, 2024-07-08:
I don't understand where the python linter comes in for C++ code.

kbx81, 2024-07-10:
You need to `#include <cinttypes>` and use the macro `PRIu32` wherever you've used `uint32_t` and `PRId32` wherever you've used `int32_t`. This is because of how IDF 5+ define the type (int vs. long int).

TikiBill, 2024-07-11:
Thanks @kbx81 for your guidance. Not a bug per-se, just my lack of c++ experience.

closed 2024-07-11 by TikiBill