esphome/issues #6029: ESP32-C3 consumptions and internal temperature peak

Rihan9, 2024-07-09:
### The problem

Hi all,
I'm not sure if here is the correct plate to write about it.
I recently bought some devKit for ESP32-C3 (SuperMini version with and without led, you can easily find them on aliexpress) to use them as brain on my multi-sensor device (presence, light, temperature and humidity). I then noticed that the temperature sensor have some strange peaks of temperature and realize that the problem was on the esp board. I remove all the attached sensor, write a simple yaml configuration and let the esp run alone through the night. I didn't solve the issue, so something is strange on the esp itself.
I'm posting here* the code and here** the screenshot of the internal temperature monitoring.

I have already tried to change the device. The results showed here are from the second device

I'm going to disable one peripheral at a time and restart the monitoring. I'll post in this issue the results.

*
```
esphome:
  name: esp32c3-test
  friendly_name: Esp32-C3 Test
  platformio_options:
    board_build.f_cpu: 80000000L

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "SLfIfLtuPmm0GUBQcgJ7EdWbAJNNSVkV+dBgktriIw4="

ota:
  password: "372d535c23d038f3f36a2c3f49451188"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-C3-Test Fallback Hotspot"
    password: "QzsVJ8kcMTBq"

captive_portal:

uart:
  id: uart_bus
  rx_pin: 
    number: GPIO20
    mode:
      input: true
      pullup: true
  tx_pin: 
    number: GPIO21
    mode:
      input: true
      pullup: true
  baud_rate: 256000
  parity: NONE
  stop_bits: 1
  data_bits: 8

i2c:
  sda: GPIO10
  scl: GPIO09
  scan: false
  id: bus_a

sensor:
  - platform: internal_temperature
    name: "Internal Temperature"
    entity_category: "diagnostic"

light:
  - platform: status_led
    name: "Status Led"
    pin:
      number: GPIO08
      inverted: true
```

**
[image]

### Which version of ESPHome has the issue?

2024.5.5

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2024.6.4

### What platform are you using?

ESP32

### Board

esp32-c3-devkitm-1

Rihan9, 2024-07-09:
Hi,
I completed the tests with UART and I2C disabled, with the same behavior. I noticed that 2 board with a different start time have the consumption peak at the same hour

gabest11, 2024-07-12:
```
esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_BT_CTRL_MODEM_SLEEP: y
```

Rihan9, 2024-07-12:
I'll check.
This is the first time i've changed the framework, so I don't know the impact of this change. I'll do more test if the framework change resolve the issue. Thank.

P.S.
I checked if the power_safe_mode for the wifi change this issue but no, some behavior

Rihan9, 2024-07-12:
Hi,
another consumption/temperature peak registered at 13:35 GMT+2.
The config option suggested wasn't the answer.

I start to think that there is some sort of problem on the wifi module, like the router pinging all the devices for some sort of reason or something like this. I don't want to set a sniffer on my entire network, it will be a nightmare, but I don't see anithing in the device log even in VERY VERBOSE debug level.

I'll try to set up a different AP and check this, but the routing table to keep the 2 network separated without losing the monitoring is going to be a problem. I'll update this issue with the results if I find a way to doing it

Rihan9, 2024-07-22:
Hi all! The problem seems to be related to my Wi-Fi setup. I used the project esp32_nat_router and it works perfectly. I'll do some other tests to find why this strange behavior, but I'm not an expert in the Wi-Fi protocol.
In the screenshot below you can clearly see the moment when I switched the WiFi
[image]

Rihan9, 2024-07-25:
Hi,
I completed my tests using another wifi6 network and the problem exposed in this defect does not accour. I think there is some kind  of problem in my router.

The router itself does not have a decent way to debug the problem, so I'm closing this issues because I don't have other tests that can maybe be useful for other people, and the problem is not related tu the Esp32-c3 or ESPHome.

Have a good day!

closed 2024-07-25 by Rihan9