esphome/issues #6033: M5Atom Echo Ready-Made-Project fails after adoption due to partion problem

darth-hp, 2024-07-10:
### The problem

I flashed a M5Atom Echo using https://esphome.io/projects/ as Voice assistant. After configuring it appeared in my esphome dashboard. I then tried to change the name and yaml file name. Compile and flash went fine. But then it did not come back. In the log you can see it is stuck in a boot loop with this error message:

```
[13:27:06]I (57) boot: ESP-IDF 4.4.7 2nd stage bootloader
[13:27:06]I (57) boot: compile time 10:30:30
[13:27:06]I (58) boot: Multicore bootloader
[13:27:06]I (61) boot: chip revision: v1.1
[13:27:06]I (65) boot.esp32: SPI Speed      : 40MHz
[13:27:06]I (70) boot.esp32: SPI Mode       : DIO
[13:27:06]I (74) boot.esp32: SPI Flash Size : 4MB
[13:27:06]I (79) boot: Enabling RNG early entropy source...
[13:27:06]I (84) boot: Partition Table:
[13:27:06]I (88) boot: ## Label            Usage          Type ST Offset   Length
[13:27:06]I (95) boot:  0 otadata          OTA data         01 00 00009000 00002000
[13:27:06]I (103) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[13:27:06]I (110) boot:  2 app0             OTA app          00 10 00010000 001c0000
[13:27:06]I (118) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[13:27:06]I (125) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[13:27:06]I (133) boot: End of partition table
[13:27:06]I (137) boot: No factory image, trying OTA 0
[13:27:06]I (142) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=659f8h (416248) map
[13:27:06]I (301) esp_image: segment 1: paddr=00075a20 vaddr=3ffbdb60 size=051bch ( 20924) load
[13:27:06]I (310) esp_image: segment 2: paddr=0007abe4 vaddr=40080000 size=05434h ( 21556) load
[13:27:06]I (319) esp_image: segment 3: paddr=00080020 vaddr=400d0020 size=136e50h (1273424) map
[13:27:07]I (780) esp_image: segment 4: paddr=001b6e78 vaddr=40085434 size=19b50h (105296) load
[13:27:07]E (823) esp_image: Image length 1837552 doesn't fit in partition length 1835008
[13:27:07]E (823) boot: OTA app partition slot 0 is not bootable
[13:27:07]E (826) esp_image: image at 0x1d0000 has invalid magic byte (nothing flashed here?)
[13:27:07]E (834) boot: OTA app partition slot 1 is not bootable
[13:27:07]E (840) boot: No bootable app partitions in the partition table
[13:27:07]ets Jun  8 2016 00:22:57
[13:27:07]
[13:27:07]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:27:07]configsip: 188777542, SPIWP:0xee
[13:27:07]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:27:07]mode:DIO, clock div:2
[13:27:07]load:0x3fff0030,len:6652
[13:27:07]ho 0 tail 12 room 4
[13:27:07]load:0x40078000,len:15052
[13:27:07]load:0x40080400,len:3836
[13:27:07]entry 0x4008069c
```

Here the two YAMLs used:
mini-jarvis-01.yaml
```yaml
substitutions:
  name: "mini-jarvis-01"
  friendly_name: "Mini-Jarvis-01"

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main

<<: !include .common.yaml
```

.common.yaml
```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .fritz.box
  fast_connect: True 
  ap:

api:
  encryption:
    key: !secret esp_api_key
```

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.1 HAOS 12.4

### What platform are you using?

ESP32-IDF

### Board

M5Atom Echo

### Component causing the issue

unknown

Ceralor, 2024-07-10:
Same issue here. I tried re-running the command generated manually and tried reflashing from 0x0 with firmware.factory.bin but the device is showing that the partition is smaller than the binary.

Ceralor, 2024-07-11:
I converted the OTA yaml to a local config to bypass the issues with the firmware image being too large thanks to the new components, which do not fit in the partition structure. Was this verified prior to release of the new feature with this firmware? I'm only raising this question because this is the second time in 2024 there's been an issue with firmware being generated that's too large for the partition structure.

darth-hp, 2024-07-11:
> I converted the OTA yaml to a local config to bypass the issues with the firmware image being too large thanks to the new components, which do not fit in the partition structure. Was this verified [...]

How did you do that?

Ceralor, 2024-07-11:
> How did you do that?

Remove the reference to the dashboard part, https requests, and replace the wifi section with a configured one for my wifi, ditching improv, and update the OTA section to just be the one entry for platform esphome using existing OTA password.

You can see an example here, just modify to use your own secrets if so. https://github.com/Ceralor/esphome_configs/blob/main/bedroom-speaker.yaml

randomletters, 2024-07-12:
I followed the instructions from @Ceralor (except for the OTA stuff since I couldn't get the secret to work) and it worked for me.  I was able to get past the size issue.  My steps were:

1. Goto the yaml file on github shown in "packages"
2. Copy the contents into the esphome yaml
3. Comment out the package reference
4. Fix the errors caused by duplicate keys by rearranging the yaml and commenting the default values out
5. Remove the dashboard, https requests, and wifi options mentioned above
6. Install

darth-hp, 2024-07-12:
I used https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml and removed dashboard_import, http_request, update, improv_serial and esp32_improv. That did it

<details><summary>m5atom-echo.yaml</summary>

```yaml
substitutions:
  name: mini-jarvis-01
  friendly_name: Mini-Jarvis-01

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: m5stack.atom-echo-voice-assistant
    version: "24.7.4.1"
  compile_process_limit: 2

<<: !include .wifi.yaml
<<: !include .ota.yaml
<<: !include .api.yaml

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

logger:

button:
  - platform: safe_mode
    id: button_safe_mode
    name: Safe Mode Boot

  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset

i2s_audio:
  - id: i2s_audio_bus
    i2s_lrclk_pin: GPIO33
    i2s_bclk_pin: GPIO19

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO23
    adc_type: external
    pdm: true

speaker:
  - platform: i2s_audio
    id: echo_speaker
    i2s_dout_pin: GPIO22
    dac_type: external
    mode: mono

voice_assistant:
  id: va
  microphone: echo_microphone
  speaker: echo_speaker
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  vad_threshold: 3
  on_listening:
    - light.turn_on:
        id: led
        blue: 0%
[... 195 lines cut ...]
  - platform: template
    id: timer_ringing
    optimistic: true
    internal: true
    restore_mode: ALWAYS_OFF
    on_turn_on:
      - delay: 15min
      - switch.turn_off: timer_ringing

external_components:
  - source: github://pr#5230
    components:
      - esp_adf
    refresh: 0s
  - source: github://jesserockz/esphome-components
    components: [file]
    refresh: 0s

esp_adf:

file:
  - id: timer_finished_wave_file
    file: https://github.com/esphome/firmware/raw/main/voice-assistant/sounds/timer_finished.wav
```
</details>

txwindsurfer, 2024-07-18:
The new update of ESPHome to 2024.7.0 doesn't fix this issue.  I had assumed the issue will get resolved without manually taking control over the yaml. Am I incorrect?

darth-hp, 2024-09-16:
Workaround did the trick. Closing