esphome/issues #6035: `!remove` and replace give a duplicate key error

HarvsG, 2024-07-10:
### The problem

the `!remove` tag can be used to remove an unwanted component:
E.g
```yml
voice_assistant:
  on_timer_finished: !remove
```
however if one tries to replace it
```yml
voice_assistant:
  on_timer_finished: !remove
  on_timer_finished:
    - voice_assistant.stop:
    - switch.turn_on: timer_ringing
    - wait_until:
        not:
  ...

Gives a duplicate key error
```
Duplicate key "on_timer_finished"
  in "/config/m5-stack-atom-echo-lt.yaml", line 39, column 3
NOTE: Previous declaration here:
  in "/config/m5-stack-atom-echo-lt.yaml", line 38, column 3
```

You might say why not use `!extend` but that only works for dictionaries

```yml
voice_assistant:
  id: !extend va
  on_timer_finished:
    - voice_assistant.stop:
    - switch.turn_on: timer_ringing
    - wait_until:
        not:
   ... 
```
Gives an error `Source for extension of ID 'va' was not found.`

### Which version of ESPHome has the issue?

2024.6.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2024.7.0

### What platform are you using?

ESP32

### Board

M5 Stack Atom Echo

### Component causing the issue

configuration-types

### Example YAML snippet

```yaml
substitutions:
  name: m5stack-atom-echo-b836b0
  friendly_name: M5Stack Atom Echo b836b0
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: +some key=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Until https://github.com/esphome/firmware/pull/233 is fixed/merged
update:
  - platform: http_request
    id: !remove update_http_request

speaker:
  - platform: i2s_audio
    id: !extend echo_speaker
    i2s_dout_pin: GPIO21 # <- It is actually on 22, so this disables the speaker

voice_assistant:
  id: va
  on_tts_end:
     - homeassistant.service:
         service: media_player.play_media
         data:
           entity_id: media_player.study_speaker # <- this is hard-coded
           media_content_id: !lambda 'return x;'
           media_content_type: music
           announce: "true"
  on_timer_finished: !remove
  on_timer_finished:
    - voice_assistant.stop:
    - switch.turn_on: timer_ringing
    - wait_until:
        not:
          microphone.is_capturing:
    - light.turn_on:
        id: led
        red: 0%
        green: 100%
        blue: 0%
        brightness: 100%
        effect: "Fast Pulse"
    - while:
        condition:
          switch.is_on: timer_ringing
        then:
          - homeassistant.service:
              service: media_player.play_media
              data:
                entity_id: media_player.study_speaker # <- this is hard-coded
                media_content_id: id(timer_finished_wave_file)
                media_content_type: music
                announce: "true"
          - delay: 1s
    - wait_until:
        not:
          speaker.is_playing:
    - light.turn_off: led
    - switch.turn_off: timer_ringing
    - if:
        condition:
          switch.is_on: use_wake_word
        then:
          - voice_assistant.start_continuous:
          - script.execute: reset_led
```

HarvsG, 2024-07-10:
If this is a limitation of the YAML spec then solutions could be:

Option 1: add a new `!replace` tag
Options 2: allow this:
```yml
voice_assistant:
  on_timer_finished: !remove
  stuff here that wont be deleted (it currently is)
``` 
Option 3: Allow `!extend` to act in this scenario

mwaterbu, 2025-11-27:
Are there any workarounds for this? I am experiencing the same issue.